awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #1026: Is there any way to create kinesis video producer from any android or web device without needing to install producer-sdk ?

sumitmandal13, 2023-06-28:
I want to create a kinesis video stream from 10 different mobile and web device? How can I achieve this without installing producer-sdk in their devices?
Background: I want to use all their streams and pass them to open-cv for further post processing.

hassanctech, 2023-07-05:
The creation API `CreateStream` is available via the AWS Amplify Mobile SDKs.

https://github.com/aws-amplify/aws-sdk-android
https://github.com/aws-amplify/aws-sdk-ios
https://github.com/aws/aws-sdk-js-v3

Can you please clarify what you mean by:
> I want to use all their streams

If you want to send video data to the streams you will need to invoke the `PutMedia` API which is *not* available via the Mobile SDKs.

sumitmandal13, 2023-07-05:
That would create the stream. RIght? How can I be the man in middle of master and viewer? Such that I can process a video in let's say OpenCV and share the processed video to the viewer? To put it more simply, let's imagine I want to create an app which tell how many times a viewer blinked on the live stream and pass number of times he blinked to the viewer. 
How can I achieve this, using Signalling channel in KVS?

hassanctech, 2023-07-05:
> How can I be the man in middle of master and viewer?

This is not supported out of the box.  Can you please explain your architecture a bit more, what device (SDK) is master and what SDK is viewer?

There is no SFU or any type of processing unit in the middle, whatever master sends is what viewer will receive.  If you want to implement something different either you will have to implement something like your own MCU to do this.  

You can create a master/viewer setup where the viewer gets the stream and does the post processing, then it becomes a "master" which then your viewers consume from.  But out of the box, the only way to modify a stream is at the source on the master side before you "send" it over to the viewer, once you've sent it, that is what will go to the viwer.

closed 2023-07-05 by hassanctech

sumitmandal13, 2023-07-05:
[image]

I am taking Master producer from browser which is js SDK, My Viewer is AWS LAMBDA - javascript sdk and sagemaker processing unit is python or opencv. There's no such MCU or IOT involved but I just want to establish a video call connection b/w a smart phone browser and a processing unit, lambda/sagemaker.