awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #1027: [CONTRIBUTION] Add fragment number inside persisted-ack signal for gstreamer applications can obtain this information

aman-aguptaaa, 2023-06-29:
PLEASE ADD THE APPROPRIATE TAG TO ALLOW BUCKETIZATION OF THE SOLUTIONS. IF THE TAG IS NOT AVAILABLE, DO NOT WORRY, WE WILL TAKE CARE OF IT!

** Describe the issue you are trying to solve **
Add a one line description of the issue you are trying to solve
Add fragment number inside persisted-ack signal, so that gstreamer applications can obtain this information.

** Details of the changes **
Give a couple of points to describe the changes you have made
Add fragment number parameter inside persisted-ack signal in [fragmentAcCallback](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/add-signals-kvssink/src/gstreamer/KvsSinkStreamCallbackProvider.cpp#L87) on add-signals-kvssink branch.

** Test cases **
Give a detailed description of the tests you are running, if applicable
Listen to persisted-ack signal and check that fragment number is being received from the signal.

** Additional context **
Any details that can be preserved for the future

disa6302, 2023-06-29:
@aman-aguptaaa ,

If you are looking for changes in that branch, please use the develop branch instead. The changes from this branch has been merged into develop and has some other fixes too, including your request. Closing this assuming answered.

closed 2023-06-29 by disa6302

aman-aguptaaa, 2023-06-29:
Thanks @disa6302 , did not know about the develop branch. 
Yes, that branch has the desired functionality I was looking for.