awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #1030: [BUG] cannot build on Apple M1 Pro

robert-hanuschke, 2023-07-05:
**Logging**

```
cmake .. -DBUILD_GSTREAMER_PLUGIN=ON               
-- Kinesis Video Cpp Producer path is /my/home/path/amazon-kinesis-video-streams-producer-sdk-cpp
-- dependencies install path is /my/home/path/amazon-kinesis-video-streams-producer-sdk-cpp/open-source
-- Begin building dependencies.
-- autoconf already built
-- automake already built
-- log4cplus already built
-- Finished building dependencies.
-- Configuring done (0.1s)
-- Generating done (0.0s)
-- Build files have been written to: /my/home/path/amazon-kinesis-video-streams-producer-sdk-cpp/dependency/libkvscproducer
[ 11%] Performing update step for 'libkvscproducer-download'
[ 22%] No patch step for 'libkvscproducer-download'
[ 33%] No configure step for 'libkvscproducer-download'
[ 44%] No build step for 'libkvscproducer-download'
[ 55%] No install step for 'libkvscproducer-download'
[ 66%] No test step for 'libkvscproducer-download'
[ 77%] Completed 'libkvscproducer-download'
[100%] Built target libkvscproducer-download
-- Kinesis Video Producer path is /my/home/path/amazon-kinesis-video-streams-producer-sdk-cpp/dependency/libkvscproducer/kvscproducer-src
-- Begin building dependencies.
-- Configuring done (0.1s)
-- Generating done (0.0s)
-- Build files have been written to: /my/home/path/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/libopenssl
[ 11%] Creating directories for 'project_libopenssl'
[ 22%] Performing download step (git clone) for 'project_libopenssl'
Cloning into 'project_libopenssl'...
HEAD is now at e2e09d9fba Prepare for 1.1.1g release
Submodule 'boringssl' (https://boringssl.googlesource.com/boringssl) registered for path 'boringssl'
Submodule 'krb5' (https://github.com/krb5/krb5) registered for path 'krb5'
Submodule 'pyca.cryptography' (https://github.com/pyca/cryptography.git) registered for path 'pyca-cryptography'
Cloning into '/my/home/path/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/libopenssl/build/src/project_libopenssl/boringssl'...
Cloning into '/my/home/path/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/libopenssl/build/src/project_libopenssl/krb5'...
Cloning into '/my/home/path/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/libopenssl/build/src/project_libopenssl/pyca-cryptography'...
Submodule path 'boringssl': checked out '2070f8ad9151dc8f3a73bffaa146b5e6937a583f'
Submodule path 'krb5': checked out '890ca2f401924cdcb88f493950b04700bbe52db3'
Submodule path 'pyca-cryptography': checked out '09403100de2f6f1cdd0d484dcb8e620f1c335c8f'
[ 33%] Performing update step for 'project_libopenssl'
[ 44%] No patch step for 'project_libopenssl'
[ 55%] Performing configure step for 'project_libopenssl'
Operating system: i686-apple-darwinDarwin Kernel Version 21.6.0: Mon Apr 24 21:11:17 PDT 2023; root:xnu-8020.240.18.701.5~1/RELEASE_ARM64_T6000
Configuring OpenSSL version 1.1.1g (0x1010107fL) for darwin-i386-cc
Using os-specific seed configuration
Creating configdata.pm
Creating Makefile

**********************************************************************
***                                                                ***
***   OpenSSL has been successfully configured                     ***
***                                                                ***
***   If you encounter a problem while building, please open an    ***
***   issue on GitHub <https://github.com/openssl/openssl/issues>  ***
***   and include the output from the following command:           ***
***                                                                ***
***       perl configdata.pm --dump                                ***
***                                                                ***
***   (If you are new to OpenSSL, you might want to consult the    ***
***   'Troubleshooting' section in the INSTALL file first)         ***
***                                                                ***
**********************************************************************
[ 66%] Performing build step for 'project_libopenssl'
ar: creating archive apps/libapps.a
ar: creating archive libcrypto.a
/Library/Developer/CommandLineTools/usr/bin/ranlib: file: libcrypto.a(async_null.o) has no symbols
/Library/Developer/CommandLineTools/usr/bin/ranlib: file: libcrypto.a(async_win.o) has no symbols
/Library/Developer/CommandLineTools/usr/bin/ranlib: file: libcrypto.a(cms_cd.o) has no symbols
/Library/Developer/CommandLineTools/usr/bin/ranlib: file: libcrypto.a(dso_dl.o) has no symbols
[... 627 lines cut ...]

**To Reproduce**
Steps to reproduce the behavior:
1. `brew install pkg-config openssl cmake gstreamer gst-plugins-base gst-plugins-good gst-plugins-bad gst-plugins-ugly log4cplus gst-libav`
2. `git clone https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp.git`
3. `mkdir -p amazon-kinesis-video-streams-producer-sdk-cpp/build`
4. `cd amazon-kinesis-video-streams-producer-sdk-cpp/build`
5. `cmake -DBUILD_GSTREAMER_PLUGIN=TRUE ..`

**Expected behavior**
cmake command to finish successfully for continuing with make in the installation instruction.

**SDK version number**
3.3.1

**Open source building**

**Screenshots**

**Platform (please complete the following information):**
 - OS: MacOS
 - Processor: Apple M1 Pro
 - Version: 12.6.6

**Additional context**

hassanctech, 2023-07-05:
Yes this is an outstanding issue we have addressed in our develop branch, but have not yet cut a release with the change:  https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/tree/develop

The crux of the issue is basically openssl 1.1.1g does not support a build for Apple arm64 and needs to be updated.  This is done in the `develop` branch.  Can you please try that out it should work.  In that branch we have updated openssl to `1.1.1t`.

robert-hanuschke, 2023-07-05:
I checked out the develop branch and tried again. Getting a different error:

```
-- The C compiler identification is AppleClang 14.0.0.14000029
-- The CXX compiler identification is AppleClang 14.0.0.14000029
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Library/Developer/CommandLineTools/usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Kinesis Video Cpp Producer path is /my/home/path/amazon-kinesis-video-streams-producer-sdk-cpp
-- dependencies install path is /my/home/path/amazon-kinesis-video-streams-producer-sdk-cpp/open-source
-- Begin building dependencies.
CMake Warning (dev) at /opt/homebrew/Cellar/cmake/3.26.4/share/cmake/Modules/ExternalProject.cmake:3091 (message):
  The DOWNLOAD_EXTRACT_TIMESTAMP option was not given and policy CMP0135 is
  not set.  The policy's OLD behavior will be used.  When using a URL
  download, the timestamps of extracted files should preferably be that of
  the time of extraction, otherwise code that depends on the extracted
  contents might not be rebuilt if the URL changes.  The OLD behavior
  preserves the timestamps from the archive instead, but this is usually not
  what you want.  Update your project to the NEW behavior or specify the
  DOWNLOAD_EXTRACT_TIMESTAMP option with a value of true to avoid this
  robustness issue.
Call Stack (most recent call first):
  /opt/homebrew/Cellar/cmake/3.26.4/share/cmake/Modules/ExternalProject.cmake:4208 (_ep_add_download_command)
  CMakeLists.txt:8 (ExternalProject_Add)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Configuring done (0.0s)
-- Generating done (0.0s)
-- Build files have been written to: /my/home/path/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/libautoconf
[ 11%] Creating directories for 'project_libautoconf'
[ 22%] Performing download step (download, verify and extract) for 'project_libautoconf'
-- Downloading...
   dst='/my/home/path/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/libautoconf/build/src/autoconf-2.69.tar.gz'
   timeout='none'
   inactivity timeout='none'
-- Using src='http://ftp.gnu.org/gnu/autoconf/autoconf-2.69.tar.gz'
-- [download 0% complete]
-- [download 1% complete]
-- [download 2% complete]
-- [download 4% complete]
-- [download 5% complete]
-- [download 6% complete]
-- [download 7% complete]
-- [download 8% complete]
-- [download 9% complete]
-- [download 10% complete]
-- [download 11% complete]
-- [download 12% complete]
-- [download 13% complete]
-- [download 14% complete]
-- [download 15% complete]
-- [download 23% complete]
-- [download 24% complete]
-- [download 25% complete]
-- [download 26% complete]
-- [download 27% complete]
-- [download 28% complete]
-- [download 29% complete]
-- [download 30% complete]
-- [download 32% complete]
-- [download 33% complete]
-- [download 34% complete]
-- [download 35% complete]
[... 802 lines cut ...]
pyexecdir=''
pythondir=''
sbindir='${exec_prefix}/sbin'
sharedstatedir='${prefix}/com'
sysconfdir='${prefix}/etc'
target='arm-apple-darwin21.6.0'
target_alias=''
target_cpu='arm'
target_os='darwin21.6.0'
target_vendor='apple'

## ----------- ##
## confdefs.h. ##
## ----------- ##

/* confdefs.h */
#define PACKAGE_NAME "log4cplus"
#define PACKAGE_TARNAME "log4cplus"
#define PACKAGE_VERSION "2.0.1"
#define PACKAGE_STRING "log4cplus 2.0.1"
#define PACKAGE_BUGREPORT ""
#define PACKAGE_URL ""

configure: exit 77
```

hassanctech, 2023-07-05:
Yes now looks like you're hitting: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/issues/978, you'll just have to run `cmake` a second time and it should work!

robert-hanuschke, 2023-07-05:
There is progress - but I unfortunately run into another error which does not go away by repeating the `cmake` command: 

```
-- Kinesis Video Cpp Producer path is /my/home/path/amazon-kinesis-video-streams-producer-sdk-cpp
-- dependencies install path is /my/home/path/amazon-kinesis-video-streams-producer-sdk-cpp/open-source
-- Begin building dependencies.
-- autoconf already built
-- automake already built
-- log4cplus already built
-- Finished building dependencies.
-- Configuring done (0.1s)
-- Generating done (0.0s)
-- Build files have been written to: /my/home/path/amazon-kinesis-video-streams-producer-sdk-cpp/dependency/libkvscproducer
[ 11%] Performing update step for 'libkvscproducer-download'
[ 22%] No patch step for 'libkvscproducer-download'
[ 33%] No configure step for 'libkvscproducer-download'
[ 44%] No build step for 'libkvscproducer-download'
[ 55%] No install step for 'libkvscproducer-download'
[ 66%] No test step for 'libkvscproducer-download'
[ 77%] Completed 'libkvscproducer-download'
[100%] Built target libkvscproducer-download
-- Kinesis Video Producer path is /my/home/path/amazon-kinesis-video-streams-producer-sdk-cpp/dependency/libkvscproducer/kvscproducer-src
-- Begin building dependencies.
-- Building non-local OpenSSL
-- openssl already built
-- Configuring done (0.0s)
-- Generating done (0.0s)
CMake Warning:
  Manually-specified variables were not used by the project:

    USE_OPENSSL

-- Build files have been written to: /my/home/path/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/libcurl
[ 11%] Creating directories for 'project_libcurl'
[ 22%] Performing download step (git clone) for 'project_libcurl'
Cloning into 'project_libcurl'...
HEAD is now at 2cfac302f RELEASE-NOTES: 7.68.0
[ 33%] Performing update step for 'project_libcurl'
[ 44%] No patch step for 'project_libcurl'
[ 55%] Performing configure step for 'project_libcurl'
-- The C compiler identification is AppleClang 14.0.0.14000029
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
CMake Warning at CMakeLists.txt:50 (message):
  the curl cmake build system is poorly maintained.  Be aware

-- curl version=[7.68.0-DEV]
-- Performing Test HAVE_SOCKADDR_IN6_SIN6_ADDR
-- Performing Test HAVE_SOCKADDR_IN6_SIN6_ADDR - Success
-- Performing Test HAVE_SOCKADDR_IN6_SIN6_SCOPE_ID
-- Performing Test HAVE_SOCKADDR_IN6_SIN6_SCOPE_ID - Success
Found *nroff option: -- -man
-- Found Perl: /usr/bin/perl (found version "5.30.3") 
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Looking for dlopen in ;
-- Looking for dlopen in ; - found
-- Looking for connect in socket;
-- Looking for connect in socket; - not found
-- Looking for gethostbyname in c
-- Looking for gethostbyname in c - found
-- Looking for gethostname
-- Looking for gethostname - found
-- Found OpenSSL: /my/home/path/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/lib/libcrypto.dylib (found version "1.1.1t")  
-- Looking for openssl/crypto.h
-- Looking for openssl/crypto.h - found
[... 1767 lines cut ...]
[ 16%] Building C object lib/CMakeFiles/libcurl.dir/vquic/quiche.c.o
[ 16%] Building C object lib/CMakeFiles/libcurl.dir/vssh/libssh2.c.o
[ 16%] Building C object lib/CMakeFiles/libcurl.dir/vssh/libssh.c.o
[ 16%] Linking C shared library libcurl.dylib
Undefined symbols for architecture arm64:
  "_EVP_PKEY_get_id", referenced from:
      _cert_stuff in openssl.c.o
      _get_cert_chain in openssl.c.o
  "_SSL_get1_peer_certificate", referenced from:
      _servercert in openssl.c.o
ld: symbol(s) not found for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[5]: *** [lib/libcurl.dylib] Error 1
make[4]: *** [lib/CMakeFiles/libcurl.dir/all] Error 2
make[3]: *** [all] Error 2
make[2]: *** [build/src/project_libcurl-stamp/project_libcurl-build] Error 2
make[1]: *** [CMakeFiles/project_libcurl.dir/all] Error 2
make: *** [all] Error 2
CMake Error at dependency/libkvscproducer/kvscproducer-src/CMake/Utilities.cmake:93 (message):
  CMake step for libcurl failed: 2
Call Stack (most recent call first):
  dependency/libkvscproducer/kvscproducer-src/CMakeLists.txt:93 (build_dependency)

-- Configuring incomplete, errors occurred!
```

hassanctech, 2023-07-05:
Looks like your ssl might be not building for arm64, to confirm this can you run:

```
file ./open-source/lib/libssl.dylib
```

Modify the path to wherever your libssl.dylib is.

On my Intel mac I see:

```
$ file ./open-source/lib/libssl.dylib
./open-source/lib/libssl.dylib: Mach-O 64-bit dynamically linked shared library x86_64
```

If yours is building correctly it should be `arm64` but I suspect it is not building the right one.  Once you confirm that is the case, we can modify/fix the build flags so that it works for you.

sirknightj, 2023-07-05:
Duplicate of #1018

sirknightj, 2023-07-05:
To reiterate: Use `develop` branch, and perform a clean build.

robert-hanuschke, 2023-07-06:
@sirknightj : tried that, didn't work, am at the situation of my last comment with that approach.

as for my libssl.dylib:

```
% file open-source/local/lib/libssl.dylib 
open-source/local/lib/libssl.dylib: Mach-O 64-bit dynamically linked shared library arm64
```

Possibly notable: I don't have a "lib" folder directly in open-source, there's only "local" in there which then contains it

elizatlawy, 2023-07-27:
Any update on this?

timetxt, 2023-07-29:
I am using MAC Pro M1 with Monterey, and here is the resolution working for me. 
1.  run this as README.md suggested: `$ brew install pkg-config openssl cmake gstreamer gst-plugins-base gst-plugins-good gst-plugins-bad gst-plugins-ugly log4cplus gst-libav`
2. run this : `cmake .. -DBUILD_GSTREAMER_PLUGIN=TRUE -DBUILD_DEPENDENCIES=OFF`
3. run this: `cmake .. -DBUILD_GSTREAMER_PLUGIN=TRUE`
4. You are good to go!

elizatlawy, 2023-07-30:
> 

Now it seems that the build is completed successfully but still when I run  'gst-launch-1.0' I get: `WARNING: erroneous pipeline: no element "kvssink"`

timetxt, 2023-07-30:
@elizatlawy   this could be related to the Gstream plugin environment variable. 

check herer: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp#run

elizatlawy, 2023-07-30:
@timetxt - It was the issue, once I exported the environment variable, everything is working, thank you!

robert-hanuschke, 2023-08-02:
Thanks @timetxt , I feel that gets me a step closer with the build now working - but `gst-inspect-1.0 kvssink` unfortunately tells me `No such element or plugin 'kvssink'`

```bash
user@machine build % cmake .. -DBUILD_GSTREAMER_PLUGIN=TRUE -DBUILD_DEPENDENCIES=OFF
-- The C compiler identification is AppleClang 14.0.0.14000029
-- The CXX compiler identification is AppleClang 14.0.0.14000029
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Library/Developer/CommandLineTools/usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Kinesis Video Cpp Producer path is /my/home/path/amazon-kinesis-video-streams-producer-sdk-cpp
-- dependencies install path is /my/home/path/amazon-kinesis-video-streams-producer-sdk-cpp/open-source
CMake Warning:
  Ignoring extra path from command line:

   "/my/home/path/amazon-kinesis-video-streams-producer-sdk-cpp/dependency/libkvscproducer/Unix Makefiles"

-- Configuring done (0.0s)
-- Generating done (0.0s)
-- Build files have been written to: /my/home/path/amazon-kinesis-video-streams-producer-sdk-cpp/dependency/libkvscproducer
[ 11%] Creating directories for 'libkvscproducer-download'
[ 22%] Performing download step (git clone) for 'libkvscproducer-download'
Cloning into 'kvscproducer-src'...
HEAD is now at a3dd1d0 Add new APIs to allow setting curl call timeouts (#366)
[ 33%] Performing update step for 'libkvscproducer-download'
[ 44%] No patch step for 'libkvscproducer-download'
[ 55%] No configure step for 'libkvscproducer-download'
[ 66%] No build step for 'libkvscproducer-download'
[ 77%] No install step for 'libkvscproducer-download'
[ 88%] No test step for 'libkvscproducer-download'
[100%] Completed 'libkvscproducer-download'
[100%] Built target libkvscproducer-download
-- Kinesis Video Producer path is /my/home/path/amazon-kinesis-video-streams-producer-sdk-cpp/dependency/libkvscproducer/kvscproducer-src
-- Found PkgConfig: /opt/homebrew/bin/pkg-config (found version "0.29.2") 
-- Checking for module 'libkvspicUtils'
--   No package 'libkvspicUtils' found
CMake Warning:
  Ignoring extra path from command line:

   "/my/home/path/amazon-kinesis-video-streams-producer-sdk-cpp/dependency/libkvscproducer/kvscproducer-src/dependency/libkvspic/Unix Makefiles"

-- Configuring done (0.1s)
-- Generating done (0.0s)
-- Build files have been written to: /my/home/path/amazon-kinesis-video-streams-producer-sdk-cpp/dependency/libkvscproducer/kvscproducer-src/dependency/libkvspic
[ 11%] Creating directories for 'libkvspic-download'
[ 22%] Performing download step (git clone) for 'libkvspic-download'
Cloning into 'kvspic-src'...
HEAD is now at 532178b Merge pull request #186 from awslabs/set-service-timeouts
[ 33%] Performing update step for 'libkvspic-download'
[ 44%] No patch step for 'libkvspic-download'
[ 55%] No configure step for 'libkvspic-download'
[ 66%] No build step for 'libkvspic-download'
[ 77%] No install step for 'libkvspic-download'
[ 88%] No test step for 'libkvspic-download'
[100%] Completed 'libkvspic-download'
[100%] Built target libkvspic-download
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Found OpenSSL: /opt/homebrew/Cellar/openssl@3/3.1.1_1/lib/libcrypto.dylib (found version "3.1.1")  
-- Checking for module 'libcurl'
--   Found libcurl, version 7.79.1
-- Bitness 64 bits
-- Found CURL: /Library/Developer/CommandLineTools/SDKs/MacOSX13.1.sdk/usr/lib/libcurl.tbd (found version "7.85.0")  
[... 23 lines cut ...]
--   No package 'libkvspicUtils' found
-- Configuring done (0.1s)
-- Generating done (0.0s)
-- Build files have been written to: /my/home/path/amazon-kinesis-video-streams-producer-sdk-cpp/dependency/libkvscproducer/kvscproducer-src/dependency/libkvspic
[ 11%] Performing update step for 'libkvspic-download'
[ 22%] No patch step for 'libkvspic-download'
[ 33%] No configure step for 'libkvspic-download'
[ 44%] No build step for 'libkvspic-download'
[ 55%] No install step for 'libkvspic-download'
[ 66%] No test step for 'libkvspic-download'
[ 77%] Completed 'libkvspic-download'
[100%] Built target libkvspic-download
-- Bitness 64 bits
found log4cplus.
-- Configuring done (1.0s)
-- Generating done (0.1s)
-- Build files have been written to: /my/home/path/amazon-kinesis-video-streams-producer-sdk-cpp/build
user@machine build % cd ..
user@machine amazon-kinesis-video-streams-producer-sdk-cpp % export GST_PLUGIN_PATH=`pwd`/build
export LD_LIBRARY_PATH=`pwd`/open-source/local/lib
user@machine amazon-kinesis-video-streams-producer-sdk-cpp % gst-inspect-1.0 kvssink
No such element or plugin 'kvssink'
user@machine amazon-kinesis-video-streams-producer-sdk-cpp % git rev-parse --abbrev-ref HEAD
develop
```

disa6302, 2023-08-16:
@robert-hanuschke ,

When building with `BUILD_DEPENDENCIES=ON`, can you try running `brew unlink openssl` and then running your cmake command again?

Would suggest cleaning up the existing build directory before trying this.

disa6302, 2023-08-22:
Closing due to no response and issue seems to have been fixed.

closed 2023-08-22 by disa6302