awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #1035: [QUESTION] How can change default device used by sample application ?

timetxt, 2023-07-29:
A one liner description about the use case and what you are trying to achieve

** Logging **
Add relevent SDK logging.  IMPORTANT NOTE:  Please make sure to NOT share AWS access credentials under any circumstance!  Please make sure they are not in the logs.

./kvs_gstreamer_audio_video_sample stream-name

and I reached this far with log output and stall here:

[DEBUG] [29-07-2023 02:03:29:256.409 UTC] getStreamingTokenHandler invoked
[DEBUG] [29-07-2023 02:03:29:256.467 UTC] Refreshing credentials. Force refreshing: 1 Now time is: 1690596209256450 Expiration: 1690599808
[INFO ] [29-07-2023 02:03:29:256.499 UTC] New credentials expiration is 1690599809
[INFO ] [29-07-2023 02:03:29:256.536 UTC] getStreamingTokenResultEvent(): Get streaming token result event.
[DEBUG] [29-07-2023 02:03:29:256.574 UTC] streamReadyHandler invoked
[DEBUG] [29-07-2023 02:03:29:256.637 UTC] Stream is ready: stream-name
[DEBUG] [29-07-2023 02:03:29:564.374 UTC] appsink configured
[INFO ] [29-07-2023 02:03:29:572.976 UTC] Using video device 0
[INFO ] [29-07-2023 02:03:29:572.995 UTC] Using audio device 0
[DEBUG] [29-07-2023 02:03:30:955.357 UTC] Pipeline playing

I knew device 0 means that sample application is trying to use my build-in Cam.  I am using a Mac Pro with M1 and the The Mac lid is closed.  But I have already modified  the source code here [line 873](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/e8125ee363c4e77e11b15398a8b77f4e8a5791a5/samples/kvs_gstreamer_audio_video_sample.cpp#L873)

gint audio_device_index = 56, video_device_index = 1; // default devices

And I have clean CMAKE and Make it again. still the application is using device 0 which is the build-in Cam

gst-device-monitor-1.0

name  : FaceTime HD Camera
	class : Video/Source
	caps  : video/x-raw(memory:GLMemory), width=640, height=480, format=UYVY, framerate=[ 1/1, 30/1 ], texture-target=rectangle
	        video/x-raw(memory:GLMemory), width=1280, height=720, format=UYVY, framerate=[ 1/1, 30/1 ], texture-target=rectangle
	        video/x-raw, width=640, height=480, format={ (string)UYVY, (string)YUY2, (string)NV12, (string)BGRA }, framerate=[ 1/1, 30/1 ]
	        video/x-raw, width=1280, height=720, format={ (string)UYVY, (string)YUY2, (string)NV12, (string)BGRA }, framerate=[ 1/1, 30/1 ]
	properties:
		device.api = avf
		avf.unique_id = EAB7A68FEC2B4487AADFD8A91C1CB782
		avf.model_id = FaceTime HD Camera
		avf.has_flash = false
		avf.has_torch = false
		avf.manufacturer = Apple Inc.
	gst-launch-1.0 avfvideosrc device-index=0 ! ...

Device found:

	name  : HD Webcam eMeet C970L
	class : Video/Source
	caps  : video/x-raw(memory:GLMemory), width=1920, height=1080, format=UYVY, framerate=[ 25/1, 5000000/83333 ], texture-target=rectangle
	        video/x-raw(memory:GLMemory), width=1280, height={ (int)960, (int)720 }, format=UYVY, framerate=[ 25/1, 5000000/83333 ], texture-target=rectangle
	        video/x-raw(memory:GLMemory), width=1024, height=576, format=UYVY, framerate=[ 25/1, 5000000/83333 ], texture-target=rectangle
	        video/x-raw(memory:GLMemory), width=800, height=600, format=UYVY, framerate=[ 25/1, 5000000/83333 ], texture-target=rectangle
	        video/x-raw(memory:GLMemory), width=640, height={ (int)480, (int)360 }, format=UYVY, framerate=[ 25/1, 5000000/83333 ], texture-target=rectangle
	        video/x-raw, width=1920, height=1080, format={ (string)UYVY, (string)YUY2, (string)NV12, (string)BGRA }, framerate=[ 25/1, 5000000/83333 ]
	        video/x-raw, width=1280, height={ (int)720, (int)960 }, format={ (string)UYVY, (string)YUY2, (string)NV12, (string)BGRA }, framerate=[ 25/1, 5000000/83333 ]
	        video/x-raw, width=1024, height=576, format={ (string)UYVY, (string)YUY2, (string)NV12, (string)BGRA }, framerate=[ 25/1, 5000000/83333 ]
	        video/x-raw, width=800, height=600, format={ (string)UYVY, (string)YUY2, (string)NV12, (string)BGRA }, framerate=[ 25/1, 5000000/83333 ]
	        video/x-raw, width=640, height={ (int)360, (int)480 }, format={ (string)UYVY, (string)YUY2, (string)NV12, (string)BGRA }, framerate=[ 25/1, 5000000/83333 ]
	properties:
		device.api = avf
		avf.unique_id = 0x120000328f0034
		avf.model_id = UVC Camera VendorID_12943 ProductID_52
		avf.has_flash = false
		avf.has_torch = false
		avf.manufacturer = EMEET
	gst-launch-1.0 avfvideosrc device-index=1 ! ...

Device found:

	name  : MacBook Pro Microphone
	class : Audio/Source
[... 89 lines cut ...]

	name  : MacBook Pro Microphone
	class : Audio/Source
	caps  : audio/x-raw, format=F32LE, layout=interleaved, rate=48000, channels=1
	        audio/x-raw, format={ (string)F64LE, (string)F64BE, (string)F32LE, (string)F32BE, (string)S32LE, (string)S32BE, (string)U32LE, (string)U32BE, (string)S24_32LE, (string)S24_32BE, (string)U24_32LE, (string)U24_32BE, (string)S24LE, (string)S24BE, (string)U24LE, (string)U24BE, (string)S20LE, (string)S20BE, (string)U20LE, (string)U20BE, (string)S18LE, (string)S18BE, (string)U18LE, (string)U18BE, (string)S16LE, (string)S16BE, (string)U16LE, (string)U16BE, (string)S8, (string)U8 }, layout=interleaved, rate=48000, channels=2, channel-mask=0x0000000000000003
	        audio/x-raw, format={ (string)F64LE, (string)F64BE, (string)F32LE, (string)F32BE, (string)S32LE, (string)S32BE, (string)U32LE, (string)U32BE, (string)S24_32LE, (string)S24_32BE, (string)U24_32LE, (string)U24_32BE, (string)S24LE, (string)S24BE, (string)U24LE, (string)U24BE, (string)S20LE, (string)S20BE, (string)U20LE, (string)U20BE, (string)S18LE, (string)S18BE, (string)U18LE, (string)U18BE, (string)S16LE, (string)S16BE, (string)U16LE, (string)U16BE, (string)S8, (string)U8 }, layout=interleaved, rate=48000, channels=1
	gst-launch-1.0 osxaudiosrc device=51 ! ...

Device found:

	name  : HD Webcam eMeet C970L
	class : Audio/Source
	caps  : audio/x-raw, format=F32LE, layout=interleaved, rate=16000, channels=1
	        audio/x-raw, format={ (string)F64LE, (string)F64BE, (string)F32LE, (string)F32BE, (string)S32LE, (string)S32BE, (string)U32LE, (string)U32BE, (string)S24_32LE, (string)S24_32BE, (string)U24_32LE, (string)U24_32BE, (string)S24LE, (string)S24BE, (string)U24LE, (string)U24BE, (string)S20LE, (string)S20BE, (string)U20LE, (string)U20BE, (string)S18LE, (string)S18BE, (string)U18LE, (string)U18BE, (string)S16LE, (string)S16BE, (string)U16LE, (string)U16BE, (string)S8, (string)U8 }, layout=interleaved, rate=16000, channels=2, channel-mask=0x0000000000000003
	        audio/x-raw, format={ (string)F64LE, (string)F64BE, (string)F32LE, (string)F32BE, (string)S32LE, (string)S32BE, (string)U32LE, (string)U32BE, (string)S24_32LE, (string)S24_32BE, (string)U24_32LE, (string)U24_32BE, (string)S24LE, (string)S24BE, (string)U24LE, (string)U24BE, (string)S20LE, (string)S20BE, (string)U20LE, (string)U20BE, (string)S18LE, (string)S18BE, (string)U18LE, (string)U18BE, (string)S16LE, (string)S16BE, (string)U16LE, (string)U16BE, (string)S8, (string)U8 }, layout=interleaved, rate=16000, channels=1
	gst-launch-1.0 osxaudiosrc device=56 ! ...

And why it is using audio device 0 while the audio index is 51 for Mac in-build Mic ?

I am looking forward to your inputs here. Thank you! 

** Any design considerations/constraints **
Explain in detail how you would like to integrate our SDK into your solution 

** If you would not like to open an issue to discuss your solution in open-platform, please email your question to kinesis-video-support@amazon.com **

hassanctech, 2023-08-03:
By any chance do you have this env var set:  `AWS_KVS_AUDIO_DEVICE`.  If this is set it will override whatever value you hard coded.  Perhaps you can try to set this env var to the value you desire and see if it works that way.  There is an analog for video, `AWS_KVS_VIDEO_DEVICE`

sirknightj, 2023-08-03:
Documentation for the item @hassanctech is referring to: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/e8125ee363c4e77e11b15398a8b77f4e8a5791a5/docs/macos.md#running-the-gstreamer-sample-application-to-stream-audio-and-video-from-live-source

timetxt, 2023-08-04:
`% cat ~/.zshrc

export AWS_DEFAULT_REGION=ap-southeast-2

export GST_PLUGIN_PATH=~/Documents/services/kvs/cpp/amazon-kinesis-video-streams-producer-sdk-cpp/build
export LD_LIBRARY_PATH=~/Documents/services/kvs/cpp/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/lib

export AWS_KVS_VIDEO_DEVICE=1
export AWS_KVS_AUDIO_DEVICE=56`

Work like a champ! 

Remember to `source ~/.zshrc` after modifying the file

Thank you!

closed 2023-08-04 by timetxt