awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #1036: [QUESTION] How can I output images in S3 bucket with kvs_gstreamer_audio_video_sample ?

timetxt, 2023-07-29:
A one liner description about the use case and what you are trying to achieve

** Logging **
Add relevent SDK logging.  IMPORTANT NOTE:  Please make sure to NOT share AWS access credentials under any circumstance!  Please make sure they are not in the logs.

In code [kvs_gstreamer_audio_video_sample my-k](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/e8125ee363c4e77e11b15398a8b77f4e8a5791a5/samples/kvs_gstreamer_audio_video_sample.cpp#L1007),

How could I use option '-e' to set gEvent value to 'Image'?  

I run command `./kvs_gstreamer_audio_video_sample <kvs-channel> -f ./<mp4-file> -e image`, but only '-f' was effective. I did not have see images were uploaded to into my S3 bucket. 

I used `aws kinesisvideo update-image-generation-configuration` to configure my S3 destination.

I also tried with  `./kvs_gstreamer_sample <kvs-channel>  ./<mp4-file>` with the same 'update-image-generation-configuration' configuration file, I was able to see image files were uploaded into S3 bucket. 

'update-image-generation-configuration' template was used from [aws blog: Building machine learning pipelines with Amazon Kinesis Video Streams](https://aws.amazon.com/blogs/iot/building-machine-learning-pipelines-with-amazon-kinesis-video-streams/)

From [source code of 'kvs_gstreamer_sample'](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/e8125ee363c4e77e11b15398a8b77f4e8a5791a5/samples/kvs_gstreamer_sample.cpp#L409), it outputs images into s3 by default if the configuration is set.

One thing w

** Any design considerations/constraints **
Explain in detail how you would like to integrate our SDK into your solution 

** If you would not like to open an issue to discuss your solution in open-platform, please email your question to kinesis-video-support@amazon.com **

sirknightj, 2023-07-30:
The `-e image` flag is working correctly. You can check for yourself by placing a print statement [here](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/e8125ee363c4e77e11b15398a8b77f4e8a5791a5/samples/kvs_gstreamer_audio_video_sample.cpp#L478C26-L478C26).

However, running the sample, I did see this line:
```
[ERROR] [30-07-2023 07:56:51:857.031 UTC] putKinesisVideoEventMetadata(): operation returned status code: 0x5200008c
[ERROR] [30-07-2023 07:56:51:857.087 UTC] Failed to insert event metadata with: 0x5200008c
```
Which is the same issue fixed in #1019, just in this sample.

We can fix this by moving the switch statement to the end of the method. Note that you may also want to adjust the [keyframe event interval](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/e8125ee363c4e77e11b15398a8b77f4e8a5791a5/samples/kvs_gstreamer_audio_video_sample.cpp#L70) to add the eventMetadata more often, since the default will only attach it to every 200th keyframe, which may be way too far apart, according to your use case.

In the patch below, I have moved `putEventMetadata` to the end of the method, and changed the interval to 5.

[putEventMetadata.patch](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/12208815/putEventMetadata.patch)

Remember to run `make` after applying this.

timetxt, 2023-07-30:
confirm the patch is working. Thank you!

closed 2023-07-30 by timetxt