awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #1037: [QUESTION]kvs_gstreamer_sample cannot stream by using built-in camera in Mac Pro M1

timetxt, 2023-07-29:
A one liner description about the use case and what you are trying to achieve

** Logging **
Add relevent SDK logging.  IMPORTANT NOTE:  Please make sure to NOT share AWS access credentials under any circumstance!  Please make sure they are not in the logs.

Issue as title.  but I can stream MP4 file with kvs_gstreamer_sample. 

In the same machine, I can use another sample application kvs_gstreamer_audio_video_sample to stream with built-in camera. 

#1035 also state another issue about switching camera selection with sample code. 

** Any design considerations/constraints **
Explain in detail how you would like to integrate our SDK into your solution 

** If you would not like to open an issue to discuss your solution in open-platform, please email your question to kinesis-video-support@amazon.com **

hassanctech, 2023-08-04:
@timetxt were you able to resolve his issue by exporting the relevant video device env var as found in this ticket:  https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/issues/1035 ?

timetxt, 2023-08-04:
no, not this one. It did not work with this particular sample application.

sirknightj, 2023-08-16:
Hi timetxt, please use either of the following GStreamer pipeline to stream from your mac webcam to kvssink.

This one uses vtenc_h264_hw, a hardware encoder.
`gst-launch-1.0 autovideosrc ! videoconvert ! video/x-raw,format=I420,width=640,height=480,framerate=30/1 ! vtenc_h264_hw allow-frame-reordering=FALSE realtime=TRUE max-keyframe-interval=45 bitrate=500 ! h264parse ! video/x-h264,stream-format=avc,alignment=au,profile=baseline ! kvssink stream-name="demo-stream"`

disa6302, 2023-08-28:
Closing due to no response.