awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #1043: [BUG] Can't connect kvs endpoint on China Beijing region with kvssink

aqiao, 2023-08-13:
**Logging**
[INFO ] [09-08-2023 08:22:55:265.400 UTC] Using region: cn-north-1
[INFO ] [09-08-2023 08:22:55:265.611 UTC] Using aws credentials for Kinesis Video Streams
[INFO ] [09-08-2023 08:22:55:265.626 UTC] No session token was detected.
[INFO ] [09-08-2023 08:22:55:269.856 UTC] createKinesisVideoClient(): Creating Kinesis Video Client
[INFO ] [09-08-2023 08:22:55:270.035 UTC] heapInitialize(): Initializing native heap with limit size 134217728, spill ratio 0% and flags 0x00000001
[INFO ] [09-08-2023 08:22:55:270.063 UTC] heapInitialize(): Creating AIV heap.
[INFO ] [09-08-2023 08:22:55:270.121 UTC] heapInitialize(): Heap is initialized OK
[DEBUG] [09-08-2023 08:22:55:270.186 UTC] getSecurityTokenHandler invoked
[DEBUG] [09-08-2023 08:22:55:270.207 UTC] Refreshing credentials. Force refreshing: 0 Now time is: 1691569375270202 Expiration: 0
[INFO ] [09-08-2023 08:22:55:270.228 UTC] New credentials expiration is 1691572975
[INFO ] [09-08-2023 08:22:55:270.310 UTC] createDeviceResultEvent(): Create device result event.
[DEBUG] [09-08-2023 08:22:55:270.393 UTC] clientReadyHandler invoked
[DEBUG] [09-08-2023 08:22:55:270.407 UTC] Client is ready
[INFO ] [09-08-2023 08:22:55:270.470 UTC] Creating Kinesis Video Stream anqing-data-stream
[INFO ] [09-08-2023 08:22:55:270.480 UTC] createKinesisVideoStream(): Creating Kinesis Video Stream.
[INFO ] [09-08-2023 08:22:55:270.489 UTC] logStreamInfo(): SDK version: e8125ee363c4e77e11b15398a8b77f4e8a5791a5
[DEBUG] [09-08-2023 08:22:55:270.498 UTC] logStreamInfo(): Kinesis Video Stream Info
[DEBUG] [09-08-2023 08:22:55:270.506 UTC] logStreamInfo(): 	Stream name: anqing-data-stream 
[DEBUG] [09-08-2023 08:22:55:270.514 UTC] logStreamInfo(): 	Streaming type: STREAMING_TYPE_REALTIME 
[DEBUG] [09-08-2023 08:22:55:270.522 UTC] logStreamInfo(): 	Content type: video/h264 
[DEBUG] [09-08-2023 08:22:55:270.530 UTC] logStreamInfo(): 	Max latency (100ns): 600000000
[DEBUG] [09-08-2023 08:22:55:270.539 UTC] logStreamInfo(): 	Fragment duration (100ns): 20000000
[DEBUG] [09-08-2023 08:22:55:270.546 UTC] logStreamInfo(): 	Key frame fragmentation: Yes
[DEBUG] [09-08-2023 08:22:55:270.554 UTC] logStreamInfo(): 	Use frame timecode: Yes
[DEBUG] [09-08-2023 08:22:55:270.562 UTC] logStreamInfo(): 	Absolute frame timecode: Yes
[DEBUG] [09-08-2023 08:22:55:270.569 UTC] logStreamInfo(): 	Nal adaptation flags: 0
[DEBUG] [09-08-2023 08:22:55:270.577 UTC] logStreamInfo(): 	Average bandwith (bps): 4194304
[DEBUG] [09-08-2023 08:22:55:270.585 UTC] logStreamInfo(): 	Framerate: 25
[DEBUG] [09-08-2023 08:22:55:270.592 UTC] logStreamInfo(): 	Buffer duration (100ns): 1200000000
[DEBUG] [09-08-2023 08:22:55:270.600 UTC] logStreamInfo(): 	Replay duration (100ns): 400000000
[DEBUG] [09-08-2023 08:22:55:270.608 UTC] logStreamInfo(): 	Connection Staleness duration (100ns): 600000000
[DEBUG] [09-08-2023 08:22:55:270.616 UTC] logStreamInfo(): 	Store Pressure Policy: 1
[DEBUG] [09-08-2023 08:22:55:270.623 UTC] logStreamInfo(): 	View Overflow Policy: 1
[DEBUG] [09-08-2023 08:22:55:270.631 UTC] logStreamInfo(): 	Segment UUID: NULL
[DEBUG] [09-08-2023 08:22:55:270.639 UTC] logStreamInfo(): 	Frame ordering mode: 0
[DEBUG] [09-08-2023 08:22:55:270.646 UTC] logStreamInfo(): Track list
[DEBUG] [09-08-2023 08:22:55:270.654 UTC] logStreamInfo(): 	Track id: 1
[DEBUG] [09-08-2023 08:22:55:270.661 UTC] logStreamInfo(): 	Track name: kinesis_video
[DEBUG] [09-08-2023 08:22:55:270.670 UTC] logStreamInfo(): 	Codec id: V_MPEG4/ISO/AVC
[DEBUG] [09-08-2023 08:22:55:270.677 UTC] logStreamInfo(): 	Track type: TRACK_INFO_TYPE_VIDEO
[DEBUG] [09-08-2023 08:22:55:270.685 UTC] logStreamInfo(): 	Track cpd: NULL
[WARN ] [09-08-2023 08:22:55:335.730 UTC] curlCompleteSync(): **curl perform failed for url https://kinesisvideo.cn-north-1.amazonaws.com/describeStream** with result Couldn't resolve host name: Could not resolve host: kinesisvideo.cn-north-1.amazonaws.com
[WARN ] [09-08-2023 08:22:55:335.791 UTC] curlCompleteSync(): HTTP Error 0 : Response: (null)
Request URL: https://kinesisvideo.cn-north-1.amazonaws.com/describeStream
Request Headers:
    Authorization: AWS4-HMAC-SHA256 Credential=test/20230809/cn-north-1/kinesisvideo/aws4_request, SignedHeaders=host;user-agent;x-amz-date, Signature=591fe1aeee6f08dbdaf05a34296e877fb80332f23ff116e6bb4c1da4e3fc8413
    content-length: 39
    content-type: application/json
    host: kinesisvideo.cn-north-1.amazonaws.com
    user-agent: AWS-SDK-KVS/3
[DEBUG] [09-08-2023 08:22:55:335.827 UTC] describeStreamCurlHandler(): DescribeStream API response: 
[INFO ] [09-08-2023 08:22:55:335.862 UTC] describeStreamResultEvent(): Describe stream result event.
[DEBUG] [09-08-2023 08:22:55:335.881 UTC] defaultStreamStateTransitionHook(): Stream state machine retry count: 0
[DEBUG] [09-08-2023 08:22:55:335.891 UTC] defaultStreamStateTransitionHook(): 

**Describe the bug**
According to AWS official doc, the kvs endpoint on Beijing region should be https://kinesisvideo.cn-north-1.amazonaws.com**.cn**
so the subfix cn lost.

**To Reproduce**
gst-launch-1.0 -v autovideosrc device=/dev/video0 \
! video/x-raw,framerate=20/1 \
! videoscale \
! videoconvert \
! x264enc tune=zerolatency bitrate=200 speed-preset=superfast \
! kvssink  name=sink stream-name="test-stream" access-key="myak" secret-key="mysk" osxaudiosrc ! audioconvert ! avenc_aac ! queue ! sink.

**Expected behavior**
KVSSink should support data stream endpoint on Beijing region:https://kinesisvideo.cn-north-1.amazonaws.com**.cn**

**SDK version number**
Include the SDK version you are running. If it is a specific commit on master branch, include that

**Open source building**
If it is a build issue, include 3rd party library version and steps to how you are building it

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Platform (please complete the following information):**
 - OS: [Mac OS ]
 - Processor: [Apple M1 Pro arm]
 - Version [Ventura 13.1]

**Additional context**
I checked the cpp producer code found bellow code:
```
    if (control_plane_uri_.empty()) {
        // Create a fully qualified URI
        control_plane_uri_ = CONTROL_PLANE_URI_PREFIX
                             + std::string(KINESIS_VIDEO_SERVICE_NAME)
                             + "."
                             + region_
                             + CONTROL_PLANE_URI_POSTFIX;
        // If region is in CN, add CN region uri postfix
        if(region_.rfind("cn-", 0) == 0) {
            control_plane_uri_ += ".cn";
        }
    }
```
Seems it considered the Beijing region case, but i don't know why kvssink does't work

aqiao, 2023-08-13:
add aws-region parameter, but seems it doesn't get the value
```
user1@localhost build % gst-launch-1.0 -v autovideosrc device=/dev/video0 \                 
! video/x-raw,framerate=20/1 \
! videoscale \
! videoconvert \
! x264enc tune=zerolatency bitrate=200 speed-preset=superfast \
! kvssink  name=sink stream-name="test-stream" access-key="testak" secret-key="testsk" aws-region="cn-north-1" osxaudiosrc ! audioconvert ! avenc_aac ! queue ! sink.
设置暂停管道 ...
[DEBUG] [13-08-2023 04:01:15:108.592 UTC] control_plane_uri_ before empty check 
[DEBUG] [13-08-2023 04:01:15:108.625 UTC] region_ before empty check us-west-2
[DEBUG] [13-08-2023 04:01:15:108.637 UTC] control_plane_uri_ after empty check https://kinesisvideo.us-west-2.amazonaws.com
[DEBUG] [13-08-2023 04:01:15:108.645 UTC] control_plane_uri_ after appending cn https://kinesisvideo.us-west-2.amazonaws.com
[ERROR] [13-08-2023 04:01:15:113.693 UTC] Unable to create Rotating Credential provider. Error status: 0x15000022
```

aqiao, 2023-08-13:
BTW, i added some debug info for control_plane_uri_ and region_ variable

sirknightj, 2023-08-17:
Hi @aqiao, thanks for the report.

---------
### Summary of reported issue:

It is trying:

https://kinesisvideo.cn-north-1.amazonaws.com

Whereas it should be:

https://kinesisvideo.cn-north-1.amazonaws.com.cn

---------

I've tried reproducing this, but it chooses the correct endpoint for me:
```sh
export AWS_DEFAULT_REGION=cn-north-1
gst-launch-1.0 -v videotestsrc is-live=true \
  ! video/x-raw,framerate=30/1,width=720,height=480 \
  ! clockoverlay time-format="%a %B %d, %Y %I:%M:%S %p" \
  ! x264enc \
  ! h264parse \
  ! kvssink stream-name=demo-stream
```
```
Request URL: https://kinesisvideo.cn-north-1.amazonaws.com.cn/describeStream
```
This is correct.

```sh
unset AWS_DEFAULT_REGION
gst-launch-1.0 -v videotestsrc is-live=true \
  ! video/x-raw,framerate=30/1,width=720,height=480 \
  ! clockoverlay time-format="%a %B %d, %Y %I:%M:%S %p" \
  ! x264enc \
  ! h264parse \
  ! kvssink stream-name=demo-stream aws-region="cn-north-1"
```
```
Request URL: https://kinesisvideo.cn-north-1.amazonaws.com.cn/describeStream
```
This is also correct.

If you could also unset CONTROL_PLANE_URI variable?
If that still doesn't work, please try pulling latest `develop` branch and performing a clean build to see if the error still persists.

aqiao, 2023-08-17:
Hi @sirknightj ,
May i know the environment you run above command?

I run above command on MacOS M1 at first, it seems gstreamer can't start the buildin camera, then i have to swtich to a Windows x64 laptop. Now i can use below command to start buildin camera 
```
gst-launch-1.0 mfvideosrc  ! queue !autovideosink
```
so i try to build `kvssink` further on the laptop, however i faced the build issue 
https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/issues/569#issuecomment-1681642702

and i saw your response, but unfortunately it doesn't work. 

Please could guide me how to build it on Windowns laptop, i have been blocked by this issue for almost one week

aqiao, 2023-08-17:
I pulled the latest code ,and build with below command
```
cmake .. -DBUILD_GSTEAMER_PLUGIN=TRUE -DBUILD_DEPENDENCIES=OFF -DCURL_INCLUDE_DIR="C:\Users\user1\Downloads\curl-8.2.1_5-win64-mingw\include" -DCURL_LIBRARY="C:\Users\user1\Downloads\curl-8.2.1_5-win64-mingw\lib"

```
it shows following error
```
CMake Error at CMakeLists.txt:119 (find_package):
  Could not find a package configuration file provided by "Log4cplus" with
  any of the following names:

    log4cplusConfig.cmake
    log4cplus-config.cmake

  Add the installation prefix of "Log4cplus" to CMAKE_PREFIX_PATH or set
  "Log4cplus_DIR" to a directory containing one of the above files.  If
  "Log4cplus" provides a separate development package or SDK, be sure it has
  been installed.

-- Configuring incomplete, errors occurred!
See also "C:/Users/shubingzhang/amazon-kinesis-video-streams-producer-sdk-cpp/build/CMakeFiles/CMakeOutput.log".
See also "C:/Users/shubingzhang/amazon-kinesis-video-streams-producer-sdk-cpp/build/CMakeFiles/CMakeError.log".

```

aqiao, 2023-08-17:
Hi @sirknightj 
Can we setup a chime meeting now?

niyatim23, 2023-08-17:
Hi @aqiao, please refer to the steps of a working Windows build in our CI:
https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/4c4afff63f3371a69be2472bb0bee654ef09abd3/.github/workflows/ci.yml#L278-L297

Be sure to modify the folder paths as per the location on your system

aqiao, 2023-08-17:
@niyatim23 ,
I have installed gstreamer (1.22.5) from official site as below
MSVC 64-bit (VS 2019, Release CRT)
[1.22.5 runtime installer](https://gstreamer.freedesktop.org/data/pkg/windows/1.22.5/msvc/gstreamer-1.0-msvc-x86_64-1.22.5.msi)
[1.22.5 development installer](https://gstreamer.freedesktop.org/data/pkg/windows/1.22.5/msvc/gstreamer-1.0-devel-msvc-x86_64-1.22.5.msi)

But in the ci.yml, i noticed it install 1.16.2
```
run: |
     choco install nasm strawberryperl
     choco install gstreamer --version=1.16.2
     choco install gstreamer-devel --version=1.16.2 
```
Can i comment above install code, or must keep the same version as 1.16.2

niyatim23, 2023-08-17:
I think the new version should be fine. Although, I don't know if the environment variables set the same way for the GUI-based and the command line approach. I guess you can try to use the GStreamer you already have and if you run into any issues, you can move to installing it this way

aqiao, 2023-08-18:
@niyatim23，
I can build with vs2022 now, but i use vs2022 build-in build method, which caused the outputs are placed in `out` folder, and i can't found the libkvssink.dll , do i need run `nmake` command further, if so, where to run

sirknightj, 2023-08-21:
Hi @aqiao, this thread is getting off-topic, as we've drifted from the original issue. I see you've created a new issue #1049. We'll keep windows-build discussion there.

Is the original issue you are encountering (URL with .cn) still occurring? https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/issues/1043#issuecomment-1681652650

aqiao, 2023-08-21:
Hi @sirknightj , this issue is still here. Actually i juse run below command and it shows same error
```
gst-launch-1.0 ksvideosrc do-timestamp=TRUE ! video/x-raw,width=640,height=480,framerate=30/1 ! videoconvert ! x264enc bframes=0 key-int-max=45 bitrate=512 ! video/x-h264,profile=baseline,stream-format=avc,alignment=au ! kvssink stream-name="stream-name" access-key="myak" secret-key="mysk" aws-region="cn-north-1"
```

It shows:
```
[DEBUG] [21-08-2023 07:53:08:798.907 中国标准时间] describeStreamCurlHandler(): DescribeStream API response:
[INFO ] [21-08-2023 07:53:08:798.907 中国标准时间] describeStreamResultEvent(): Describe stream result event.
[DEBUG] [21-08-2023 07:53:08:799.909 中国标准时间] defaultStreamStateTransitionHook(): Stream state machine retry count: 9
[DEBUG] [21-08-2023 07:53:08:799.909 中国标准时间] defaultStreamStateTransitionHook():
 KinesisVideoStream base result is [10006]. Executing KVS retry handler of retry strategy type [1]
[WARN ] [21-08-2023 07:53:18:842.526 中国标准时间] curlCompleteSync(): curl perform failed for url https://kinesisvideo.cn-north-1.amazonaws.com/describeStream with result Couldn't resolve host name: Could not resolve host: kinesisvideo.cn-north-1.amazonaws.com
```

aqiao, 2023-08-22:
@niyatim23 there is only a folder named `local` under `open-source` ,  you mean to export these `dll`, if so please guide me how and where to export this `dll` files

[image]

niyatim23, 2023-08-22:
@aqiao, please set the following and see if that works for you:
```
set GST_PLUGIN_PATH=%CD%\build
set PATH=%PATH%;%CD%\open-source\local\bin;%CD%\open-source\local\lib
```

aqiao, 2023-08-25:
exported above variables, but still doesn't work

jdelapla, 2023-09-01:
Which branch are you using @aqiao ?

jdelapla, 2023-09-01:
I've also tried to reproduce this, and am somehow missing how this simple string parser could be failing.
Could you print the result of this line in the same place the control_plane_uri_ is being calculated?
```
auto result = region_.rfind("cn-", 0);
```

aqiao, 2023-09-05:
@jdelapla , i' using 3.4.0 master branch,  `region_.rfind("cn-", 0);` run as expected, it will append `.cn`, but the final kvs endpoint still doesn't `cn`.

BTW, the `control panel` is kvs with webRTC related conception from aws doc, the kvs datastream also refers to `control panel`?  In my case, i just use kvs datatream without webRTC, i'm not sure if there is any relationship

niyatim23, 2023-09-05:
@aqiao, we are not able to reproduce your issue. When I use a kvvsink gstreamer pipeline with cn-north-1, I see the correct URL being chosen: `Request URL: https://kinesisvideo.cn-north-1.amazonaws.com.cn/describeStream`. I would recommend building clean with dependencies. 

GStreamer pipeline:
```
gst-launch-1.0 -v videotestsrc is-live=true ! video/x-raw,framerate=30/1,width=720,height=480 ! clockoverlay time-format="%a %B %d, %Y %I:%M:%S %p" ! openh264enc ! h264parse ! kvssink stream-name=demo-stream
```

sirknightj, 2023-09-09:
You can find which APIs belong to control plane vs data plane here: https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/limits.html

The control plane endpoints can be found here:
https://docs.aws.amazon.com/general/latest/gr/akv.html#limits-akv or https://docs.amazonaws.cn/en_us/aws/latest/userguide/endpoints-Beijing.html or

aqiao, 2023-09-09:
@niyatim23, will try and update you next week

niyatim23, 2023-09-18:
Closing due to no response. Feel free to reopen once you have something

closed 2023-09-18 by niyatim23