awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #1044: [QUESTION]

umerm64, 2023-08-13:
How to update kvssink parameters at run time

I want to update kvssink parameters like stream-name, access-key, secret-key for a running pipeline.
The objective is to send stream to multiple kinesis accounts based on certain events.
I am trying to use g_object_set function to do this but right now I cant seem to do it.
Need help in this regard.

Thank you.

closed 2023-08-22 by disa6302

umerm64, 2023-08-22:
@disa6302 Thank you for reaching out.
Unfortunately this doesn't solve the issue of mine.
I actually tried to pause the pipeline (changed state to PAUSED), updated kvssink parameters and then changed state to PLAY again but the params were not updated (the stream was not directed to new kinesis account)

disa6302, 2023-08-22:
As I mentioned, pausing is something kvssink does not support, so naturally, the update would not work. You would have to stop the pipeline and restart.

umerm64, 2023-08-22:
@disa6302 does this mean destroying the pipeline and recreating again?

disa6302, 2023-08-22:
It would mean setting the pipeline state to NULL and potentially stopping your main loop when you want to change the parameters