awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #1049: [BUG] Build failure on Windows

aqiao, 2023-08-20:
**Logging**

NMAKE : fatal error U1073: Dont know how to build “"apps\apps.c"”
Stop.
NMAKE : fatal error U1077: “"D:\Program Files\Microsoft Visual Studio\2022\Community\VC\Tools\MSVC\14.37.32822\bin\Hostx64\x64\nmake.exe" /                   depend && "D:\Program Files\Microsoft Visual Studio\2022\Community\VC\Tools\MSVC\14.37.32822\bin\Hostx64\x64\nmake.exe" /                   _all”: 返回代码“0x2”
Stop.
NMAKE : fatal error U1077: “echo >nul && "D:\Program Files\Microsoft Visual Studio\2022\Community\VC\Tools\MSVC\14.37.32822\bin\Hostx64\x64\nmake.exe"”: 返回代码“0x2”
Stop.
NMAKE : fatal error U1077: “"D:\Program Files\Microsoft Visual Studio\2022\Community\VC\Tools\MSVC\14.37.32822\bin\HostX64\x6Submodule path 'krb5': checked out '890ca2f401924cdcb88f493950b04700bbe52db3'

**Describe the bug**
Build failure on Window

**To Reproduce**
Steps to reproduce the behavior:
1.`call "D:\Program Files\Microsoft Visual Studio\2022\Community\VC\Auxiliary\Build\vcvars64.bat"`
2. `cmake -G "NMake Makefiles" -DBUILD_TEST=TRUE -DBUILD_GSTREAMER_PLUGIN=TRUE -DEBUILD_DEPENDENCIES=OFF -DBUILD_JNI=TRUE ..`

**Platform (please complete the following information):**
 - OS: [Windows]
 - Processor: [x64]
 - Version [10]

**Additional context**
BTW, i have installed below software followed `build_windows.bat`
```
choco install nasm strawberryperl
choco install gstreamer --version=1.22.5
choco install gstreamer-devel --version=1.22.5
```

aqiao, 2023-08-20:
Also added below info to PATH
          $env:Path += ';C:\Strawberry\perl\site\bin;C:\Strawberry\perl\bin;C:\Strawberry\c\bin;C:\Program Files\NASM;E:\test\amazon-kinesis-video-streams-producer-sdk-cpp\open-source\local\lib;E:\test\amazon-kinesis-video-streams-producer-sdk-cpp\open-source\local\bin'
          git config --system core.longpaths true

aqiao, 2023-08-20:
i installed openssl with choco command 
`choco install openssl`
above error disappeared but now it raised another error
```
  No build step for 'project_libopenssl'
  'MAKE_EXE-NOTFOUND' is not recognized as internal or external command,
  operable program or batch file。
```

niyatim23, 2023-08-21:
@aqiao, do you see the same error if you're trying to build with dependencies? Also, at what stage do you see the error you have shared?

niyatim23, 2023-09-05:
@aqiao, we recommend building with dependencies on Windows. I was able to build with dependencies and after exporting the paths correctly with the following, I could stream through a gstreamer pipeline as well:
```
set GST_PLUGIN_PATH=%CD%\build
set PATH=%PATH%;%CD%\open-source\local\bin;%CD%\open-source\local\lib
```

Gstreamer pipeline:
```
gst-launch-1.0 -v videotestsrc is-live=true ! video/x-raw,framerate=30/1,width=720,height=480 ! clockoverlay time-format="%a %B %d, %Y %I:%M:%S %p" ! openh264enc ! h264parse ! kvssink stream-name=demo-stream
```

niyatim23, 2023-09-11:
Closing assuming answered

closed 2023-09-11 by niyatim23