awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #1050: [BUG]

girma336, 2023-08-21:
**Logging**
Add relevent SDK logging.  IMPORTANT NOTE:  Please make sure to NOT share AWS access credentials under any circumstance!  Please make sure they are not in the logs.

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**SDK version number**
Include the SDK version you are running. If it is a specific commit on master branch, include that

**Open source building**
If it is a build issue, include 3rd party library version and steps to how you are building it

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Platform (please complete the following information):**
 - OS: [e.g. Linux]
 - Processor: [e.g. ARMv6]
 - Version [e.g. ubuntu 22.04]

**Additional context**
-- Checking for module 'gstreamer-app-1.0'
--   No package 'gstreamer-app-1.0' found
CMake Error at /usr/share/cmake-3.22/Modules/FindPkgConfig.cmake:603 (message):
  A required package was not found
Call Stack (most recent call first):
  /usr/share/cmake-3.22/Modules/FindPkgConfig.cmake:825 (_pkg_check_modules_internal)
  CMakeLists.txt:198 (pkg_check_modules)

-- Configuring incomplete, errors occurred!
See also "/home/girma/Downloads/amazon-kinesis-video-streams-producer-sdk-cpp/build/CMakeFiles/CMakeOutput.log".

sirknightj, 2023-08-21:
Hi @girma336,

The build for ubuntu is currently passing https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/actions/runs/5882086381/workflow#L379

Looking at the error you are running, it appears that CMake couldn't find GStreamer, or it isn't installed. 

Want to confirm that you installed all of the required dependencies.
```
sudo apt-get install cmake m4 git build-essential libssl-dev libcurl4-openssl-dev liblog4cplus-dev libgstreamer1.0-dev libgstreamer-plugins-base1.0-dev gstreamer1.0-plugins-base-apps gstreamer1.0-plugins-bad gstreamer1.0-plugins-good gstreamer1.0-plugins-ugly gstreamer1.0-tools
```

If so, could you list which commands you used to build, and also which branch you are using?

girma336, 2023-08-21:
> <awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/issues/1050/1685696290
> @github.com>

@Jeremy Gunawan
I have used the master branch and  ' mkdir build and cd build; cmake ..
-DBUILD_GSTREAMER_PLUGIN=ON -DBUILD_JNI=TRUE;' at once.

niyatim23, 2023-08-21:
@girma336, can you please try @sirknightj's suggestion with a clean build. You can do that by doing `git clean -ffxd` from the root of the repository and create the build folder again by installing the dependencies correctly

closed 2023-08-22 by disa6302