awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #1051: [BUG] No package 'libkvspicUtils' found

AbhinavSandru, 2023-08-22:
[image]

OS : Linux
Version : Ubuntu 20.04.5

Checking for module 'libkvspicUtils'
--   No package 'libkvspicUtils' found

AbhinavSandru, 2023-08-22:
[image]
That wasnt the problem actually, when I give the make command the missing module is getting built but there are actually more missing libraries that need to be built like for example "kvs_gstreamer_sample" "gstkvssink"
[image]

disa6302, 2023-08-22:
Not sure I understand. Can you clarify the exact issue? Your screenshots suggest you have the samples built and the libraries. What is the error?

AbhinavSandru, 2023-08-22:
The second screenshot is from a youtube video with all the libraries built the first screen are those that were built for me they are incomplete

AbhinavSandru, 2023-08-22:
I require the kvs_gstreamer_audio_video_sample but it is not built in mine

disa6302, 2023-08-22:
@AbhinavSandru , 

Please provide the following details:
- What version of the SDK are you building?
- Have you followed the README to install gstreamer on your system?
- What Cmake command are you running? To generate the samples, you need to run `cmake .. -DBUILD_GSTREAMER_PLUGIN=ON`.

closed 2023-08-28 by disa6302