awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #1052: Unable to Stream RTSP Feed from EZVIZ C3TN Camera to Kinesis Video Stream Using GStreamer

Mr-MeerMoazzam, 2023-08-25:
**Problem:**
I'm facing difficulties while attempting to stream a live RTSP feed from an EZVIZ C3TN camera to an Amazon Kinesis Video Stream using GStreamer. The GStreamer pipeline doesn't seem to work as expected, and I'm encountering issues related to receiving UDP packets, linking pads, and internal data stream errors.
Add relevent SDK logging. 
**Steps to Reproduce**
Run the following GStreamer pipeline command to stream the RTSP feed from an EZVIZ C3TN camera to a Kinesis Video Stream:

`AWS_ACCESS_KEY_ID=$Access_Key AWS_SECRET_ACCESS_KEY=$Secret_Key AWS_DEFAULT_REGION="us-east-1" gst-launch-1.0 rtspsrc location=rtsp://admin:Password@RTSP/H.264 ! rtph264depay ! h264parse ! kvssink stream-name=camera_148 storage-size=512
`

**Expected Behavior:**
The GStreamer pipeline should establish a stable connection to the RTSP feed from the EZVIZ C3TN camera, process the H.264 data appropriately, and successfully stream it to the specified Kinesis Video Stream without encountering errors.

**Observed Behavior:**

-> The pipeline logs display warnings indicating that UDP packets from the RTSP source were not received for a specific duration and that TCP is being used as a fallback.
-> Delayed linking failures occur between GstRTSPSrc and GstRtpH264Depay elements.
-> The pipeline encounters internal data stream errors and terminates.
**Additional Information:**
-> The EZVIZ C3TN camera's live feed has been tested successfully using other RTSP clients, indicating that the camera is operational.
-> I've reviewed the firewall settings and there are no indications of the firewall blocking UDP packets.
-> The GStreamer plugins for RTSP and H.264 processing are up-to-date.
**Environment:**

Operating System: Ubuntu (20.04)
GStreamer Version: (1.16.3)

**Logs**
```
[INFO ] [25-08-2023 10:25:40:142.198 GMT] createKinesisVideoClient(): Creating Kinesis Video Client
[INFO ] [25-08-2023 10:25:40:142.231 GMT] heapInitialize(): Initializing native heap with limit size 536870912, spill ratio 0% and flags 0x00000001
[INFO ] [25-08-2023 10:25:40:142.238 GMT] heapInitialize(): Creating AIV heap.
[INFO ] [25-08-2023 10:25:40:142.249 GMT] heapInitialize(): Heap is initialized OK
[DEBUG] [25-08-2023 10:25:40:142.309 GMT] getSecurityTokenHandler invoked
[DEBUG] [25-08-2023 10:25:40:142.318 GMT] Refreshing credentials. Force refreshing: 0 Now time is: 1692959140142317049 Expiration: 0
[INFO ] [25-08-2023 10:25:40:142.328 GMT] createDeviceResultEvent(): Create device result event.
[DEBUG] [25-08-2023 10:25:40:142.336 GMT] clientReadyHandler invoked
[INFO ] [25-08-2023 10:25:40:142.353 GMT] try creating stream
[INFO ] [25-08-2023 10:25:40:142.367 GMT] Creating Kinesis Video Stream camera_148
[INFO ] [25-08-2023 10:25:40:142.375 GMT] createKinesisVideoStream(): Creating Kinesis Video Stream.
[INFO ] [25-08-2023 10:25:40:142.380 GMT] logStreamInfo(): SDK version: e8125ee363c4e77e11b15398a8b77f4e8a5791a5
[DEBUG] [25-08-2023 10:25:40:142.384 GMT] logStreamInfo(): Kinesis Video Stream Info
[DEBUG] [25-08-2023 10:25:40:142.388 GMT] logStreamInfo(): 	Stream name: camera_148 
[DEBUG] [25-08-2023 10:25:40:142.392 GMT] logStreamInfo(): 	Streaming type: STREAMING_TYPE_REALTIME 
[DEBUG] [25-08-2023 10:25:40:142.396 GMT] logStreamInfo(): 	Content type: video/h264 
[DEBUG] [25-08-2023 10:25:40:142.400 GMT] logStreamInfo(): 	Max latency (100ns): 600000000
[DEBUG] [25-08-2023 10:25:40:142.404 GMT] logStreamInfo(): 	Fragment duration (100ns): 20000000
[DEBUG] [25-08-2023 10:25:40:142.409 GMT] logStreamInfo(): 	Key frame fragmentation: Yes
[DEBUG] [25-08-2023 10:25:40:142.413 GMT] logStreamInfo(): 	Use frame timecode: Yes
[DEBUG] [25-08-2023 10:25:40:142.417 GMT] logStreamInfo(): 	Absolute frame timecode: Yes
[DEBUG] [25-08-2023 10:25:40:142.422 GMT] logStreamInfo(): 	Nal adaptation flags: 0
[DEBUG] [25-08-2023 10:25:40:142.426 GMT] logStreamInfo(): 	Average bandwith (bps): 4194304
[DEBUG] [25-08-2023 10:25:40:142.431 GMT] logStreamInfo(): 	Framerate: 25
[DEBUG] [25-08-2023 10:25:40:142.435 GMT] logStreamInfo(): 	Buffer duration (100ns): 1200000000
[DEBUG] [25-08-2023 10:25:40:142.440 GMT] logStreamInfo(): 	Replay duration (100ns): 400000000
[DEBUG] [25-08-2023 10:25:40:142.444 GMT] logStreamInfo(): 	Connection Staleness duration (100ns): 600000000
[DEBUG] [25-08-2023 10:25:40:142.447 GMT] logStreamInfo(): 	Store Pressure Policy: 1
[DEBUG] [25-08-2023 10:25:40:142.451 GMT] logStreamInfo(): 	View Overflow Policy: 1
[DEBUG] [25-08-2023 10:25:40:142.455 GMT] logStreamInfo(): 	Segment UUID: NULL
[DEBUG] [25-08-2023 10:25:40:142.459 GMT] logStreamInfo(): 	Frame ordering mode: 0
[DEBUG] [25-08-2023 10:25:40:142.463 GMT] logStreamInfo(): Track list
[DEBUG] [25-08-2023 10:25:40:142.467 GMT] logStreamInfo(): 	Track id: 1
[DEBUG] [25-08-2023 10:25:40:142.471 GMT] logStreamInfo(): 	Track name: kinesis_video
[DEBUG] [25-08-2023 10:25:40:142.475 GMT] logStreamInfo(): 	Codec id: V_MPEG4/ISO/AVC
[DEBUG] [25-08-2023 10:25:40:142.493 GMT] logStreamInfo(): 	Track type: TRACK_INFO_TYPE_VIDEO
[DEBUG] [25-08-2023 10:25:40:142.499 GMT] logStreamInfo(): 	Track cpd: NULL
[INFO ] [25-08-2023 10:25:41:221.875 GMT] writeHeaderCallback(): RequestId: 5bf951a4-9fe6-44de-bb01-67bcaa42b9ae
[DEBUG] [25-08-2023 10:25:41:221.926 GMT] describeStreamCurlHandler(): DescribeStream API response: {"StreamInfo":{"CreationTime":1.692946442192E9,"DataRetentionInHours":8760,"DeviceName":null,"IngestionConfiguration":null,"KmsKeyId":"arn:aws:kms:us-east-1:820417679840:alias/aws/kinesisvideo","MediaType":null,"Status":"ACTIVE","StreamARN":"arn:aws:kinesisvideo:us-east-1:820417679840:stream/camera_148/1692946442192","StreamName":"camera_148","Version":"9RbiPgzmA3cyWwSF8iQQ"}}
[INFO ] [25-08-2023 10:25:41:222.451 GMT] describeStreamResultEvent(): Describe stream result event.
[WARN ] [25-08-2023 10:25:41:222.474 GMT] describeStreamResult(): Retention period returned from the DescribeStream call doesn't match the one specified in the StreamInfo
[WARN ] [25-08-2023 10:25:41:222.504 GMT] describeStreamResult(): Content type returned from the DescribeStream call doesn't match the one specified in the StreamInfo
[... 17 lines cut ...]
Progress: (request) Sending PLAY request
Progress: (request) Sending PLAY request
Progress: (request) Sent PLAY request
WARNING: from element /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0: Could not read from resource.
Additional debug info:
gstrtspsrc.c(5769): gst_rtspsrc_reconnect (): /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0:
Could not receive any UDP packets for 5.0000 seconds, maybe your firewall is blocking it. Retrying using a tcp connection.
WARNING: from element /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0: Delayed linking failed.
Additional debug info:
./grammar.y(506): gst_parse_no_more_pads (): /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0:
failed delayed linking some pad of GstRTSPSrc named rtspsrc0 to some pad of GstRtpH264Depay named rtph264depay0
ERROR: from element /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0: Internal data stream error.
Additional debug info:
gstrtspsrc.c(6057): gst_rtspsrc_loop (): /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0:
streaming stopped, reason not-linked (-1)
Execution ended after 0:00:07.124925275
Setting pipeline to NULL ...
Freeing pipeline ...
[INFO ] [25-08-2023 10:25:49:363.094 GMT] Freeing Kinesis Video Stream camera_148
[INFO ] [25-08-2023 10:25:49:363.148 GMT] freeKinesisVideoStream(): Freeing Kinesis Video stream.
[DEBUG] [25-08-2023 10:25:49:363.179 GMT] curlApiCallbacksShutdownActiveRequests(): pActiveRequests hashtable is empty
[INFO ] [25-08-2023 10:25:49:763.416 GMT] freeKinesisVideoClient(): Freeing Kinesis Video Client
[DEBUG] [25-08-2023 10:25:49:763.456 GMT] curlApiCallbacksShutdownActiveRequests(): pActiveRequests hashtable is empty
[DEBUG] [25-08-2023 10:25:50:163.695 GMT] freeKinesisVideoClientInternal(): Total allocated memory 0
```

Mr-MeerMoazzam, 2023-08-25:
with GST_DEBUG=4 log is

```
Setting pipeline to PAUSED ...
[INFO ] [25-08-2023 10:25:40:142.198 GMT] createKinesisVideoClient(): Creating Kinesis Video Client
[INFO ] [25-08-2023 10:25:40:142.231 GMT] heapInitialize(): Initializing native heap with limit size 536870912, spill ratio 0% and flags 0x00000001
[INFO ] [25-08-2023 10:25:40:142.238 GMT] heapInitialize(): Creating AIV heap.
[INFO ] [25-08-2023 10:25:40:142.249 GMT] heapInitialize(): Heap is initialized OK
[DEBUG] [25-08-2023 10:25:40:142.309 GMT] getSecurityTokenHandler invoked
[DEBUG] [25-08-2023 10:25:40:142.318 GMT] Refreshing credentials. Force refreshing: 0 Now time is: 1692959140142317049 Expiration: 0
[INFO ] [25-08-2023 10:25:40:142.328 GMT] createDeviceResultEvent(): Create device result event.
[DEBUG] [25-08-2023 10:25:40:142.336 GMT] clientReadyHandler invoked
[INFO ] [25-08-2023 10:25:40:142.353 GMT] try creating stream
[INFO ] [25-08-2023 10:25:40:142.367 GMT] Creating Kinesis Video Stream camera_148
[INFO ] [25-08-2023 10:25:40:142.375 GMT] createKinesisVideoStream(): Creating Kinesis Video Stream.
[INFO ] [25-08-2023 10:25:40:142.380 GMT] logStreamInfo(): SDK version: e8125ee363c4e77e11b15398a8b77f4e8a5791a5
[DEBUG] [25-08-2023 10:25:40:142.384 GMT] logStreamInfo(): Kinesis Video Stream Info
[DEBUG] [25-08-2023 10:25:40:142.388 GMT] logStreamInfo(): 	Stream name: camera_148 
[DEBUG] [25-08-2023 10:25:40:142.392 GMT] logStreamInfo(): 	Streaming type: STREAMING_TYPE_REALTIME 
[DEBUG] [25-08-2023 10:25:40:142.396 GMT] logStreamInfo(): 	Content type: video/h264 
[DEBUG] [25-08-2023 10:25:40:142.400 GMT] logStreamInfo(): 	Max latency (100ns): 600000000
[DEBUG] [25-08-2023 10:25:40:142.404 GMT] logStreamInfo(): 	Fragment duration (100ns): 20000000
[DEBUG] [25-08-2023 10:25:40:142.409 GMT] logStreamInfo(): 	Key frame fragmentation: Yes
[DEBUG] [25-08-2023 10:25:40:142.413 GMT] logStreamInfo(): 	Use frame timecode: Yes
[DEBUG] [25-08-2023 10:25:40:142.417 GMT] logStreamInfo(): 	Absolute frame timecode: Yes
[DEBUG] [25-08-2023 10:25:40:142.422 GMT] logStreamInfo(): 	Nal adaptation flags: 0
[DEBUG] [25-08-2023 10:25:40:142.426 GMT] logStreamInfo(): 	Average bandwith (bps): 4194304
[DEBUG] [25-08-2023 10:25:40:142.431 GMT] logStreamInfo(): 	Framerate: 25
[DEBUG] [25-08-2023 10:25:40:142.435 GMT] logStreamInfo(): 	Buffer duration (100ns): 1200000000
[DEBUG] [25-08-2023 10:25:40:142.440 GMT] logStreamInfo(): 	Replay duration (100ns): 400000000
[DEBUG] [25-08-2023 10:25:40:142.444 GMT] logStreamInfo(): 	Connection Staleness duration (100ns): 600000000
[DEBUG] [25-08-2023 10:25:40:142.447 GMT] logStreamInfo(): 	Store Pressure Policy: 1
[DEBUG] [25-08-2023 10:25:40:142.451 GMT] logStreamInfo(): 	View Overflow Policy: 1
[DEBUG] [25-08-2023 10:25:40:142.455 GMT] logStreamInfo(): 	Segment UUID: NULL
[DEBUG] [25-08-2023 10:25:40:142.459 GMT] logStreamInfo(): 	Frame ordering mode: 0
[DEBUG] [25-08-2023 10:25:40:142.463 GMT] logStreamInfo(): Track list
[DEBUG] [25-08-2023 10:25:40:142.467 GMT] logStreamInfo(): 	Track id: 1
[DEBUG] [25-08-2023 10:25:40:142.471 GMT] logStreamInfo(): 	Track name: kinesis_video
[DEBUG] [25-08-2023 10:25:40:142.475 GMT] logStreamInfo(): 	Codec id: V_MPEG4/ISO/AVC
[DEBUG] [25-08-2023 10:25:40:142.493 GMT] logStreamInfo(): 	Track type: TRACK_INFO_TYPE_VIDEO
[DEBUG] [25-08-2023 10:25:40:142.499 GMT] logStreamInfo(): 	Track cpd: NULL
[INFO ] [25-08-2023 10:25:41:221.875 GMT] writeHeaderCallback(): RequestId: 5bf951a4-9fe6-44de-bb01-67bcaa42b9ae
[DEBUG] [25-08-2023 10:25:41:221.926 GMT] describeStreamCurlHandler(): DescribeStream API response: {"StreamInfo":{"CreationTime":1.692946442192E9,"DataRetentionInHours":8760,"DeviceName":null,"IngestionConfiguration":null,"KmsKeyId":"arn:aws:kms:us-east-1:820417679840:alias/aws/kinesisvideo","MediaType":null,"Status":"ACTIVE","StreamARN":"arn:aws:kinesisvideo:us-east-1:820417679840:stream/camera_148/1692946442192","StreamName":"camera_148","Version":"9RbiPgzmA3cyWwSF8iQQ"}}
[INFO ] [25-08-2023 10:25:41:222.451 GMT] describeStreamResultEvent(): Describe stream result event.
[WARN ] [25-08-2023 10:25:41:222.474 GMT] describeStreamResult(): Retention period returned from the DescribeStream call doesn't match the one specified in the StreamInfo
[WARN ] [25-08-2023 10:25:41:222.504 GMT] describeStreamResult(): Content type returned from the DescribeStream call doesn't match the one specified in the StreamInfo
[INFO ] [25-08-2023 10:25:42:164.694 GMT] writeHeaderCallback(): RequestId: 9db5822c-c4b2-4f6d-8ccf-ea8f5cdbf850
[DEBUG] [25-08-2023 10:25:42:164.725 GMT] getStreamingEndpointCurlHandler(): GetStreamingEndpoint API response: {"DataEndpoint":"https://s-d8eb3b21.kinesisvideo.us-east-1.amazonaws.com"}
[INFO ] [25-08-2023 10:25:42:165.023 GMT] getStreamingEndpointResultEvent(): Get streaming endpoint result event.
[DEBUG] [25-08-2023 10:25:42:165.041 GMT] getStreamingTokenHandler invoked
[DEBUG] [25-08-2023 10:25:42:165.050 GMT] Refreshing credentials. Force refreshing: 1 Now time is: 1692959142165048381 Expiration: 18446744073709551615
[INFO ] [25-08-2023 10:25:42:165.061 GMT] getStreamingTokenResultEvent(): Get streaming token result event.
[DEBUG] [25-08-2023 10:25:42:165.070 GMT] streamReadyHandler invoked
Stream is ready
Pipeline is live and does not need PREROLL ...
Progress: (open) Opening Stream
Progress: (connect) Connecting to rtsp://admin:HPAIWD@203.175.70.148:554/H.264
Progress: (open) Retrieving server options
Progress: (open) Retrieving media info
[... cut ...]
0:00:40.443964713 230119 0x557ab5d78000 INFO         GST_REFCOUNTING gstelement.c:3341:gst_element_finalize:<pipeline0> 0x557ab6006160 finalize
0:00:40.443970300 230119 0x557ab5d78000 INFO         GST_REFCOUNTING gstelement.c:3346:gst_element_finalize:<pipeline0> 0x557ab6006160 finalize parent
0:00:40.443976446 230119 0x557ab5d78000 INFO                GST_INIT gst.c:1083:gst_deinit: deinitializing GStreamer
0:00:40.445600493 230119 0x557ab5d78000 INFO                GST_INIT gst.c:1232:gst_deinit: deinitialized GStreamer
```

Mr-MeerMoazzam, 2023-08-25:
[gst_debug.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/12438542/gst_debug.txt)

sirknightj, 2023-08-25:
[gst_debug.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/12440520/gst_debug.txt)

sirknightj, 2023-08-25:
Edited to remove credentials. Please remember to remove credentials when uploading logs. You may also want to redact those credentials and generate new ones.

disa6302, 2023-08-25:
@Mr-MeerMoazzam ,

Have you tested your pipeline with fakesink or other sinks to confirm there isnt an issue with your source pipeline? The issue you are seeing is not related to kvssink since the frames are not getting to kvssink in the first place. 

On the other hand, you could trying using this property here: https://gstreamer.freedesktop.org/documentation/rtsp/rtspsrc.html?gi-language=c#rtspsrc:protocols to see if it helps your situation.

You could also use the sample pipeline here: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/docs/linux.md#running-the-gst-launch-10-command-to-start-streaming-from-a-rtsp-camera-source and replace your RTSP URL to see if it helps the issue.

Resolving since this is not kvssink specific. Feel free to open a new issue or start a discussion thread if you still face issues after trying out the suggestions.

closed 2023-08-25 by disa6302