awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #1064: Failed to initialize kinesis video with an exception: Unable to create Iot Credential provider

ajaypsiborg, 2023-09-14:
INFO ] [14-09-2023 13:12:54:928.659 GMT] Using region: ap-south-1
[INFO ] [14-09-2023 13:12:54:928.910 GMT] Using IoT credentials for Kinesis Video Streams
[DEBUG] [14-09-2023 13:12:54:932.110 GMT] Creating IoT auth callbacks.
2023-09-14 13:12:55.081 ERROR   blockingCurlCall(): Curl call response failed with http status 403
[ERROR] [14-09-2023 13:12:55:081.844 GMT] Unable to create Iot Credential provider. Error status: 0x16000001
[ERROR] [14-09-2023 13:12:55:082.132 GMT] Failed to initialize kinesis video with an exception: Unable to create Iot Credential provider. Error status: 0x16000001

sirknightj, 2023-09-14:
Hi @ajaypsiborg, please help us by providing the following:

1. What GStreamer command are you using? (with credentials redacted, if any)
3. What does your policy look like?
4. Are you able to fetch the credentials manually by making the curl call, then using the temporary credentials obtained to make a describe call?  Step 4, points 2 and 3 https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/how-iot.html#how-iot-test-it,

ajaypsiborg, 2023-09-15:
>    2. What does your policy look like?
>    3. Are you able to fetch the credentials manually by making the curl
> <awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/issues/1064/1719610202
> @github.com>

Please check the below commands and i want to stream from  raspberry pi
camera

  export GST_PLUGIN_PATH=`pwd`/build
  export LD_LIBRARY_PATH=`pwd`/open-source/local/lib
   gst-inspect-1.0 kvssink
export CERT_PATH=cert/certificate.pem.crt
export PRIVATE_KEY_PATH=cert/private.pem.key
export CA_CERT_PATH=cert/AmazonRootCA.pem
aws iot describe-endpoint --endpoint-type iot:CredentialProvider=>This one
is used for getting end point from cloud shell
export ROLE_ALIAS=cameraIotRoleAlice
export IOT_GET_CREDENTIAL_ENDPOINT=
c1zs83tjsf8zps.credentials.iot.ap-south-1.amazonaws.com
  ./kvs_gstreamer_sample camera_102

Policy
{
    "Version": "2012-10-17",
    "Statement": [
        {
            "Effect": "Allow",
            "Action": [
                "kinesisvideo:DescribeStream",
                "kinesisvideo:PutMedia",
                "kinesisvideo:TagStream",
                "kinesisvideo:GetDataEndpoint"
            ],
            "Resource":
"arn:aws:kinesisvideo:*:*:stream/${credentials-iot:ThingName}/*"
        }
    ]
}

Trust policy
{
    "Version": "2012-10-17",
    "Statement": [
        {
            "Effect": "Allow",
            "Principal": {
                "Service": "credentials.iot.amazonaws.com"
            },
            "Action": "sts:AssumeRole"
        }
    ]
}

ajaypsiborg, 2023-09-15:
gst-launch-1.0 -e raspivid preview=0 rotation=0 width=640 height=480 framerate=30 ! video/x-h264,format=avc,alignment=au ! h264parse ! kvssink stream-name="kvs_example_camera_stream" aws-region="ap-south-1" iot-certificate="iot-certificate,endpoint=[credential-account-specific-prefix.credentials.iot.aws-region.amazonaws.com](http://credential-account-specific-prefix.credentials.iot.aws-region.amazonaws.com/),cert-path=./certs/certificate.pem,key-path=./certs/private.pem.key,ca-path=./certs/cacert.pem,role-aliases=KvsCameraIoTRoleAlias"

I have also applied the second method with the provided documentation  but got this error
https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/how-iot.html#how-iot-test-it

WARNING: erroneous pipeline: no element "raspivid"

ajaypsiborg, 2023-09-15:
I have also applied the second method with the provided documentation  but
got this error
https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/how-iot.html#how-iot-test-it

ajaypsiborg, 2023-09-15:
It start uploading live video streaming with udb web cam but not with resperry pi camera 

Getting error in kinisis
StartTimestamp must be before the current time for LIVE_REPLAY PlaybackMode.
Status code: 400
Request ID: cb574964-c928-4e9a-8241-f8fbab32b672
Region: ap-south-1
Time: September 15, 2023 6:13:36 PM

ajaypsiborg, 2023-09-15:
https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/issues/1064

niyatim23, 2023-09-17:
Hi @ajaypsiborg, the error code you're seeing is `0x16000001: STATUS_CURL_PERFORM_FAILED`. I think the iot credentials are not configured on your device correctly on your raspberry pi. Please make sure the curl call @sirknightj mentioned works before you try to send media with KVS