awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #1068: [BUG] kvssink build on Alpine linux fails with segfault

alan-willard-simplisafe, 2023-09-21:
**Logging**
##### Output when running `gst-launch-1.0` with the `kvssink`
>/opt/amazon-kinesis-video-streams-producer-sdk-cpp/build # gst-launch-1.0 -v videotestsrc ! x264enc ! kvssink stream-name=test aws-region=us-east-1 access-key=test secret-key=test
Setting pipeline to PAUSED ...
[INFO ] [21-09-2023 15:42:31:208.267 UTC] createKinesisVideoClient(): Creating Kinesis Video Client
[INFO ] [21-09-2023 15:42:31:208.363 UTC] heapInitialize(): Initializing native heap with limit size 134217728, spill ratio 0% and flags 0x00000001
[INFO ] [21-09-2023 15:42:31:208.383 UTC] heapInitialize(): Creating AIV heap.
[INFO ] [21-09-2023 15:42:31:208.414 UTC] heapInitialize(): Heap is initialized OK
[DEBUG] [21-09-2023 15:42:31:208.565 UTC] getSecurityTokenHandler invoked
[DEBUG] [21-09-2023 15:42:31:208.602 UTC] Refreshing credentials. Force refreshing: 0 Now time is: 1695310951208599221 Expiration: 0
[INFO ] [21-09-2023 15:42:31:208.632 UTC] createDeviceResultEvent(): Create device result event.
[DEBUG] [21-09-2023 15:42:31:208.645 UTC] clientReadyHandler invoked
[INFO ] [21-09-2023 15:42:31:208.718 UTC] try creating stream
[INFO ] [21-09-2023 15:42:31:208.761 UTC] Creating Kinesis Video Stream test
[INFO ] [21-09-2023 15:42:31:208.791 UTC] createKinesisVideoStream(): Creating Kinesis Video Stream.
[INFO ] [21-09-2023 15:42:31:208.841 UTC] logStreamInfo(): SDK version: 0cfcd36648ced65a1cf740bcd9f46034ef8fff50
[DEBUG] [21-09-2023 15:42:31:208.859 UTC] logStreamInfo(): Kinesis Video Stream Info
[DEBUG] [21-09-2023 15:42:31:208.868 UTC] logStreamInfo(): 	Stream name: test
[DEBUG] [21-09-2023 15:42:31:208.876 UTC] logStreamInfo(): 	Streaming type: STREAMING_TYPE_REALTIME
[DEBUG] [21-09-2023 15:42:31:208.900 UTC] logStreamInfo(): 	Content type: video/h264
[DEBUG] [21-09-2023 15:42:31:208.909 UTC] logStreamInfo(): 	Max latency (100ns): 600000000
[DEBUG] [21-09-2023 15:42:31:208.920 UTC] logStreamInfo(): 	Fragment duration (100ns): 20000000
[DEBUG] [21-09-2023 15:42:31:208.935 UTC] logStreamInfo(): 	Key frame fragmentation: Yes
[DEBUG] [21-09-2023 15:42:31:208.955 UTC] logStreamInfo(): 	Use frame timecode: Yes
[DEBUG] [21-09-2023 15:42:31:208.977 UTC] logStreamInfo(): 	Absolute frame timecode: Yes
[DEBUG] [21-09-2023 15:42:31:208.996 UTC] logStreamInfo(): 	Nal adaptation flags: 0
[DEBUG] [21-09-2023 15:42:31:209.005 UTC] logStreamInfo(): 	Average bandwith (bps): 4194304
[DEBUG] [21-09-2023 15:42:31:209.027 UTC] logStreamInfo(): 	Framerate: 25
[DEBUG] [21-09-2023 15:42:31:209.047 UTC] logStreamInfo(): 	Buffer duration (100ns): 1200000000
[DEBUG] [21-09-2023 15:42:31:209.055 UTC] logStreamInfo(): 	Replay duration (100ns): 400000000
[DEBUG] [21-09-2023 15:42:31:209.088 UTC] logStreamInfo(): 	Connection Staleness duration (100ns): 600000000
[DEBUG] [21-09-2023 15:42:31:209.100 UTC] logStreamInfo(): 	Store Pressure Policy: 1
[DEBUG] [21-09-2023 15:42:31:209.111 UTC] logStreamInfo(): 	View Overflow Policy: 1
[DEBUG] [21-09-2023 15:42:31:209.121 UTC] logStreamInfo(): 	Segment UUID: NULL
[DEBUG] [21-09-2023 15:42:31:209.136 UTC] logStreamInfo(): 	Frame ordering mode: 0
[DEBUG] [21-09-2023 15:42:31:209.152 UTC] logStreamInfo(): Track list
[DEBUG] [21-09-2023 15:42:31:209.248 UTC] logStreamInfo(): 	Track id: 1
[DEBUG] [21-09-2023 15:42:31:209.340 UTC] logStreamInfo(): 	Track name: kinesis_video
[DEBUG] [21-09-2023 15:42:31:209.351 UTC] logStreamInfo(): 	Codec id: V_MPEG4/ISO/AVC
[DEBUG] [21-09-2023 15:42:31:209.371 UTC] logStreamInfo(): 	Track type: TRACK_INFO_TYPE_VIDEO
[DEBUG] [21-09-2023 15:42:31:209.392 UTC] logStreamInfo(): 	Track cpd: NULL
Segmentation fault

##### GDB backtrace
>Starting program: /usr/bin/gst-launch-1.0 -v videotestsrc \! x264enc \! kvssink stream-name=test aws-region=us-east-1 access-key=test secret-key=test
warning: Error disabling address space randomization: Operation not permitted
Setting pipeline to PAUSED ...
[INFO ] [21-09-2023 15:53:26:594.387 UTC] createKinesisVideoClient(): Creating Kinesis Video Client
[INFO ] [21-09-2023 15:53:26:594.461 UTC] heapInitialize(): Initializing native heap with limit size 134217728, spill ratio 0% and flags 0x00000001
[INFO ] [21-09-2023 15:53:26:594.473 UTC] heapInitialize(): Creating AIV heap.
[INFO ] [21-09-2023 15:53:26:594.492 UTC] heapInitialize(): Heap is initialized OK
[New LWP 113]
[DEBUG] [21-09-2023 15:53:26:594.684 UTC] getSecurityTokenHandler invoked
[DEBUG] [21-09-2023 15:53:26:594.716 UTC] Refreshing credentials. Force refreshing: 0 Now time is: 1695311606594714427 Expiration: 0
[INFO ] [21-09-2023 15:53:26:594.728 UTC] createDeviceResultEvent(): Create device result event.
[DEBUG] [21-09-2023 15:53:26:594.786 UTC] clientReadyHandler invoked
[INFO ] [21-09-2023 15:53:26:594.855 UTC] try creating stream
[INFO ] [21-09-2023 15:53:26:594.883 UTC] Creating Kinesis Video Stream test
[INFO ] [21-09-2023 15:53:26:594.898 UTC] createKinesisVideoStream(): Creating Kinesis Video Stream.
[INFO ] [21-09-2023 15:53:26:594.907 UTC] logStreamInfo(): SDK version: 0cfcd36648ced65a1cf740bcd9f46034ef8fff50
[DEBUG] [21-09-2023 15:53:26:594.916 UTC] logStreamInfo(): Kinesis Video Stream Info
[DEBUG] [21-09-2023 15:53:26:594.930 UTC] logStreamInfo(): 	Stream name: test
[DEBUG] [21-09-2023 15:53:26:594.936 UTC] logStreamInfo(): 	Streaming type: STREAMING_TYPE_REALTIME
[DEBUG] [21-09-2023 15:53:26:594.941 UTC] logStreamInfo(): 	Content type: video/h264
[DEBUG] [21-09-2023 15:53:26:594.948 UTC] logStreamInfo(): 	Max latency (100ns): 600000000
[DEBUG] [21-09-2023 15:53:26:594.958 UTC] logStreamInfo(): 	Fragment duration (100ns): 20000000
[DEBUG] [21-09-2023 15:53:26:594.970 UTC] logStreamInfo(): 	Key frame fragmentation: Yes
[DEBUG] [21-09-2023 15:53:26:594.986 UTC] logStreamInfo(): 	Use frame timecode: Yes
[DEBUG] [21-09-2023 15:53:26:594.993 UTC] logStreamInfo(): 	Absolute frame timecode: Yes
[DEBUG] [21-09-2023 15:53:26:594.998 UTC] logStreamInfo(): 	Nal adaptation flags: 0
[... 133 lines cut ...]
**Open source building**
#### Dockerfile

>FROM alpine:3.17
>
>RUN apk add alpine-sdk cmake autoconf gstreamer-dev pkgconf gst-plugins-base-dev gstreamer gst-plugins-bad gst-plugins-good gst-plugins-base gst-plugins-ugly openssl gst-libav
>
>COPY . /opt/amazon-kinesis-video-streams-producer-sdk-cpp
>WORKDIR /opt/amazon-kinesis-video-streams-producer-sdk-cpp/build/
>
>RUN cmake .. -DBUILD_GSTREAMER_PLUGIN=ON -DCMAKE_VERBOSE_MAKEFILE:BOOL=ON && \
    make
>
>ENV LD_LIBRARY_PATH=/opt/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/lib
>ENV GST_PLUGIN_PATH=/opt/amazon-kinesis-video-streams-producer-sdk-cpp/build/:$GST_PLUGIN_PATH

**Screenshots**
n/a

**Platform (please complete the following information):**
 - OS: MacOS Ventura Version 13.5.1 
 - Processor: arm64 (Apple M1)

**Additional context**
n/a

alan-willard-simplisafe, 2023-09-21:
I do have a core dump available, but it is quite large, let me know the preferred way I can make that available, thanks.

jdelapla, 2023-09-21:
Hello, could you please add 
```
-DCMAKE_BUILD_TYPE=Debug
```
To your CMake command so we could get line numbers in the gdb backtrace? additionally please use 
```
bt -full
```
So we can see variables with the added debug symbols, thank you.

alan-willard-simplisafe, 2023-09-22:
Additional Information

#### bt -full
```
(gdb) bt -full
#0  0x0000ffff8c7aabdc in curlCompleteSyncPython Exception <class 'gdb.MemoryError'>: Cannot access memory at address 0xffff83f5d678
 (pCurlResponse=#1  0x0000ffff8c7a2db0 in curlCallApi (pCurlRequest=0xffff83fda3a0) at /opt/amazon-kinesis-video-streams-producer-sdk-cpp/dependency/libkvscproducer/kvscproducer-src/src/source/CurlApiCallbacks.c:650
        __status = 2349079808
        retStatus = 0
        pCurlResponse = 0xffff8cde22a0
        __FUNCTION__ = "curlCallApi"
#2  0x0000ffff8c7a4eb4 in describeStreamCurlHandler (arg=0xffff83fda3a0) at /opt/amazon-kinesis-video-streams-producer-sdk-cpp/dependency/libkvscproducer/kvscproducer-src/src/source/CurlApiCallbacks.c:1339
        __status = 0
        retStatus = 0
        pCurlRequest = 0xffff83fda3a0
        pCurlApiCallbacks = 0xffff8c0512e0
        pCallbacksProvider = 0xffff8d02e9f0
        pCurlResponse = 0x0
        pResponseStr = 0x0
        parser = {pos = 0, toknext = 0, toksuper = 0}
        tokens = {{type = JSMN_UNDEFINED, start = 0, end = 0, size = 0} <repeats 100 times>}
        i = 0
        strLen = 0
        resultLen = 0
        tokenCount = 0
        retention = 0
        jsonInStreamDescription = 0
        requestTerminating = 0

                  streamDescription = {version = 0, deviceName = '\000' <repeats 128 times>, streamName = '\000' <repeats 256 times>, contentType = '\000' <repeats 128 times>, updateVersion = '\000' <repeats 64 times>, streamArn = '\000' <repeats 2048 times>, streamStatus = STREAM_STATUS_CREATING, creationTime = 0, retention = 0, kmsKeyId = '\000' <repeats 2048 times>}
        streamHandle = 0
        callResult = SERVICE_CALL_RESULT_NOT_SET
        __func__ = "describeStreamCurlHandler"
        __FUNCTION__ = "describeStreamCurlHandler"
#3  0x0000ffff8d5b69a4 in  () at /lib/ld-musl-aarch64.so.1
#4  0x0000ffff83fa7000 in  ()
```

##### loaded libraries
```
(gdb) info sharedlibrary
From                To                  Syms Read   Shared Object Library
0x0000ffff8d56c5a0  0x0000ffff8d5c0f70  Yes (*)     /lib/ld-musl-aarch64.so.1
0x0000ffff8d454820  0x0000ffff8d4f7ff0  Yes (*)     /usr/lib/libgstreamer-1.0.so.0
0x0000ffff8d30fdc0  0x0000ffff8d385e30  Yes (*)     /usr/lib/libglib-2.0.so.0
0x0000ffff8d2d2e50  0x0000ffff8d2d8204  Yes (*)     /usr/lib/libintl.so.8
0x0000ffff8d26ed60  0x0000ffff8d298260  Yes (*)     /usr/lib/libgobject-2.0.so.0
0x0000ffff8d23f2a0  0x0000ffff8d23ffb8  Yes (*)     /usr/lib/libgmodule-2.0.so.0
0x0000ffff8d19f4d0  0x0000ffff8d2043bc  Yes (*)     /usr/lib/libpcre2-8.so.0
0x0000ffff8d17dc20  0x0000ffff8d18235c  Yes (*)     /usr/lib/libffi.so.8
0x0000ffff8d13b700  0x0000ffff8d141060  Yes (*)     /usr/lib/gstreamer-1.0/libgstvideotestsrc.so
0x0000ffff8d0a1de0  0x0000ffff8d0fc760  Yes (*)     /usr/lib/libgstvideo-1.0.so.0
0x0000ffff8cd67210  0x0000ffff8cda9f80  Yes (*)     /usr/lib/libgstbase-1.0.so.0
0x0000ffff8ccd0790  0x0000ffff8cd2246c  Yes (*)     /usr/lib/liborc-0.4.so.0
0x0000ffff8d069560  0x0000ffff8d06ebbc  Yes (*)     /usr/lib/gstreamer-1.0/libgstx264.so
0x0000ffff8cc832e0  0x0000ffff8cc9bd94  Yes (*)     /usr/lib/libgstpbutils-1.0.so.0
0x0000ffff8c9d5140  0x0000ffff8cb3b534  Yes (*)     /usr/lib/libx264.so.164
0x0000ffff8c958de0  0x0000ffff8c993e20  Yes (*)     /usr/lib/libgstaudio-1.0.so.0
0x0000ffff8c901140  0x0000ffff8c91b0dc  Yes (*)     /usr/lib/libgsttag-1.0.so.0
0x0000ffff8d036570  0x0000ffff8d043258  Yes (*)     /lib/libz.so.1
0x0000ffff8c8bb010  0x0000ffff8c8ce930  Yes         /opt/amazon-kinesis-video-streams-producer-sdk-cpp/build/libgstkvssink.so
0x0000ffff8c846180  0x0000ffff8c859690  Yes         /opt/amazon-kinesis-video-streams-producer-sdk-cpp/build/libKinesisVideoProducer.so
0x0000ffff8c79b310  0x0000ffff8c7ecea0  Yes         /opt/amazon-kinesis-video-streams-producer-sdk-cpp/build/dependency/libkvscproducer/kvscproducer-src/libcproducer.so
0x0000ffff8c71b750  0x0000ffff8c757920  Yes         /opt/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/lib/liblog4cplus-1.2.so.5
0x0000ffff8c565af0  0x0000ffff8c66d01c  Yes (*)     /usr/lib/libstdc++.so.6
0x0000ffff8c480aa0  0x0000ffff8c48f064  Yes (*)     /usr/lib/libgcc_s.so.1
0x0000ffff8c241000  0x0000ffff8c3a7890  Yes         /opt/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/lib/libcrypto.so.1.1
0x0000ffff8c11fe40  0x0000ffff8c17e9ac  Yes         /opt/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/lib/libcurl.so
0x0000ffff8c090c20  0x0000ffff8c0d9ca0  Yes         /opt/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/lib/libssl.so.1.1
(*): Shared library is missing debugging information.
```

jdelapla, 2023-10-11:
Hello,
We've resolve the issue in a PR that's undergoing review. The issue is due to the smaller default stack size of Alpine Linux. You'll need to call `cmake -DCONSTRAINED_DEVICE=ON` with the fix to get around this problem.

alan-willard-simplisafe, 2023-10-12:
Thanks for the update! Will give this a shot at my earliest opportunity

hassanctech, 2023-10-16:
Looks like this issue is resolved, closing this issue.  @alan-willard-simplisafe if you still see the issue, please re-open.

closed 2023-10-16 by hassanctech