awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #1071: [BUG] SDK are faling to cross-compile for ARM64 platform

koratshail, 2023-09-21:
**Logging**
/home/ambadocker/dev/h32/output/h32_ambalink_defconfig/build/aws_kinesis_kvs/open-source/local/libcurl/build/src/project_libcurl/lib/md4.c:31:10: fatal error: openssl/opensslconf.h: No such file or directory 31 | #include <openssl/opensslconf.h> |          ^~~~~~~~~~~~~~~~~~~~~~~ compilation terminated. lib/CMakeFiles/libcurl.dir/build.make:598: recipe for target 'lib/CMakeFiles/libcurl.dir/md4.c.o' failed make[7]: *** [lib/CMakeFiles/libcurl.dir/md4.c.o] Error 1 make[7]: Leaving directory '/home/ambadocker/dev/h32/output/h32_ambalink_defconfig/build/aws_kinesis_kvs/open-source/local/libcurl/build/src/project_libcurl-build' CMakeFiles/Makefile2:1572: recipe for target 'lib/CMakeFiles/libcurl.dir/all' failed make[6]: Leaving directory '/home/ambadocker/dev/h32/output/h32_ambalink_defconfig/build/aws_kinesis_kvs/open-source/local/libcurl/build/src/project_libcurl-build' make[6]: *** [lib/CMakeFiles/libcurl.dir/all] Error 2 Makefile:143: recipe for target 'all' failed make[5]: *** [all] Error 2 make[5]: Leaving directory '/home/ambadocker/dev/h32/output/h32_ambalink_defconfig/build/aws_kinesis_kvs/open-source/local/libcurl/build/src/project_libcurl-build' CMakeFiles/project_libcurl.dir/build.make:115: recipe for target 'build/src/project_libcurl-stamp/project_libcurl-build' failed make[4]: *** [build/src/project_libcurl-stamp/project_libcurl-build] Error 2 make[4]: Leaving directory '/home/ambadocker/dev/h32/output/h32_ambalink_defconfig/build/aws_kinesis_kvs/open-source/local/libcurl' CMakeFiles/Makefile2:78: recipe for target 'CMakeFiles/project_libcurl.dir/all' failed make[3]: *** [CMakeFiles/project_libcurl.dir/all] Error 2 make[3]: Leaving directory '/home/ambadocker/dev/h32/output/h32_ambalink_defconfig/build/aws_kinesis_kvs/open-source/local/libcurl' Makefile:86: recipe for target 'all' failed make[2]: *** [all] Error 2 make[2]: Leaving directory '/home/ambadocker/dev/h32/output/h32_ambalink_defconfig/build/aws_kinesis_kvs/open-source/local/libcurl' CMake Error at dependency/libkvscproducer/kvscproducer-src/CMake/Utilities.cmake:93 (message): CMake step for libcurl failed: 2 Call Stack (most recent call first): dependency/libkvscproducer/kvscproducer-src/CMakeLists.txt:95 (build_dependency)

**Describe the bug**
I am trying to build a Kinesis video streams producer sdk cpp package for my ARM64 platform.

all attempts to build this package with the dependencies failed.
First of all, the target platform needed to be specified explicitly as shown below:
$(pkg)_CONF_OPTS = -DBUILD_GSTREAMER_PLUGIN=ON -DBUILD_OPENSSL_PLATFORM=linux-aarch64 -DBUILD_LOG4CPLUS_HOST=arm-linux

The problem that couldn't be solved was how to build libcurl: the compilation failed because one source couldn't find the <openssl/opensslconf.h> include file.
Providing the file's location using an environment variable, as export OPENSSL_INCLUDE_DIR = $(@d)/open-source/local/include
made the compilations pass, but failed at the linking stage, due to wrong file format, because the files were compiled for the build host platform, not the target one.

**To Reproduce**
Steps to reproduce the behavior:
Try to build this SDK by enabling build dependencies like OpenSSL, cURL etc. 

**Expected behavior**
It should cross-compile without any issue.

**SDK version number**
Release number v3.4.1

**Open source building**
Adding the following third-party lib compilation flags.
-DBUILD_GSTREAMER_PLUGIN=ON -DBUILD_OPENSSL_PLATFORM=linux-aarch64 -DBUILD_LOG4CPLUS_HOST=arm-linux

**Platform (please complete the following information):**
 - OS: Linux
 - Version [e.g. 20.04]

sirknightj, 2023-09-21:
Hi, the cross compiling build for arm64 is passing on the ci.

Can you confirm this instructions work for you?
https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/.github/workflows/ci.yml#L311

koratshail, 2023-09-22:
I am trying to add this SDK into build root system. Following is the cmake file to add this package to buildroot.

`#############################################################
#
# aws_kinesis_kvs
#
#############################################################

pkg                     = AWS_KINESIS_KVS

$(pkg)_SITE             = https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/archive/refs/tags
$(pkg)_SOURCE           = v3.3.1.tar.gz
$(pkg)_INSTALL_TARGET   = YES
$(pkg)_CONF_OPTS        = -DBUILD_GSTREAMER=ON  -DBUILD_GSTREAMER_PLUGIN=ON -DBUILD_OPENSSL_PLATFORM=linux-aarch64 -DBUILD_LOG4CPLUS_HOST=arm-linux -DBUILD_DEPENDENCIES=ON
$(pkg)_DEPENDENCIES     = openssl libcurl log4cplus gstreamer1 gst1-plugins-base

define $(pkg)_INSTALL_TARGET_CMDS
	$(INSTALL) -m 755 -D $(@D)/libgstkvssink.so $(TARGET_DIR)/usr/lib/gstreamer-1.0
	$(INSTALL) -m 755 -D $(@D)/libKinesisVideoProducer.so $(TARGET_DIR)/usr/lib
	$(INSTALL) -m 755 -D $(@D)/dependency/libkvscproducer/kvscproducer-src/libcproducer.so $(TARGET_DIR)/usr/lib
	$(INSTALL) -m 644 -D $(@D)/certs/cert.pem $(TARGET_DIR)/etc/ssl/certs/aws_root_ca.pem
endef

$(eval $(cmake-package))`

I am building this SDK with Gstreamer pluging = ON, Am I missing anything else here?

koratshail, 2023-09-22:
@sirknightj  It is working as per without Gstremmer https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/.github/workflows/ci.yml#L311

koratshail, 2023-09-25:
@sirknightj  Could you us how to add this SDK to build root system?

disa6302, 2023-09-28:
Closing this as a duplicate of the discussion: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/discussions/1073

closed 2023-09-28 by disa6302