awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #1093: [BUG] Unable to open file when using the example README commands on Windows

junglie85, 2023-10-25:
The following command fails when executed from `C:\Dev\amazon-kinesis-video-streams-producer-sdk-cpp\build`
```
gst-launch-1.0 -v filesrc location="YourAudioVideo.mkv" ! matroskademux name=demux ! queue ! h264parse ! kvssink name=sink stream-name="my_stream_name" access-key="YourAccessKeyId" secret-key="YourSecretAccessKey" aws-region="region" streaming-type=offline demux. ! queue ! aacparse ! sink.
```

Fail to open file:
```
[DEBUG] [25-10-2023 15:29:26:245.162 GMT Standard Time] Kinesis Video client and stream metrics for my_stream_name
	>> Overall storage byte size: 134217728
	>> Available storage byte size: 134217728
	>> Allocated storage byte size: 0
	>> Total view allocation byte size: 576080
	>> Total streams elementary frame rate (fps): 100
	>> Total streams transfer rate (bps): 33554432 (32768 Kbps)
	>> Current view duration (ms): 0
	>> Overall view duration (ms): 0
	>> Current view byte size: 0
	>> Overall view byte size: 0
	>> Current elementary frame rate (fps): 100
	>> Current transfer rate (bps): 33554432 (32768 Kbps)
[WARN ] [25-10-2023 15:29:26:250.990 GMT Standard Time] putKinesisVideoFrame(): [my_stream_name] Failed to submit frame to Kinesis Video client. status: 0x00000009 decoding timestamp: 0 presentation timestamp: 16982477650000000
[ERROR] [25-10-2023 15:29:26:251.082 GMT Standard Time] Put frame for my_stream_name failed with 0x9
```

But it finds the file just fine:
```
0:00:01.155650000 11700 000002133376D000 DEBUG           GST_PIPELINE parse.l:91:priv_gst_parse_yylex: flex: ASSIGNMENT: location=YourAudioVideo.mkv
0:00:01.155681000 11700 000002133376D000 DEBUG                default types.h:84:gst_parse_unescape: unescaping YourAudioVideo.mkv
0:00:01.155729000 11700 000002133376D000 DEBUG                GST_URI gsturi.c:946:gst_filename_to_uri: 'YourAudioVideo.mkv' -> 'file:///C:/Dev/amazon-kinesis-video-streams-producer-sdk-cpp/build/YourAudioVideo.mkv'
0:00:01.155761000 11700 000002133376D000 INFO                 filesrc gstfilesrc.c:261:gst_file_src_set_location: filename : YourAudioVideo.mkv
0:00:01.155791000 11700 000002133376D000 INFO                 filesrc gstfilesrc.c:262:gst_file_src_set_location: uri      : file:///C:/Dev/amazon-kinesis-video-streams-producer-sdk-cpp/build/YourAudioVideo.mkv
...
0:00:01.711844000 11700 000002133376D000 INFO                 filesrc gstfilesrc.c:468:gst_file_src_start:<filesrc0> opening file YourAudioVideo.mkv
0:00:01.712190000 11700 000002133376D000 DEBUG                basesrc gstbasesrc.c:3498:gst_base_src_start_complete:<filesrc0> starting source
0:00:01.712235000 11700 000002133376D000 DEBUG                basesrc gstbasesrc.c:3511:gst_base_src_start_complete:<filesrc0> setting size 431058044
0:00:01.712268000 11700 000002133376D000 DEBUG                basesrc gstbasesrc.c:3522:gst_base_src_start_complete:<filesrc0> format: bytes, have size: 1, size: 431058044, duration: 431058044
0:00:01.712300000 11700 000002133376D000 DEBUG                basesrc gstbasesrc.c:3525:gst_base_src_start_complete:<filesrc0> is seekable: 1
0:00:01.712332000 11700 000002133376D000 DEBUG                basesrc gstbasesrc.c:3530:gst_base_src_start_complete:<filesrc0> is random_access: 1
0:00:01.712363000 11700 000002133376D000 DEBUG                basesrc gstbasesrc.c:3635:gst_base_src_start_wait:<filesrc0> got ok
```

**Expected behavior**
Video to stream to my KVS.

**SDK version number**
Commit [de4299f](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/commit/de4299f942bb89c867d0dc73aeb4bc875d560a8f)

**Open source building**
If it is a build issue, include 3rd party library version and steps to how you are building it

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Platform (please complete the following information):**
 - OS: Windows
 - Version 10

disa6302, 2023-10-27:
This is interesting because the SDK does not open media files, rather it expects H.264/H.265 video frames and/or AAC/A_MS/ACM audio frames. Could you attach complete verbose logs? You can get this by setting log level to TRACE here: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/develop/kvs_log_configuration#L1

junglie85, 2023-10-30:
I copy / pasted the same commands I was running last week when I received these errors... and it all works as I expect. Going to put this down to user error somewhere.

closed 2023-10-30 by junglie85

sirknightj, 2025-02-04:
I ran into this issue in https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/actions/runs/13128407619/job/36628920227.

Traced it to setting `KVS_DEBUG_DUMP_DATA_FILE_DIR`, when the directory didn't exist.