awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #1098: How to use rtmp stream format for kvs ingestion, i don't have rtsp camera

sundaram123krishnan, 2023-11-04:
A one liner description about the use case and what you are trying to achieve

** Logging **
Add relevent SDK logging.  IMPORTANT NOTE:  Please make sure to NOT share AWS access credentials under any circumstance!  Please make sure they are not in the logs.

** Any design considerations/constraints **
Explain in detail how you would like to integrate our SDK into your solution 

** If you would not like to open an issue to discuss your solution in open-platform, please email your question to kinesis-video-support@amazon.com **

sirknightj, 2023-11-04:
Hi @sundaram123krishnan, KVS is [source-agnostic](https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/what-is-kinesis-video.html). To grab media from RTMP, you would need to use [rtmpsrc](https://gstreamer.freedesktop.org/documentation/rtmp/rtmpsrc.html?gi-language=c) GStreamer element. There are some sample GStreamer pipelines in this other issue: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/issues/433#issuecomment-621420133.

We also have a [file uploader sample](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/samples/kvs_gstreamer_file_uploader_sample.cpp#L203). There is a sample video file [here](https://catalog.us-east-1.prod.workshops.aws/kinesis-video-streams/en-US/3-stream/2-sample-app).

If you wanted to try streaming from RTSP, you can also set up a docker container as an RTSP source. 
```
sudo docker run --rm -e RTSP_PORT=8554 -p 8554:8554 -e GST_PIPELINE="videotestsrc pattern=ball ! clockoverlay time-format='%a %B %d, %Y %I:%M:%S %p' ! textoverlay text='DemoStream' ! videoscale ! video/x-raw,width=640,height=480 ! capsfilter ! queue ! x264enc bframes=0 key-int-max=45 bitrate=512 ! rtph264pay name=pay0 pt=96" ullaakut/rtspatt
```
Pipeline to upload that to kvs:
```
gst-launch-1.0 -v rtspsrc location="rtsp://0.0.0.0:8554/live.sdp" short-header=TRUE ! rtph264depay ! h264parse ! kvssink stream-name="demo-stream" access-key="x" secret-key="x" aws-region="us-west-2"
```

closed 2023-11-04 by sirknightj

sundaram123krishnan, 2023-11-04:
@sirknightj thx for ur kind response, so i just need to add " gst-launch-1.0 rtmpsrc location="rtmp://somehost/someurl live=1" ! fakesink" this while producing sdk? sry if it's an newbie question, i am new to it

sirknightj, 2023-11-05:
To use this producer SDK, you would need to build and use the `kvssink` GStreamer element as the **sink** (last element) in your GStreamer pipeline. Different GStreamer elements are connected via `!`. `rtmpsrc` cannot connect to `kvssink` directly; you will need other elements will transform the media received from `rtmpsrc` to something `kvssink` can handle. From other issues (https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/issues/197#issuecomment-585034352) the suggested pipeline looks to be rtmpsrc -> flvdemux -> h264parse -> kvssink.

`kvssink` doesn't come installed with GStreamer. To build it, follow the README in this repository. After running the `cmake` and `make` commands, set the `GST_PLUGIN_PATH` environment variable to the directory containing the `kvssink` binary, then you can use `kvssink`.

You can find more information about `kvssink` [here](https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/examples-gstreamer-plugin.html).

sundaram123krishnan, 2023-11-05:
ohh ok thx a lot @sirknightj