awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #1123: [QUESTION] Set retention to 0 hours and play video

lucabergamini, 2024-01-08:
I'm trying to set a stream with a retention of 0 and play the live video in the aws console, but I'm getting an error about the retention being 0.

[image]

This should be possible according to https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/API_CreateStream.html :
```
When the DataRetentionInHours value is 0, 
consumers can still consume the fragments that remain in the service host buffer, 
which has a retention time limit of 5 minutes and a retention memory limit of 200 MB. 
Fragments are removed from the buffer when either limit is reached.
```

The stream is successfully created and the retention is correctly set to 0. The stream is not set in offline mode in the producer library (as that returns a clear error). 

The reason behind this setting is to minimise the cost of the BytesHr metric.

sirknightj, 2024-01-09:
Hi @lucabergamini,

Thanks for bringing this to our attention. We will be working on updating the documentation for this.

[GetMedia](https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/API_dataplane_GetMedia.html) is the only API where you can consume media from a stream with 0 retention. The AWS console uses the [DASH](https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/API_reader_GetDASHStreamingSessionURL.html), which requires retention > 0.

Note that if you are looking to view live media via peer-to-peer streaming (e.g. from a security camera on your phone), we'd recommend checking out Kinesis Video Streams with WebRTC to see if it fits your use case.

lucabergamini, 2024-01-10:
Hi @sirknightj 
Thanks for the clarification on this. I still have a question though. We are using HLS in our case but to play videos in the browser we don't go through [GetMedia](https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/API_dataplane_GetMedia.html) but through [GetHLSStreamingSessionURL](https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/API_reader_GetHLSStreamingSessionURL.html). Does that mean we can't use 0 retention in this case?

sirknightj, 2024-01-10:
Correct. HLS playback requires retention > 0.

lucabergamini, 2024-01-10:
Got it, thanks a lot!

closed 2024-01-10 by lucabergamini