awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #1124: [FEATURE] Writing frames to content store even before kinesis video stream initialization.

AnilNS, 2024-01-15:
**Is your feature request related to a problem? Please describe.**
Writing frames to hybrid content store won't work if during start of the application, there is network offline. 

**Describe the solution you'd like**
We are using the kvs gstreamer sample application for the video streaming application.
We do want to leverage hybrid content store mechanism in our application.
Currently we are able to utilize hybrid content store and it is working fine.
Suppose at the start of application, there is no network connectivity then kinesis video initialization fails.
Kinesis video SDK needs a unique pointer to write frames to content store. Due to this limitation we will loose the frames.
Can we bypass this limitation and write frames to content store from start of the application?

schmidty1913, 2024-06-07:
@AnilNS Can you elaborate more on what you are using for the hybrid content store mechanism. 
I am looking to utilize RAM / Disk for buffering when the device is offline as well.
I have experience with C++ and could make these changes to this library if we define the requirements properly.

MushMal, 2024-06-07:
This is a relatively easy feature to implement with the main hinderance being the state machinery. It would be great to perhaps capture the full details and what edge cases are with this feature and why this is important to implement in the SDK rather than the application layers. For example, the SDK has no way of knowing when the network is back on/available as it's application and platform specific. I remember a while back one of the patrons had a mobile application which would create a temp storage and store the frames (with the metadata). The application was also listening on the airplane mode and would reply the collected frame buffer into the SDK with their timestamps faster than realtime. Perhaps sub-optimal for sure but nevertheless a solution.