awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #1133: [QUESTION] All my frames are getting dropped I think and data upload rate is very low (rtsp H265 at 1020*1080)

msalmonw, 2024-01-26:
Im using a Dahua rtsp camera to stream to kvssink but I do not succeed. If im using h264 on the camera settings than I get "Bus Error". I switched to x265 at 1920x1080 but then again my frames are getting dropped and data upload is very very slow. 

** Logging **
```
gatewise@devomech:~/Desktop/aws-certs $ gst-launch-1.0 rtspsrc location=rtsp://admin:GatewiseCamera123@192.168.1.141:554 ! rtph265depay ! h265parse ! video/x-h265,stream-format=byte-stream,width=1920,height=1080 ! kvssink stream-name="$STREAM_NAME" aws-region="$AWS_REGION" access-key="$AWS_ACCESS_KEY_ID" secret-key="$AWS_SECRET_ACCESS_KEY"
Setting pipeline to PAUSED ...
log4cplus:ERROR could not open file ../kvs_log_configuration
INFO - Logger config being used: ../kvs_log_configuration
2024-01-26 03:34:44 [4157881344] INFO - Setting session token from env for gatewise-camera-01
2024-01-26 03:34:44 [4157881344] INFO - User agent string: AWS-SDK-KVS-CPP-CLIENT/3.4.1
2024-01-26 03:34:44 [4157881344] INFO - createKinesisVideoClient(): Creating Kinesis Video Client
2024-01-26 03:34:44 [4157881344] INFO - heapInitialize(): Initializing native heap with limit size 134217728, spill ratio 0% and flags 0x00000001
2024-01-26 03:34:44 [4157881344] INFO - heapInitialize(): Creating AIV heap.
2024-01-26 03:34:44 [4157881344] INFO - heapInitialize(): Heap is initialized OK
2024-01-26 03:34:44 [4157881344] DEBUG - getSecurityTokenHandler invoked
2024-01-26 03:34:44 [4157881344] DEBUG - Refreshing credentials. Force refreshing: 0 Now time is: 1706261684760696328 Expiration: 0
2024-01-26 03:34:44 [4157881344] INFO - createDeviceResultEvent(): Create device result event.
2024-01-26 03:34:44 [4157881344] DEBUG - clientReadyHandler invoked
2024-01-26 03:34:44 [4157881344] INFO - Try creating stream for gatewise-camera-01
2024-01-26 03:34:44 [4157881344] INFO - Creating Kinesis Video Stream gatewise-camera-01
2024-01-26 03:34:44 [4157881344] INFO - createKinesisVideoStream(): Creating Kinesis Video Stream.
2024-01-26 03:34:44 [4157881344] INFO - logStreamInfo(): SDK version: de4299f942bb89c867d0dc73aeb4bc875d560a8f
2024-01-26 03:34:44 [4157881344] DEBUG - logStreamInfo(): Kinesis Video Stream Info
2024-01-26 03:34:44 [4157881344] DEBUG - logStreamInfo():       Stream name: gatewise-camera-01
2024-01-26 03:34:44 [4157881344] DEBUG - logStreamInfo():       Streaming type: STREAMING_TYPE_REALTIME
2024-01-26 03:34:44 [4157881344] DEBUG - logStreamInfo():       Content type: video/h265
2024-01-26 03:34:44 [4157881344] DEBUG - logStreamInfo():       Max latency (100ns): 600000000
2024-01-26 03:34:44 [4157881344] DEBUG - logStreamInfo():       Fragment duration (100ns): 20000000
2024-01-26 03:34:44 [4157881344] DEBUG - logStreamInfo():       Key frame fragmentation: Yes
2024-01-26 03:34:44 [4157881344] DEBUG - logStreamInfo():       Use frame timecodes: Yes
2024-01-26 03:34:44 [4157881344] DEBUG - logStreamInfo():       Absolute frame timecodes: Yes
2024-01-26 03:34:44 [4157881344] DEBUG - logStreamInfo():       Nal adaptation flags: 0
2024-01-26 03:34:44 [4157881344] DEBUG - logStreamInfo():       Average bandwidth (bps): 4194304
2024-01-26 03:34:44 [4157881344] DEBUG - logStreamInfo():       Framerate: 25
2024-01-26 03:34:44 [4157881344] DEBUG - logStreamInfo():       Buffer duration (100ns): 1200000000
2024-01-26 03:34:44 [4157881344] DEBUG - logStreamInfo():       Replay duration (100ns): 400000000
2024-01-26 03:34:44 [4157881344] DEBUG - logStreamInfo():       Connection Staleness duration (100ns): 600000000
2024-01-26 03:34:44 [4157881344] DEBUG - logStreamInfo():       Store Pressure Policy: 1
2024-01-26 03:34:44 [4157881344] DEBUG - logStreamInfo():       View Overflow Policy: 1
2024-01-26 03:34:44 [4157881344] DEBUG - logStreamInfo():       Allow stream creation: Yes
2024-01-26 03:34:44 [4157881344] DEBUG - logStreamInfo():       Segment UUID: NULL
2024-01-26 03:34:44 [4157881344] DEBUG - logStreamInfo():       Frame ordering mode: 0
2024-01-26 03:34:44 [4157881344] DEBUG - logStreamInfo(): Track list
2024-01-26 03:34:44 [4157881344] DEBUG - logStreamInfo():       Track id: 1
2024-01-26 03:34:44 [4157881344] DEBUG - logStreamInfo():       Track name: kinesis_video
2024-01-26 03:34:44 [4157881344] DEBUG - logStreamInfo():       Codec id: V_MPEGH/ISO/HEVC
2024-01-26 03:34:44 [4157881344] DEBUG - logStreamInfo():       Track type: TRACK_INFO_TYPE_VIDEO
2024-01-26 03:34:44 [4157881344] DEBUG - logStreamInfo():       Track cpd: NULL
2024-01-26 03:34:44 [4157881344] INFO - writeHeaderCallback(): RequestId: 588e02d0-9e9e-4a71-ab87-df8e6f7a1c2e
2024-01-26 03:34:45 [3941057600] DEBUG - describeStreamCurlHandler(): [] DescribeStream API response: {"StreamInfo":{"CreationTime":1.706103516332E9,"DataRetentionInHours":24,"DeviceName":null,"IngestionConfiguration":null,"KmsKeyId":"arn:aws:kms:us-west-2:571292287453:alias/aws/kinesisvideo","MediaType":null,"Status":"ACTIVE","StreamARN":"arn:aws:kinesisvideo:us-west-2:571292287453:stream/gatewise-camera-01/1706103516332","StreamName":"gatewise-camera-01","Version":"gcae41F4MPYEmwAhLGzl"}}
2024-01-26 03:34:45 [3941057600] INFO - describeStreamResultEvent(): Describe stream result event.
2024-01-26 03:34:45 [3941057600] WARN - describeStreamResult(): Retention period returned from the DescribeStream call doesn't match the one specified in the StreamInfo
2024-01-26 03:34:45 [3941057600] WARN - describeStreamResult(): Content type returned from the DescribeStream(null) call doesn't match the one specified in the StreamInfo(video/h265)
2024-01-26 03:34:45 [3941057600] WARN - describeStreamResult(): [gatewise-camera-01] Retention period returned from the DescribeStream call doesn't match the one specified in the StreamInfo
2024-01-26 03:34:45 [3941057600] WARN - describeStreamResult(): [gatewise-camera-01] Content type returned from the DescribeStream call doesn't match the one specified in the StreamInfo
2024-01-26 03:34:45 [3941057600] INFO - writeHeaderCallback(): RequestId: 2dc67329-bf3d-4763-ac48-cbbc1c96cd15
2024-01-26 03:34:45 [3931108416] DEBUG - getStreamingEndpointCurlHandler(): [gatewise-camera-01] GetStreamingEndpoint API response: {"DataEndpoint":"https://s-8e923eff.kinesisvideo.us-west-2.amazonaws.com"}
2024-01-26 03:34:45 [3931108416] INFO - getStreamingEndpointResultEvent(): Get streaming endpoint result event.
2024-01-26 03:34:45 [3931108416] DEBUG - getStreamingTokenHandler invoked
2024-01-26 03:34:45 [3931108416] DEBUG - Refreshing credentials. Force refreshing: 1 Now time is: 1706261685455260420 Expiration: 18446744073709551615
2024-01-26 03:34:45 [3931108416] INFO - getStreamingTokenResultEvent(): Get streaming token result event.
2024-01-26 03:34:45 [3931108416] DEBUG - streamReadyHandler invoked
2024-01-26 03:34:45 [3931108416] Stream is ready
Pipeline is live and does not need PREROLL ...
Progress: (open) Opening Stream
Pipeline is PREROLLED ...
Prerolled, waiting for progress to finish...
Progress: (connect) Connecting to rtsp://admin:GatewiseCamera123@192.168.1.141:554
Progress: (open) Retrieving server options
Progress: (open) Retrieving media info
[... 83 lines cut ...]
 KinesisVideoStream base result is [599]. Executing KVS retry handler of retry strategy type [1]
2024-01-26 03:35:46 [3826230336] INFO - putStreamResultEvent(): Put stream result event. New upload handle 2
2024-01-26 03:35:46 [3826230336] WARN - notifyDataAvailable(): [gatewise-camera-01] Failed to un-pause curl with error: 43. Curl object 0xe9192800
DEBUG - Dropping frame with flag: 1536 for gatewise-camera-01
DEBUG - Dropping frame with flag: 1536 for gatewise-camera-01
DEBUG - Dropping frame with flag: 1536 for gatewise-camera-01
DEBUG - postReadCallback(): [gatewise-camera-01] Pausing CURL read for upload handle: 2
2024-01-26 03:35:52 [3817837632] INFO - writeHeaderCallback(): RequestId: d62e2efd-4549-f05d-8803-51caaa3edc07
DEBUG - Dropping frame with flag: 1536 for gatewise-camera-01
DEBUG - Dropping frame with flag: 1536 for gatewise-camera-01
DEBUG - Dropping frame with flag: 1536 for gatewise-camera-01
DEBUG - Dropping frame with flag: 1536 for gatewise-camera-01
DEBUG - Dropping frame with flag: 1536 for gatewise-camera-01
DEBUG - Dropping frame with flag: 1536 for gatewise-camera-01
DEBUG - Dropping frame with flag: 1536 for gatewise-camera-01
DEBUG - Dropping frame with flag: 1536 for gatewise-camera-01
DEBUG - Dropping frame with flag: 1536 for gatewise-camera-01
DEBUG - Dropping frame with flag: 1536 for gatewise-camera-01
DEBUG - Dropping frame with flag: 1536 for gatewise-camera-01
DEBUG - Dropping frame with flag: 1536 for gatewise-camera-01
DEBUG - Dropping frame with flag: 1536 for gatewise-camera-01
^Chandling interrupt.3826230336] 0:01:31.8 / 99:99:99.
```

Please help!

msalmonw, 2024-01-29:
Yes it worked, I resolved the issue by compiling the with -DALIGNED_MEMORY_MODEL=TRUE.

Also now I have another problem, I can generate session tokens completely fine with my certificates, but I can not start the stream directly using the certificates. The command I use is:

`
curl -H "x-amzn-iot-thingname:gatewise-camera-01" --cert certificate.pem --key private.pem.key https://ctdaryo6b4e5l.credentials.iot.us-west-2.amazonaws.com/role-aliases/gatewise-camera-role-alias/credentials --cacert cacert.pem > token.json
`
And the output:

`log4cplus:ERROR could not open file ../kvs_log_configuration
INFO - Logger config being used: ../kvs_log_configuration
2024-01-29 11:43:48 [4159011840] INFO - Setting session token from env for gatewise-camera-01
2024-01-29 11:43:48 [4159011840] INFO - Using iot credential provider within KVS sink for gatewise-camera-01
2024-01-29 11:43:48 [4159011840] INFO - User agent string: AWS-SDK-KVS-CPP-CLIENT/3.4.1
2024-01-29 11:43:48 [4159011840] DEBUG - Creating IoT auth callbacks.
2024-01-29 11:43:48 [4159011840] ERROR - Unable to create Iot Credential provider. Error status: 0x4000000e
ERROR: from element /GstPipeline:pipeline0/GstKvsSink:kvssink0: Could not initialise supporting library.
Additional debug info:
/home/gatewise/Desktop/GateWiseCamera/amazon-kinesis-video-streams-producer-sdk-cpp/src/gstreamer/gstkvssink.cpp(1640): gst_kvs_sink_change_state (): /GstPipeline:pipeline0/GstKvsSink:kvssink0:
Failed to init kvs producer. Error: Unable to create Iot Credential provider. Error status: 0x4000000e
ERROR: pipeline doesn't want to preroll.
Failed to set pipeline to PAUSED.
2024-01-29 11:43:48 [4159011840] Setting pipeline to NULL ...
Freeing pipeline ...`

niyatim23, 2024-02-07:
The error code you're seeing is `0x4000000e`: `STATUS_DIRECTORY_ENTRY_STAT_ERROR`. This could be something related to directory access probably. Can you double check all your paths and permissions?