awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #1134: [QUESTION] Gstreamer event/message for streaming/connection releated issues in KVS SDK

hardik237, 2024-02-01:
I am developing a media gateway for Local RTSP cameras to AWS Kinesis Video stream. I  want to build a robust application that can handle network outage scenarios using the KVS Gstreamer plugin.

I guess, as of now KVS producer SDK does not support file-based buffering to a large number of frames in case of a network outage, and due to that it will drop incoming live streams.

I want to handle this case by launching a gstreamer recording pipeline in such cases. These recordings will be synced when we get the connection back. To start/stop these recording pipelines I need a connection status of the KVS stream. 

**Expected behavior:** I should get connection-related events/messages in the gstreamer bus callback or 'stream-error' callback in kvssink.
**Current behavior:** I am not getting any stream-related error message in case of network outage or connection timeout. 

I am wondering are these messages and events part of the current gstreamer KVS plugin. Or is there any other method using which I can get the stream status?

stefankiesz, 2024-05-03:
Hi Hardik, that is correct, kvssink does not readily have a write-to-file feature for network outages. Kvssink does not post any messages upon such stream errors, and we will be taking note of this as a feature request. This will likely involve emitting a message from the KVS Producer callbacks which are defined [here](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/src/gstreamer/KvsSinkStreamCallbackProvider.cpp) for kvssink. From my testing by streaming with kvssink and then turning off wifi, it seems `streamLatencyPressureHandler` would be the appropriate callback to react to for beginning writing the stream to file, or perhaps `bufferDurationOverflowPressureHandler` if the desire behavior is to first fill the local Producer buffer before writing to file.

stefankiesz, 2024-05-03:
Closing, if you would like to extend discussion on the topic, please feel free to open up a [discussions](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/discussions) topic.

closed 2024-05-03 by stefankiesz