awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #1175: [BUG] Streaming h.265 video from RTSP

deorwine, 2024-05-08:
hi, 

i have tried to stream h265 rtsp stream o kvs but it is not working. 

gst-launch-1.0 rtspsrc location="rtsp://livepickleballcourts:Astro24%21@lpcipnfw.ddns.net:55405/cam/realmonitor?channel=3&subtype=0" ! rtph265depay ! h265parse ! video/x-h265,stream-format=byte-stream,alignment=au ! kvssink stream-name=pi4n9ko44j6k storage-size=128

but it is not working when using h264  workig perfectly. can you please chek.

mateuszboryn, 2024-05-11:
Hi, I'm also trying to stream h265. Please check if changing `stream-format=byte-stream` to `stream-format=hvc1` ot `hev1` changes anything.

sirknightj, 2024-05-11:
Could you share the error you are seeing?

deorwine, 2024-05-14:
yes streaming in command not given error when trying to get hls stream medi and also checking on aws console to view video => The detected fragment codec is not supported

[image]

sirknightj, 2024-05-16:
> streaming in command not given error

It is likely that media uploading was successful. You can confirm this by downloading the video (clip) and playing it locally.

The issue mentioned on the screenshot is with browser support. H.265 isn't supported in all browsers and devices: https://caniuse.com/hevc. You will need to choose a browser where H.265 playback is supported.

If playback across wide variety of browsers is a priority for your use case, you may want to use H.264 instead, as H.264 browser support is more wide.

closed 2024-05-20 by disa6302