awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #1184: How to resolve a "MAX_FRAGMENT_DURATION_REACHED" error when trying to stream RTSP camera feed into KVS

tawiemadavz, 2024-06-03:
I have been trying for a few days to connect a HikVision camera that is on my home network to KVS and if keeps failing with error message "MAX_FRAGMENT_DURATION_REACHED". 

I have tried to adjust the camera settings for my IP camera to try get the GOP (as i understand it) as low as possible but still facing the same issue.

I am running the latest docker file for this build on an M1 Mac. I have used VLC player and ffmpeg to double check if the stream is working and it is working fine.

I have used the following commands to try start the stream:

`./kvs_gstreamer_sample FacialVideoStream rtsp://XXXXXX:XXXXX@IP/Streaming/Channels/101`

and 

`gst-launch-1.0 rtspsrc location="rtsp://XXXXXX:XXXXX@IP/Streaming/Channels/101" short-header=TRUE ! rtph264depay ! h264parse ! video/x-h264, format=avc,alignment=au ! kvssink stream-name="FacialVideoStream"`

Credentials were set i have just omitted them and commands were pointed to the correct region.

See below camera settings and logs. Any help would be appreciated. Thanks!

[image]

`[INFO ] [03-06-2024 11:35:49:490.910 GMT] Using region: us-east-1
[INFO ] [03-06-2024 11:35:49:492.788 GMT] Using aws credentials for Kinesis Video Streams
[INFO ] [03-06-2024 11:35:49:492.990 GMT] No session token was detected.
[INFO ] [03-06-2024 11:35:49:503.042 GMT] createKinesisVideoClient(): Creating Kinesis Video Client
[INFO ] [03-06-2024 11:35:49:503.734 GMT] heapInitialize(): Initializing native heap with limit size 134217728, spill ratio 0% and flags 0x00000001
[INFO ] [03-06-2024 11:35:49:503.759 GMT] heapInitialize(): Creating AIV heap.
[INFO ] [03-06-2024 11:35:49:503.938 GMT] heapInitialize(): Heap is initialized OK
[DEBUG] [03-06-2024 11:35:49:504.384 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
[DEBUG] [03-06-2024 11:35:49:504.553 GMT] getSecurityTokenHandler invoked
[DEBUG] [03-06-2024 11:35:49:505.241 GMT] Refreshing credentials. Force refreshing: 0 Now time is: 1717414549504689000 Expiration: 0
[INFO ] [03-06-2024 11:35:49:505.373 GMT] New credentials expiration is 1717418149
[DEBUG] [03-06-2024 11:35:49:505.619 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000000000000010
[INFO ] [03-06-2024 11:35:49:505.736 GMT] createDeviceResultEvent(): Create device result event.
[DEBUG] [03-06-2024 11:35:49:505.929 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000010, Next state: 0x0000000000000040
[DEBUG] [03-06-2024 11:35:49:506.102 GMT] clientReadyHandler invoked
[DEBUG] [03-06-2024 11:35:49:506.572 GMT] Client is ready
[INFO ] [03-06-2024 11:35:49:508.445 GMT] Creating Kinesis Video Stream FacialVideoStream
[INFO ] [03-06-2024 11:35:49:508.885 GMT] createKinesisVideoStream(): Creating Kinesis Video Stream.
[DEBUG] [03-06-2024 11:35:49:509.226 GMT] logStreamInfo(): Kinesis Video Stream Info
[DEBUG] [03-06-2024 11:35:49:509.239 GMT] logStreamInfo(): 	Stream name: FacialVideoStream 
[DEBUG] [03-06-2024 11:35:49:509.242 GMT] logStreamInfo(): 	Streaming type: STREAMING_TYPE_REALTIME 
[DEBUG] [03-06-2024 11:35:49:509.245 GMT] logStreamInfo(): 	Content type: video/h264 
[DEBUG] [03-06-2024 11:35:49:509.248 GMT] logStreamInfo(): 	Max latency (100ns): 600000000
[DEBUG] [03-06-2024 11:35:49:509.251 GMT] logStreamInfo(): 	Fragment duration (100ns): 20000000
[DEBUG] [03-06-2024 11:35:49:509.254 GMT] logStreamInfo(): 	Key frame fragmentation: Yes
[DEBUG] [03-06-2024 11:35:49:509.257 GMT] logStreamInfo(): 	Use frame timecode: Yes
[DEBUG] [03-06-2024 11:35:49:509.259 GMT] logStreamInfo(): 	Absolute frame timecode: Yes
[DEBUG] [03-06-2024 11:35:49:509.290 GMT] logStreamInfo(): 	Nal adaptation flags: 0
[DEBUG] [03-06-2024 11:35:49:509.298 GMT] logStreamInfo(): 	Average bandwith (bps): 4194304
[DEBUG] [03-06-2024 11:35:49:509.301 GMT] logStreamInfo(): 	Framerate: 25
[DEBUG] [03-06-2024 11:35:49:509.309 GMT] logStreamInfo(): 	Buffer duration (100ns): 1200000000
[DEBUG] [03-06-2024 11:35:49:509.312 GMT] logStreamInfo(): 	Replay duration (100ns): 400000000
[DEBUG] [03-06-2024 11:35:49:509.324 GMT] logStreamInfo(): 	Connection Staleness duration (100ns): 600000000
[DEBUG] [03-06-2024 11:35:49:509.328 GMT] logStreamInfo(): 	Store Pressure Policy: 1
[DEBUG] [03-06-2024 11:35:49:509.331 GMT] logStreamInfo(): 	View Overflow Policy: 1
[DEBUG] [03-06-2024 11:35:49:509.339 GMT] logStreamInfo(): 	Segment UUID: NULL
[DEBUG] [03-06-2024 11:35:49:509.342 GMT] logStreamInfo(): 	Frame ordering mode: 0
[DEBUG] [03-06-2024 11:35:49:509.345 GMT] logStreamInfo(): Track list
[DEBUG] [03-06-2024 11:35:49:509.348 GMT] logStreamInfo(): 	Track id: 1
[DEBUG] [03-06-2024 11:35:49:509.351 GMT] logStreamInfo(): 	Track name: kinesis_video
[DEBUG] [03-06-2024 11:35:49:509.354 GMT] logStreamInfo(): 	Codec id: V_MPEG4/ISO/AVC
[DEBUG] [03-06-2024 11:35:49:509.361 GMT] logStreamInfo(): 	Track type: TRACK_INFO_TYPE_VIDEO
[DEBUG] [03-06-2024 11:35:49:509.364 GMT] logStreamInfo(): 	Track cpd: NULL
[DEBUG] [03-06-2024 11:35:49:510.724 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
[INFO ] [03-06-2024 11:35:50:538.442 GMT] writeHeaderCallback(): RequestId: 0fe69141-e974-40ef-a2d6-866c2ebdb6ab
[DEBUG] [03-06-2024 11:35:50:543.868 GMT] describeStreamCurlHandler(): DescribeStream API response: {"StreamInfo":{"CreationTime":1.716818676779E9,"DataRetentionInHours":2,"DeviceName":"Kinesis_Video_Device","IngestionConfiguration":null,"KmsKeyId":"arn:aws:kms:us-east-1:955834732421:alias/aws/kinesisvideo","MediaType":"video/h264","Status":"ACTIVE","StreamARN":"arn:aws:kinesisvideo:us-east-1:955834732421:stream/FacialVideoStream/1716818676779","StreamName":"FacialVideoStream","Version":"g7soBwlWuBMXDRAmbHEX"}}
[INFO ] [03-06-2024 11:35:50:555.377 GMT] describeStreamResultEvent(): Describe stream result event.
[DEBUG] [03-06-2024 11:35:50:555.977 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000000000000020
[INFO ] [03-06-2024 11:35:51:445.588 GMT] writeHeaderCallback(): RequestId: 88b2b706-f5c7-452f-a997-f003a993b391
[DEBUG] [03-06-2024 11:35:51:446.002 GMT] getStreamingEndpointCurlHandler(): GetStreamingEndpoint API response: {"DataEndpoint":"https://s-808fdb44.kinesisvideo.us-east-1.amazonaws.com"}
[... 82 lines cut ...]
[WARN ] [03-06-2024 11:36:17:284.405 GMT] continuousRetryStreamErrorReportHandler(): Reporting stream error. Errored timecode: 53220000 Status: 0x5200005e
[DEBUG] [03-06-2024 11:36:17:284.936 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000100, Next state: 0x0000000000000200
[DEBUG] [03-06-2024 11:36:17:285.258 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000200, Next state: 0x0000000000000001
[DEBUG] [03-06-2024 11:36:17:285.411 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
[INFO ] [03-06-2024 11:36:17:285.517 GMT] describeStreamResultEvent(): Describe stream result event.
[DEBUG] [03-06-2024 11:36:17:285.616 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000000000000020
[INFO ] [03-06-2024 11:36:17:285.663 GMT] getStreamingEndpointResultEvent(): Get streaming endpoint result event.
[DEBUG] [03-06-2024 11:36:17:285.695 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000020, Next state: 0x0000000000000010
[DEBUG] [03-06-2024 11:36:17:285.728 GMT] getStreamingTokenHandler invoked
[DEBUG] [03-06-2024 11:36:17:285.792 GMT] Refreshing credentials. Force refreshing: 1 Now time is: 1717414577285771847 Expiration: 1717418151
[INFO ] [03-06-2024 11:36:17:285.874 GMT] New credentials expiration is 1717418177
[INFO ] [03-06-2024 11:36:17:285.933 GMT] getStreamingTokenResultEvent(): Get streaming token result event.
[DEBUG] [03-06-2024 11:36:17:286.001 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000010, Next state: 0x0000000000000040
[DEBUG] [03-06-2024 11:36:17:286.065 GMT] streamReadyHandler invoked
[DEBUG] [03-06-2024 11:36:17:286.102 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000040, Next state: 0x0000000000000080
[INFO ] [03-06-2024 11:36:17:286.760 GMT] putStreamResultEvent(): Put stream result event. New upload handle 1
[DEBUG] [03-06-2024 11:36:17:297.453 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000080, Next state: 0x0000000000000100
[... cut ...]
[DEBUG] [03-06-2024 11:36:17:300.624 GMT] postReadCallback(): Wrote 0 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG] [03-06-2024 11:36:17:303.028 GMT] putStreamCurlHandler(): Network thread for Kinesis Video stream: FacialVideoStream with upload handle: 0 exited. http status: 200
[INFO ] [03-06-2024 11:36:18:129.248 GMT] writeHeaderCallback(): RequestId: f3bebbbc-03b4-35aa-ad69-174e7cc45707
[DEBUG] [03-06-2024 11:36:18:129.666 GMT] postReadCallback(): Pausing CURL read for upload handle: 1`

niyatim23, 2024-06-04:
Hi @tawiemadavz, can you try https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/issues/147#issuecomment-440732944 if it helps

tawiemadavz, 2024-06-05:
@niyatim23 i had seen it earlier, but i'm confused as to where on this command i should add "key-int-max=10"?

Original Command:

`gst-launch-1.0 rtspsrc location="rtsp://XXXXXX:XXXXX@IP/Streaming/Channels/101" short-header=TRUE ! rtph264depay ! h264parse ! video/x-h264, format=avc,alignment=au ! kvssink stream-name="FacialVideoStream"`

When i add "frame-timestamp=1":

`gst-launch-1.0 rtspsrc location="rtsp://XXXXXX:XXXXX@IP/Streaming/Channels/101" short-header=TRUE ! rtph264depay ! h264parse ! video/x-h264, format=avc,alignment=au ! kvssink stream-name="FacialVideoStream" frame-timestamp=1`

Output:

`WARNING: erroneous pipeline: no property "frame-timestamp" in element "kvssink"`

When i add the "key-int-max" from "x264enc":

`gst-launch-1.0 rtspsrc location="rtsp://XXXXXX:XXXXX@IP/Streaming/Channels/101" short-header=TRUE ! x264enc bframes=0 key-int-max=10 bitrate=512 ! video/x-h264, format=avc,alignment=au ! kvssink stream-name="FacialVideoStream" frame-timestamp=1`

Output:

`WARNING: from element /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0: Delayed linking failed.
Additional debug info:
subprojects/gstreamer/gst/parse/grammar.y(919): gst_parse_no_more_pads (): /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0:
failed delayed linking some pad of GstRTSPSrc named rtspsrc0 to some pad of GstX264Enc named x264enc0
ERROR: from element /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0/GstUDPSrc:udpsrc0: Internal data stream error.
`

So it does not seems to be working for me because i'm using "rtspsrc" instead of "ksvideosrc". Not sure if that matters? Have a limited understanding of gstreamer so any assistance RE how to put together the command correctly would be appreciated.

Thanks!

disa6302, 2024-06-17:
@tawiemadavz , 

Please follow the example here: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/docs/linux.md#running-the-gst-launch-10-command-to-start-streaming-from-a-rtsp-camera-source

We have some sample pipelines in our docs that could help you get started.