awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #1188: Stream video to your Kinesis video stream and view the live stream

himadrii27, 2024-06-17:
On running the command ./kvs_gstreamer_sample camera_37 rtsp://user:password@192.168.1.37:554/Streaming/Channels/102/
I am getting the error
 Creating IoT auth callbacks.
[ERROR] [17-06-2024 09:22:32:860.195 GMT] Unable to create Iot Credential provider. Error status: 0x4000000e
[ERROR] [17-06-2024 09:22:32:860.422 GMT] Failed to initialize kinesis video with an exception: Unable to create Iot Credential provider. Error status: 0x4000000e
 
Do you have any idea how to fix this?

disa6302, 2024-06-17:
@himadrii27 , 

These are some common situations the error could be thrown:
1. If the path provided to fstat is invalid (e.g., the file does not exist, or there is a typo in the path
2. If the process does not have the necessary permissions to access the file or directory 
3. An I/O error occurred while trying to retrieve the file status. This could be due to hardware issues or file system corruption

Could you confirm the path provided for the IoT certs is infact valid and has all the necessary permissions?

sirknightj, 2024-06-18:
You can also double-check the YouTube tutorial to see if you are missing anything: https://www.youtube.com/watch?v=YoOYTCD_v3Q, it shows running the sample with IoT certificate (X.509) credentials.

Note that for quicker testing, you can also use a standard iam credentials (export AWS_ACCESS_KEY_ID=...): https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/gs-send-data.html#send-data-run-samples

himadrii27, 2024-06-18:
> @himadrii27 ,
>
> These are some common situations the error could be thrown:
>
> 1. If the path provided to fstat is invalid (e.g., the file does not exist, or there is a typo in the path
> 2. If the [...]

himadrii27, 2024-06-18:
> You can also double-check the YouTube tutorial to see if you are missing anything: https://www.youtube.com/watch?v=YoOYTCD_v3Q, it shows running the sample with IoT certificate (X.509) credentials. [...]

you mean having all the certificates in cert file and then proceeding? yea, i already did that

sirknightj, 2024-11-26:
Do you have the `CA_CERT_PATH` cert path set? It should point to the `AmazonRootCA1.pem` file.
You can do `ls $CA_CERT_PATH` to confirm that the path is set correctly.

sirknightj, 2025-01-06:
Apologies for the typo of the env variable.

Please make sure that the file path is specified correctly in the `AWS_KVS_CACERT_PATH` environment variable, and also please double check the file paths of all of the IoT certificates:

```
export AWS_DEFAULT_REGION=us-west-2
export AWS_IOT_CORE_CREDENTIAL_ENDPOINT=/x/xxxxx.credentials.iot.us-west-2.amazonaws.com
export AWS_IOT_CORE_PRIVATE_KEY=xxxx-private.pem.key 
export AWS_IOT_CORE_ROLE_ALIAS=xxxx
export AWS_IOT_CORE_THING_NAME=xxxx
export AWS_IOT_CORE_CERT=/x/xxxx-certificate.pem.crt
export AWS_KVS_CACERT_PATH=/x/AmazonRootCA1.pem
```

closed 2025-01-06 by sirknightj