awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #1191: [BUG] KVS Fails to build due to Threadpool.c.o issue

antmakedev, 2024-06-18:
**Describe the bug**
When trying to build I keep hitting the same issue outlined below

**To Reproduce**
Steps to reproduce the behavior:
Download amazon-kinesis-video-streams-producer-sdk-cpp-3.4.1
Make build directory
Execute `cmake -DBUILD_GSTREAMER_PLUGIN=TRUE ..  `
Execute `make`
Get error

**Expected behavior**
Expect library to build

```
[ 17%] Building C object dependency/libkvscproducer/kvscproducer-src/dependency/libkvspic/kvspic-src/CMakeFiles/kvspicUtils.dir/src/utils/src/Threadpool.c.o
/scratch/amazon-kinesis-video-streams-producer-sdk-cpp-3.4.1/dependency/libkvscproducer/kvscproducer-src/dependency/libkvspic/kvspic-src/src/utils/src/Threadpool.c:81:79: error: incompatible pointer to integer conversion passing 'PThreadData' (aka 'struct __ThreadData *') to parameter of type 'UINT64' (aka 'unsigned long long') [-Wint-conversion]
                            if (stackQueueRemoveItem(pThreadpool->threadList, pThreadData) != STATUS_SUCCESS) {
                                                                              ^~~~~~~~~~~
/scratch/amazon-kinesis-video-streams-producer-sdk-cpp-3.4.1/dependency/libkvscproducer/kvscproducer-src/dependency/libkvspic/kvspic-src/src/utils/include/com/amazonaws/kinesis/video/utils/Include.h:591:59: note: passing argument to parameter here
PUBLIC_API STATUS stackQueueRemoveItem(PStackQueue, UINT64);
                                                          ^
/scratch/amazon-kinesis-video-streams-producer-sdk-cpp-3.4.1/dependency/libkvscproducer/kvscproducer-src/dependency/libkvspic/kvspic-src/src/utils/src/Threadpool.c:95:67: error: incompatible pointer to integer conversion passing 'PThreadData' (aka 'struct __ThreadData *') to parameter of type 'UINT64' (aka 'unsigned long long') [-Wint-conversion]
                if (stackQueueRemoveItem(pThreadpool->threadList, pThreadData) != STATUS_SUCCESS) {
                                                                  ^~~~~~~~~~~
/scratch/amazon-kinesis-video-streams-producer-sdk-cpp-3.4.1/dependency/libkvscproducer/kvscproducer-src/dependency/libkvspic/kvspic-src/src/utils/include/com/amazonaws/kinesis/video/utils/Include.h:591:59: note: passing argument to parameter here
PUBLIC_API STATUS stackQueueRemoveItem(PStackQueue, UINT64);
                                                          ^
/scratch/amazon-kinesis-video-streams-producer-sdk-cpp-3.4.1/dependency/libkvscproducer/kvscproducer-src/dependency/libkvspic/kvspic-src/src/utils/src/Threadpool.c:301:61: warning: incompatible pointer types passing 'PThreadData *' (aka 'struct __ThreadData **') to parameter of type 'PUINT64' (aka 'unsigned long long *') [-Wincompatible-pointer-types]
            retStatus = stackQueueIteratorGetItem(iterator, &item);
                                                            ^~~~~
/scratch/amazon-kinesis-video-streams-producer-sdk-cpp-3.4.1/dependency/libkvscproducer/kvscproducer-src/dependency/libkvspic/kvspic-src/src/utils/include/com/amazonaws/kinesis/video/utils/Include.h:636:72: note: passing argument to parameter here
PUBLIC_API STATUS stackQueueIteratorGetItem(StackQueueIterator, PUINT64);
                                                                       ^
/scratch/amazon-kinesis-video-streams-producer-sdk-cpp-3.4.1/dependency/libkvscproducer/kvscproducer-src/dependency/libkvspic/kvspic-src/src/utils/src/Threadpool.c:309:67: error: incompatible pointer to integer conversion passing 'PThreadData' (aka 'struct __ThreadData *') to parameter of type 'UINT64' (aka 'unsigned long long') [-Wint-conversion]
                if (stackQueueRemoveItem(pThreadpool->threadList, item) != STATUS_SUCCESS) {
                                                                  ^~~~
/scratch/amazon-kinesis-video-streams-producer-sdk-cpp-3.4.1/dependency/libkvscproducer/kvscproducer-src/dependency/libkvspic/kvspic-src/src/utils/include/com/amazonaws/kinesis/video/utils/Include.h:591:59: note: passing argument to parameter here
PUBLIC_API STATUS stackQueueRemoveItem(PStackQueue, UINT64);
                                                          ^
1 warning and 3 errors generated.
make[2]: *** [dependency/libkvscproducer/kvscproducer-src/dependency/libkvspic/kvspic-src/CMakeFiles/kvspicUtils.dir/src/utils/src/Threadpool.c.o] Error 1
make[1]: *** [dependency/libkvscproducer/kvscproducer-src/dependency/libkvspic/kvspic-src/CMakeFiles/kvspicUtils.dir/all] Error 2
make: *** [all] Error 2

```

**Platform (please complete the following information):**
 - MacOS Sonoma 14.5 (23F79)
 - Processor: Apple M1 Pro

I get the same issue if I pull master (49fa6cbb905958726fe54d98b837c3edf895025d) as well.

disa6302, 2024-06-19:
@antmakedev ,

Thanks for the report. The issue is fixed but not yet released. As a workaround, could you try passing in `-DCMAKE_C_FLAGS="-Wno-int-conversion -Wno-incompatible-pointer-types"` with your cmake command?

antmakedev, 2024-06-20:
Thanks @disa6302 that worked :D

nbaillie, 2024-06-21:
thanks for posting this worked for me also

sirknightj, 2024-06-22:
Will keep this open until next release for ease of locating resolution & avoiding duplicates.

closed 2025-02-25 by sirknightj