awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #1194: kvs_gstreamer_sample fails to run on Raspberry Pi bookworm

vmalan, 2024-07-08:
**Logging**
 ./kvs_gstreamer_sample buoy_van -w 640 -h 480
[INFO ] [08-07-2024 17:13:44:269.765 GMT] Using region: us-west-2
[INFO ] [08-07-2024 17:13:44:270.073 GMT] Using aws credentials for Kinesis Video Streams
[INFO ] [08-07-2024 17:13:44:270.152 GMT] No session token was detected.
[INFO ] [08-07-2024 17:13:44:275.888 GMT] createKinesisVideoClient(): Creating Kinesis Video Client
[WARN ] [08-07-2024 17:13:44:276.209 GMT] fixupClientInfo(): Connection timeout is invalid...setting to default
[WARN ] [08-07-2024 17:13:44:276.288 GMT] fixupClientInfo(): Completion timeout is invalid...setting to default
[INFO ] [08-07-2024 17:13:44:276.428 GMT] heapInitialize(): Initializing native heap with limit size 134217728, spill ratio 0% and flags 0x00000001
[INFO ] [08-07-2024 17:13:44:276.500 GMT] heapInitialize(): Creating AIV heap.
[INFO ] [08-07-2024 17:13:44:276.640 GMT] heapInitialize(): Heap is initialized OK
[DEBUG] [08-07-2024 17:13:44:277.122 GMT] getSecurityTokenHandler invoked
[DEBUG] [08-07-2024 17:13:44:277.319 GMT] Refreshing credentials. Force refreshing: 0 Now time is: 1720458824277275649 Expiration: 0
[INFO ] [08-07-2024 17:13:44:277.460 GMT] New credentials expiration is 1720462424
[INFO ] [08-07-2024 17:13:44:277.632 GMT] createDeviceResultEvent(): Create device result event.
[DEBUG] [08-07-2024 17:13:44:277.766 GMT] clientReadyHandler invoked
[DEBUG] [08-07-2024 17:13:44:277.880 GMT] Client is ready
[INFO ] [08-07-2024 17:13:44:278.068 GMT] Creating Kinesis Video Stream buoy_van
[INFO ] [08-07-2024 17:13:44:278.201 GMT] createKinesisVideoStream(): Creating Kinesis Video Stream.
[INFO ] [08-07-2024 17:13:44:278.322 GMT] logStreamInfo(): SDK version: 0b5c45b0c304c91ccce98c1e3ddf71ff5bc7af75
[DEBUG] [08-07-2024 17:13:44:278.432 GMT] logStreamInfo(): Kinesis Video Stream Info
[DEBUG] [08-07-2024 17:13:44:278.540 GMT] logStreamInfo():      Stream name: buoy_van
[DEBUG] [08-07-2024 17:13:44:278.674 GMT] logStreamInfo():      Streaming type: STREAMING_TYPE_REALTIME
[DEBUG] [08-07-2024 17:13:44:278.810 GMT] logStreamInfo():      Content type: video/h264
[DEBUG] [08-07-2024 17:13:44:278.947 GMT] logStreamInfo():      Max latency (100ns): 600000000
[DEBUG] [08-07-2024 17:13:44:279.083 GMT] logStreamInfo():      Fragment duration (100ns): 20000000
[DEBUG] [08-07-2024 17:13:44:279.217 GMT] logStreamInfo():      Key frame fragmentation: Yes
[DEBUG] [08-07-2024 17:13:44:279.407 GMT] logStreamInfo():      Use frame timecodes: Yes
[DEBUG] [08-07-2024 17:13:44:279.576 GMT] logStreamInfo():      Absolute frame timecodes: Yes
[DEBUG] [08-07-2024 17:13:44:279.720 GMT] logStreamInfo():      Nal adaptation flags: 0
[DEBUG] [08-07-2024 17:13:44:279.853 GMT] logStreamInfo():      Average bandwidth (bps): 4194304
[DEBUG] [08-07-2024 17:13:44:279.986 GMT] logStreamInfo():      Framerate: 25
[DEBUG] [08-07-2024 17:13:44:280.120 GMT] logStreamInfo():      Buffer duration (100ns): 1200000000
[DEBUG] [08-07-2024 17:13:44:280.254 GMT] logStreamInfo():      Replay duration (100ns): 400000000
[DEBUG] [08-07-2024 17:13:44:280.390 GMT] logStreamInfo():      Connection Staleness duration (100ns): 600000000
[DEBUG] [08-07-2024 17:13:44:280.523 GMT] logStreamInfo():      Store Pressure Policy: 1
[DEBUG] [08-07-2024 17:13:44:280.658 GMT] logStreamInfo():      View Overflow Policy: 1
[DEBUG] [08-07-2024 17:13:44:280.794 GMT] logStreamInfo():      Allow stream creation: Yes
[DEBUG] [08-07-2024 17:13:44:280.935 GMT] logStreamInfo():      Segment UUID: NULL
[DEBUG] [08-07-2024 17:13:44:281.068 GMT] logStreamInfo():      Frame ordering mode: 0
[DEBUG] [08-07-2024 17:13:44:281.200 GMT] logStreamInfo(): Track list
[DEBUG] [08-07-2024 17:13:44:281.302 GMT] logStreamInfo():      Track id: 1
[DEBUG] [08-07-2024 17:13:44:281.436 GMT] logStreamInfo():      Track name: kinesis_video
[DEBUG] [08-07-2024 17:13:44:281.571 GMT] logStreamInfo():      Codec id: V_MPEG4/ISO/AVC
[DEBUG] [08-07-2024 17:13:44:281.708 GMT] logStreamInfo():      Track type: TRACK_INFO_TYPE_VIDEO
[DEBUG] [08-07-2024 17:13:44:281.840 GMT] logStreamInfo():      Track cpd: NULL
[DEBUG] [08-07-2024 17:13:44:282.522 GMT] setRequestHeader(): Appending header to request: user-agent -> AWS-SDK-KVS/1.5.1 GCC/12.2.0 Linux/6.6.31+rpt-rpi-v8 aarch64 CPPSDK
[DEBUG] [08-07-2024 17:13:44:283.213 GMT] setRequestHeader(): Appending header to request: host -> kinesisvideo.us-west-2.amazonaws.com/describeStream
[DEBUG] [08-07-2024 17:13:44:283.515 GMT] setRequestHeader(): Appending header to request: X-Amz-Date -> 20240708T171344Z
[DEBUG] [08-07-2024 17:13:44:283.653 GMT] setRequestHeader(): Appending header to request: content-type -> application/json
[DEBUG] [08-07-2024 17:13:44:283.779 GMT] setRequestHeader(): Appending header to request: content-length -> 29
[DEBUG] [08-07-2024 17:13:44:288.946 GMT] setRequestHeader(): Appending header to request: Authorization -> AWS4-HMAC-SHA256 Credential=AKIAU6GDYDZT3Z74A6PU/20240708/us-west-2/kinesisvideo/aws4_request, SignedHeaders=host;user-agent;x-amz-date, Signature=efad111ca92c6b70a0a1f7adc3fb01fe9c92b4b0c2cf9daf40668d6bb2d24d35
[INFO ] [08-07-2024 17:13:44:569.187 GMT] writeHeaderCallback(): RequestId: 5cd6b6f2-89ab-44dd-8003-04e4bf37cb08
[DEBUG] [08-07-2024 17:13:44:569.422 GMT] describeStreamCurlHandler(): [buoy_van] DescribeStream API response: {"StreamInfo":{"CreationTime":1.720113870427E9,"DataRetentionInHours":24,"DeviceName":null,"IngestionConfiguration":null,"KmsKeyId":"arn:aws:kms:us-west-2:339712941671:alias/aws/kinesisvideo","MediaType":null,"Status":"ACTIVE","StreamARN":"arn:aws:kinesisvideo:us-west-2:339712941671:stream/buoy_van/1720113870427","StreamName":"buoy_van","Version":"Uv2FEnOAdn8ZUhNk5R5u"}}
[INFO ] [08-07-2024 17:13:44:571.670 GMT] describeStreamResultEvent(): Describe stream result event.
[WARN ] [08-07-2024 17:13:44:571.848 GMT] describeStreamResult(): Retention period returned from the DescribeStream call doesn't match the one specified in the StreamInfo
[WARN ] [08-07-2024 17:13:44:571.898 GMT] describeStreamResult(): Content type returned from the DescribeStream(null) call doesn't match the one specified in the StreamInfo(video/h264)
[WARN ] [08-07-2024 17:13:44:571.939 GMT] describeStreamResult(): [buoy_van] Retention period returned from the DescribeStream call doesn't match the one specified in the StreamInfo
[WARN ] [08-07-2024 17:13:44:571.978 GMT] describeStreamResult(): [buoy_van] Content type returned from the DescribeStream call doesn't match the one specified in the StreamInfo
[DEBUG] [08-07-2024 17:13:44:572.731 GMT] setRequestHeader(): Appending header to request: user-agent -> AWS-SDK-KVS/1.5.1 GCC/12.2.0 Linux/6.6.31+rpt-rpi-v8 aarch64 CPPSDK
[DEBUG] [08-07-2024 17:13:44:573.060 GMT] setRequestHeader(): Appending header to request: host -> kinesisvideo.us-west-2.amazonaws.com/getDataEndpoint
[DEBUG] [08-07-2024 17:13:44:573.133 GMT] setRequestHeader(): Appending header to request: X-Amz-Date -> 20240708T171344Z
[DEBUG] [08-07-2024 17:13:44:573.161 GMT] setRequestHeader(): Appending header to request: content-type -> application/json
[DEBUG] [08-07-2024 17:13:44:573.193 GMT] setRequestHeader(): Appending header to request: content-length -> 54
[DEBUG] [08-07-2024 17:13:44:573.379 GMT] setRequestHeader(): Appending header to request: Authorization -> AWS4-HMAC-SHA256 Credential=AKIAU6GDYDZT3Z74A6PU/20240708/us-west-2/kinesisvideo/aws4_request, SignedHeaders=host;user-agent;x-amz-date, Signature=613dba891cb8c6672a30aae19b7205545b169aad7018419179cd7ee217e909c8
[INFO ] [08-07-2024 17:13:44:847.270 GMT] writeHeaderCallback(): RequestId: a09ea95e-4124-4dc9-b413-73c8a47d06e2
[DEBUG] [08-07-2024 17:13:44:847.453 GMT] getStreamingEndpointCurlHandler(): [buoy_van] GetStreamingEndpoint API response: {"DataEndpoint":"https://s-8451b5cd.kinesisvideo.us-west-2.amazonaws.com"}
[INFO ] [08-07-2024 17:13:44:849.628 GMT] getStreamingEndpointResultEvent(): Get streaming endpoint result event.
[DEBUG] [08-07-2024 17:13:44:849.761 GMT] getStreamingTokenHandler invoked
[DEBUG] [08-07-2024 17:13:44:849.800 GMT] Refreshing credentials. Force refreshing: 1 Now time is: 1720458824849791779 Expiration: 1720462424
[... 35 lines cut ...]
[DEBUG] [08-07-2024 17:13:45:383.957 GMT] curlApiCallbacksShutdownActiveRequests(): pActiveRequests hashtable is empty
[DEBUG] [08-07-2024 17:13:45:784.209 GMT] freeKinesisVideoClientInternal(): Total allocated memory 0
[INFO ] [08-07-2024 17:13:45:784.295 GMT] Completed freeing client
[... cut ...]
2. execute kvs_gstreamer_sample <stream-name> -w <width> -h <height>
3. see sad results

**Expected behavior**
after step 2 above, hope to see video being collected at AWS.
is this possible v4l2src vs libcamerasrc ???

**SDK version number**
Include the SDK version you are running. If it is a specific commit on master branch, include that
branches master and develop as of 7/5/2024

**Platform (please complete the following information):**
 - OS: Raspberry Pi bookworm
Linux 6.6.31+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.31-1+rpt1 (2024-05-29) aarch64 GNU/Linux

sirknightj, 2024-11-26:
It appears that you're using the raspberry pi camera and not a USB webcam?

Please try the instructions here: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/discussions/1130#discussioncomment-11378187 and see if it resolves your issue.

sirknightj, 2025-01-06:
The RPI documentation has been updated with the updated instructions. If you encounter further issues with the documentation, please leave feedback on the page.

closed 2025-01-06 by sirknightj