awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #1196: [BUG] Make resulting in ncompatible pointer to integer conversion  Errors

DerYohmen, 2024-07-16:
**Describe the bug**
Im trying to follow the documentation provided on my Mac with Sonoma 14.5 and Apple M2 Pro Chip.
After running 
`cmake .. -DBUILD_GSTREAMER_PLUGIN=ON -DBUILD_JNI=TRUE` i get the following error:
`
make[2]: *** [build/src/project_log4cplus-stamp/project_log4cplus-configure] Error 77
make[1]: *** [CMakeFiles/project_log4cplus.dir/all] Error 2
make: *** [all] Error 2
CMake Error at CMake/Utilities.cmake:99 (message):
  CMake step for liblog4cplus failed: 2
Call Stack (most recent call first):
  CMakeLists.txt:85 (build_dependency)
`

With config.log showing this errors:
`
configure:3647: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc -V >&5
clang: error: argument to '-V' is missing (expected 1 value)
clang: error: no input files
configure:3658: $? = 1
configure:3647: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc -qversion >&5
clang: error: unknown argument '-qversion'; did you mean '--version'?
clang: error: no input files
configure:3658: $? = 1
configure:3678: checking whether the C compiler works
configure:3700: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc    conftest.c  >&5
ld: library 'System' not found
clang: error: linker command failed with exit code 1 (use -v to see invocation)
`
By simply running the cmake command again this error wont show again and the command finishes without errors.

Afterwards im trying to run "make" 
Wich results in the following error:
`
/amazon-kinesis-video-streams-producer-sdk-cpp/dependency/libkvscproducer/kvscproducer-src/dependency/libkvspic/kvspic-src/src/utils/src/Threadpool.c:81:79: error: incompatible pointer to integer conversion passing 'PThreadData' (aka 'struct __ThreadData *') to parameter of type 'UINT64' (aka 'unsigned long long') [-Wint-conversion]
                            if (stackQueueRemoveItem(pThreadpool->threadList, pThreadData) != STATUS_SUCCESS) {
                                                                              ^~~~~~~~~~~
/amazon-kinesis-video-streams-producer-sdk-cpp/dependency/libkvscproducer/kvscproducer-src/dependency/libkvspic/kvspic-src/src/utils/include/com/amazonaws/kinesis/video/utils/Include.h:591:59: note: passing argument to parameter here
PUBLIC_API STATUS stackQueueRemoveItem(PStackQueue, UINT64);
                                                          ^
/amazon-kinesis-video-streams-producer-sdk-cpp/dependency/libkvscproducer/kvscproducer-src/dependency/libkvspic/kvspic-src/src/utils/src/Threadpool.c:95:67: error: incompatible pointer to integer conversion passing 'PThreadData' (aka 'struct __ThreadData *') to parameter of type 'UINT64' (aka 'unsigned long long') [-Wint-conversion]
                if (stackQueueRemoveItem(pThreadpool->threadList, pThreadData) != STATUS_SUCCESS) {
                                                                  ^~~~~~~~~~~
/amazon-kinesis-video-streams-producer-sdk-cpp/dependency/libkvscproducer/kvscproducer-src/dependency/libkvspic/kvspic-src/src/utils/include/com/amazonaws/kinesis/video/utils/Include.h:591:59: note: passing argument to parameter here
PUBLIC_API STATUS stackQueueRemoveItem(PStackQueue, UINT64);
                                                          ^
/amazon-kinesis-video-streams-producer-sdk-cpp/dependency/libkvscproducer/kvscproducer-src/dependency/libkvspic/kvspic-src/src/utils/src/Threadpool.c:301:61: warning: incompatible pointer types passing 'PThreadData *' (aka 'struct __ThreadData **') to parameter of type 'PUINT64' (aka 'unsigned long long *') [-Wincompatible-pointer-types]
            retStatus = stackQueueIteratorGetItem(iterator, &item);
                                                            ^~~~~
/amazon-kinesis-video-streams-producer-sdk-cpp/dependency/libkvscproducer/kvscproducer-src/dependency/libkvspic/kvspic-src/src/utils/include/com/amazonaws/kinesis/video/utils/Include.h:636:72: note: passing argument to parameter here
PUBLIC_API STATUS stackQueueIteratorGetItem(StackQueueIterator, PUINT64);
                                                                       ^
/amazon-kinesis-video-streams-producer-sdk-cpp/dependency/libkvscproducer/kvscproducer-src/dependency/libkvspic/kvspic-src/src/utils/src/Threadpool.c:309:67: error: incompatible pointer to integer conversion passing 'PThreadData' (aka 'struct __ThreadData *') to parameter of type 'UINT64' (aka 'unsigned long long') [-Wint-conversion]
                if (stackQueueRemoveItem(pThreadpool->threadList, item) != STATUS_SUCCESS) {
                                                                  ^~~~
/amazon-kinesis-video-streams-producer-sdk-cpp/dependency/libkvscproducer/kvscproducer-src/dependency/libkvspic/kvspic-src/src/utils/include/com/amazonaws/kinesis/video/utils/Include.h:591:59: note: passing argument to parameter here
PUBLIC_API STATUS stackQueueRemoveItem(PStackQueue, UINT64);

`
**To Reproduce**
Steps to reproduce the behavior:
Its probably important to reproduce this on a Mac with the same chip and OS.
1. Follow the Instructions from the repo https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp
2. Use cmake .. -DBUILD_GSTREAMER_PLUGIN=ON -DBUILD_JNI=TRUE
3. Run cmake again after the error
4. Try to make
5. See error

**Platform (please complete the following information):**
 - OS: MAC
 - Processor: Apple M2 Pro

Thanks for any help!

stefankiesz, 2024-07-16:
Hi Yannick, can you please try and report back whether this issue persists when building the develop branch?

stefankiesz, 2024-07-16:
And what is the compiler (looks like it's clang) and version this is building with?

DerYohmen, 2024-07-17:
Hi Stefan,
first of all thanks for answering so quickly!

When i try the develop branch i get exactly the same issue.
My Compiler is:

Apple clang version 15.0.0 (clang-1500.3.9.4)
Target: arm64-apple-darwin23.5.0
Thread model: posix

Thanks for your help!

DerYohmen, 2024-07-17:
I just ran it again and now it works!
Thanks for your help!

stefankiesz, 2024-07-18:
No problem at all, glad to hear it's working!

We've had similar behavior to this in the past where we need to run `cmake` twice in order for the build to succeed. I'd like to flesh this out for good if possible. Do add more info here if you have any more details on what you tried and what ended up working. Much thanks.