awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #1197: [BUG] globalThreadSleep linkerror

thomas-roos, 2024-07-26:
When building shared version. This is missing
```
Index: git/CMakeLists.txt
===================================================================
--- git.orig/CMakeLists.txt
+++ git/CMakeLists.txt
@@ -221,7 +221,7 @@ if(BUILD_GSTREAMER_PLUGIN)
   target_link_libraries(gstkvssink PRIVATE ${GST_APP_LIBRARIES} KinesisVideoProducer)

   add_executable(kvssink_gstreamer_sample samples/kvssink_gstreamer_sample.cpp)
-  target_link_libraries(kvssink_gstreamer_sample ${GST_APP_LIBRARIES} KinesisVideoProducer)
+  target_link_libraries(kvssink_gstreamer_sample ${GST_APP_LIBRARIES} KinesisVideoProducer kvspic)

   add_executable(kvs_gstreamer_sample samples/kvs_gstreamer_sample.cpp)
   target_link_libraries(kvs_gstreamer_sample ${GST_APP_LIBRARIES} KinesisVideoProducer kvspic)
```

stefankiesz, 2024-11-26:
Hi @thomas-roos, 

In Producer C, we publicly link `kvspic` to `cproducer`:

```
target_link_libraries(cproducer PUBLIC kvsCommonCurl kvspic)
```

And here in Producer C++, `cproducer` is publicly linked to `KinesisVideoProducer`:

```
target_link_libraries(
  KinesisVideoProducer
  PUBLIC kvsCommonCurl
         cproducer
         ${Log4cplus}
         ${LIBCURL_LIBRARIES})
```
So it seems linking both `KinesisVideoProducer` and `kvspic` to the sample would be redundant. Did you experience any issue that was resolved with this change?

thomas-roos, 2024-11-27:
Yes, getting this error: 
[error.txt](https://github.com/user-attachments/files/17933134/error.txt)
When building without this patch here: https://github.com/aws4embeddedlinux/meta-aws/blob/8575c18e1d18caa44777b7b645a025dbd22f10eb/recipes-sdk/amazon-kvs-producer-sdk/amazon-kvs-producer-sdk-cpp_3.4.2.bb#L19

unicornss, 2026-01-07:
```
FROM ubuntu:22.04

ARG BUILD_STATIC=OFF

RUN apt-get update && apt-get install -y \
    build-essential \
    cmake \
    git \
    libssl-dev \
    libcurl4-openssl-dev \
    liblog4cplus-dev \
    libgstreamer1.0-dev \
    libgstreamer-plugins-base1.0-dev \
    pkg-config \
    m4 \
    autoconf \
    automake \
    libtool \
    && rm -rf /var/lib/apt/lists/*

WORKDIR /build

RUN git clone --recursive https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp.git

WORKDIR /build/amazon-kinesis-video-streams-producer-sdk-cpp

# Apply the CMake change to link kvspic library
RUN sed -i 's/target_link_libraries(kvssink_gstreamer_sample ${GST_APP_LIBRARIES} KinesisVideoProducer)/target_link_libraries(kvssink_gstreamer_sample ${GST_APP_LIBRARIES} KinesisVideoProducer kvspic)/' CMakeLists.txt

# Verify the change was applied
RUN grep -n "target_link_libraries.*kvssink_gstreamer_sample" CMakeLists.txt

RUN mkdir build && cd build && \
    cmake .. -DBUILD_GSTREAMER_PLUGIN=ON -DBUILD_STATIC=${BUILD_STATIC} -DBUILD_DEPENDENCIES=OFF && \
    make

CMD ["./build/samples/kvssink_gstreamer_sample"]
```

-----
```

#11 [8/8] RUN mkdir build && cd build &&     cmake .. -DBUILD_GSTREAMER_PLUGIN=ON -DBUILD_STATIC=OFF -DBUILD_DEPENDENCIES=OFF &&     make

#12 exporting to image
#12 exporting layers done
#12 exporting manifest sha256:f37a7f729d693002c004d15798b57a524f88d43098b967ff3bd9e599feb2cd0d done
#12 exporting config sha256:da2991511f8748b986516bd4359d79a6bf69b11a4ad408a8b873399a91830a57 done
#12 exporting attestation manifest sha256:f7e07182afa3569fb6d9d0944e5773a311a1c53631eb2eac661f86071e30cde9 done
#12 exporting manifest list sha256:e2f743712ca52dddb6a381ee6bc89f7544b16a0187dd9e8f31ddfce41733822a done
#12 naming to docker.io/library/kvs-test:latest done
#12 unpacking to docker.io/library/kvs-test:latest done
#12 DONE 0.0s
```
-----

```
 docker build --build-arg BUILD_STATIC=ON -t kvs-static . 2>&1 | tail -20 

#11 19.20 [ 97%] Built target kvs_gstreamer_multistream_sample
#11 19.21 [ 97%] Building CXX object CMakeFiles/kvs_gstreamer_audio_video_sample.dir/samples/kvs_gstreamer_audio_video_sample.cpp.o
#11 20.12 [ 98%] Linking CXX executable kvs_gstreamer_audio_video_sample
#11 20.25 [ 98%] Built target kvs_gstreamer_audio_video_sample
#11 20.25 [ 98%] Building CXX object CMakeFiles/kvs_gstreamer_file_uploader_sample.dir/samples/kvs_gstreamer_file_uploader_sample.cpp.o
#11 20.44 [100%] Linking CXX executable kvs_gstreamer_file_uploader_sample
#11 20.51 [100%] Built target kvs_gstreamer_file_uploader_sample
#11 DONE 20.6s
```

thomas-roos, 2026-01-08:
I was looking again a few hours in the issue - I think the issue is cause of another [patch](https://github.com/aws4embeddedlinux/meta-aws/blob/master/recipes-sdk/amazon-kvs-producer-sdk/amazon-kvs-producer-sdk-cpp/001-amazon-kvs-producer-sdk-cpp-deps.patch) fixing the 

BUILD_DEPENDENCIES cmake option, as this is not disable building of Internal dependencies, which in Yocto are build separate.
And my guess is then that the transitive deps are not correctly added.

The closed I can reproduce is with this Dockerfile:

```
FROM ubuntu:22.04

ARG BUILD_STATIC=OFF

RUN apt-get update && apt-get install -y \
    build-essential \
    cmake \
    git \
    libssl-dev \
    libcurl4-openssl-dev \
    liblog4cplus-dev \
    libgstreamer1.0-dev \
    libgstreamer-plugins-base1.0-dev \
    pkg-config \
    m4 \
    autoconf \
    automake \
    libtool \
    && rm -rf /var/lib/apt/lists/*

WORKDIR /build

# Clone and build Amazon KVS PIC explicitly
RUN git clone https://github.com/awslabs/amazon-kinesis-video-streams-pic.git
WORKDIR /build/amazon-kinesis-video-streams-pic
RUN mkdir build && cd build && \
    cmake .. -DBUILD_STATIC=${BUILD_STATIC} -DBUILD_SHARED_LIBS=ON -DCMAKE_C_FLAGS="-Wno-error=implicit-function-declaration" && \
    make && make install

# Clone and build Amazon KVS Producer SDK C explicitly
WORKDIR /build
RUN git clone https://github.com/awslabs/amazon-kinesis-video-streams-producer-c.git
WORKDIR /build/amazon-kinesis-video-streams-producer-c
RUN mkdir build && cd build && \
    cmake .. -DBUILD_DEPENDENCIES=OFF -DBUILD_STATIC=${BUILD_STATIC} -DBUILD_SHARED_LIBS=ON -DCMAKE_C_FLAGS="-Wno-error=implicit-function-declaration" && \
    make && make install

# Clone the C++ SDK (non-recursive, no submodules)
WORKDIR /build
RUN git clone https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp.git && \
    cd amazon-kinesis-video-streams-producer-sdk-cpp && \
    git checkout 7acf272672ced72bbbad85fe39d366deca57280e

WORKDIR /build/amazon-kinesis-video-streams-producer-sdk-cpp

# Apply the proper patch to fix BUILD_DEPENDENCIES=OFF
RUN sed -i '/CACHE BOOL "Build ProducerC with CURL Support" FORCE)/a if(BUILD_DEPENDENCIES)' CMakeLists.txt && \
    sed -i '/add_subdirectory.*kvscproducer-src.*EXCLUDE_FROM_ALL)/a endif()' CMakeLists.txt

# Apply the CMake change to link kvspic library
RUN sed -i 's/target_link_libraries(kvssink_gstreamer_sample ${GST_APP_LIBRARIES} KinesisVideoProducer)/target_link_libraries(kvssink_gstreamer_sample ${GST_APP_LIBRARIES} KinesisVideoProducer kvspic)/' CMakeLists.txt

# Verify the change was applied
RUN grep -n "target_link_libraries.*kvssink_gstreamer_sample" CMakeLists.txt

RUN mkdir build && cd build && \
    cmake .. -DBUILD_GSTREAMER_PLUGIN=ON -DBUILD_STATIC=${BUILD_STATIC} -DBUILD_SHARED_LIBS=ON -DBUILD_DEPENDENCIES=OFF && \
    make

CMD ["./build/samples/kvssink_gstreamer_sample"]

```

which works fine, but commenting out the kvspic linking it fails:

```
docker build --build-arg BUILD_STATIC=OFF  .

[ 70%] Building CXX object CMakeFiles/kvssink_gstreamer_sample.dir/samples/kvssink_gstreamer_sample.cpp.o
[ 73%] Linking CXX executable kvssink_gstreamer_sample
/usr/bin/ld: CMakeFiles/kvssink_gstreamer_sample.dir/samples/kvssink_gstreamer_sample.cpp.o: undefined reference to symbol 'strtoi32'
/usr/bin/ld: /usr/local/lib/libkvspicUtils.so.0: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/kvssink_gstreamer_sample.dir/build.make:99: kvssink_gstreamer_sample] Error 1
make[1]: *** [CMakeFiles/Makefile2:147: CMakeFiles/kvssink_gstreamer_sample.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
The command '/bin/sh -c mkdir build && cd build &&     cmake .. -DBUILD_GSTREAMER_PLUGIN=ON -DBUILD_STATIC=${BUILD_STATIC} -DBUILD_SHARED_LIBS=ON -DBUILD_DEPENDENCIES=OFF &&     make' returned a non-zero code: 2
```

I'm ok with clossing this issue as "not applicable - yocto specific". But ofc adding that explicit linking of kvspic would remove some tech dept on the yocto side to carry on the patch forever.

thomas-roos, 2026-01-14:
thank you -> https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/pull/1277