awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #1199: [QUESTION] kvssink is taking 2 minutes to shut down after EOS is sent to the Gstreamer pipeline

bkovacs-orca, 2024-07-29:
I have a Gstreamer pipeline that ends with `kvssink`, and sometimes it is taking 2 minutes to shut down after EOS is sent to the Gstreamer pipeline.
In what situations is this possible? What can be the reason for this?

** Logging **
I don't have logs yet, because I don't have an easy way to reproduce this case, it happens sporadically.

** Any design considerations/constraints **
Using the following Gstreamer pipeline: 
```
appsrc name=appsrcvideo block=false leaky-type=downstream max-bytes=5000000 ! h264parse name=parser ! kvssink name=kvssink stream-name=X storage-size=512 streaming-type=2 file-start-time=0 retention-period=1 access-key=X secret-key=X aws-region=us-west-2 stream-tags=X  appsrc name=appsrcaudio block=false leaky-type=downstream max-bytes=5000000 ! audio/mpeg,mpegversion=4,rate=(int)48000,channels=1,stream-format=(string)raw,codec_data=(buffer)1188 ! aacparse ! queue ! kvssink.
```

stefankiesz, 2024-07-29:
Hi, can you please expand on how you are observing that an EOS is sent and what you consider as a `kvssink` shutdown?

bkovacs-orca, 2024-07-30:
Hey Stefan,

Thanks for the reply. I send EOS with `gst_element_send_event(pipeline, gst_event_new_eos());`, so I know for a fact it's sent, I don't observe it any other way.

I consider the pipeline has shut down, including `kvssink` after this call returns, but it's stuck here:
```
  gst_element_set_state(pipeline, GST_STATE_NULL);
```

bkovacs-orca, 2024-08-07:
Following up on this. Do you need any more information to debug this? Thank you!

stefankiesz, 2024-08-08:
Hey @bkovacs-orca, I'll be looking into this issue with dedicated time starting next week, apologies for the delay.

stefankiesz, 2024-08-13:
@bkovacs-orca, it's difficult to assess the shutdown delay as originating from `kvssink` solely based on that it is present in your pipeline. Can you please provide logs when they become available to you and if possible run some control experiments with `kvssink` replaced with other sink elements to see if this behavior persists.

closed 2024-09-03 by niyatim23