awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #1203: [QUESTION]How can I perform kvssink in mac without installing dependencies?

testsys-varadhu, 2024-08-29:
I am creating an app in mac which needs to stream the feed to awskvs using kvssink. My limitation is that I cannot install anything in the end user machine. Is there a way I could achieve my requirement with the current setup? Thanks.

stefankiesz, 2024-09-03:
Hi Varadharajan, can you please specify what you are asking? How is having to install dependencies affecting your use of kvssink? Also have you tried using a static build for use case with BUILD_STATIC enabled?

testsys-varadhu, 2024-09-09:
@stefankiesz -- In my POC, I was able to install the dependencies using home brew and I was able to set the 'GST_PLUGIN_PATH' environment variable so it works in my local machine, whereas in the end user machine we will not have the privilege to install anything or to set the environment variables. I want to bundle the GStreamer and necessary libraries within my app bundle and need to load it during the execution of my app. After I include GStreamer.framework when i do kvssink = gst_element_factory_make("kvssink", "sink"); the sink is not getting initialized, and the variable is still null.

stefankiesz, 2024-09-13:
Here are the two options I've come to for this case.
1. Place the _libgstkvssink.so_ plugin file into the default GStreamer plugin directory for your system.
2. Programmatically set the plugin to GStreamer using `gst_registry_add_plugin`.

I tested out the first option on Mac with success. To recreate:
1. Don't set GST_PLUGIN_PATH. Can run `unset GST_PLUGIN_PATH` to be sure.
2. After building kvssink, copy the outputted library file into the appropriate GStreamer directory: `cp libgstkvssink.so /opt/homebrew/lib/gstreamer-1.0`.
3. Run the kvssink sample app per usual.

stefankiesz, 2024-10-07:
Closing assuming resolved.

closed 2024-10-07 by stefankiesz