awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #1215: [QUESTION] AWS STREAM Stopping in middle

srivatsavavuppala, 2024-12-17:
`[INFO ] [17-12-2024 17:17:15:959.963 GMT] Using region: us-east-1

[INFO ] [17-12-2024 17:17:15:960.015 GMT] Using aws credentials for Kinesis Video Streams

[INFO ] [17-12-2024 17:17:15:960.023 GMT] Session token detected.

[INFO ] [17-12-2024 17:17:15:960.772 GMT] createKinesisVideoClient(): Creating Kinesis Video Client
[WARN ] [17-12-2024 17:17:15:960.805 GMT] fixupClientInfo(): Connection timeout is invalid...setting to default
[WARN ] [17-12-2024 17:17:15:960.812 GMT] fixupClientInfo(): Completion timeout is invalid...setting to default
[INFO ] [17-12-2024 17:17:15:960.827 GMT] heapInitialize(): Initializing native heap with limit size 134217728, spill ratio 0% and flags 0x00000001
[INFO ] [17-12-2024 17:17:15:960.833 GMT] heapInitialize(): Creating AIV heap.
[INFO ] [17-12-2024 17:17:15:960.861 GMT] heapInitialize(): Heap is initialized OK
[DEBUG] [17-12-2024 17:17:15:960.957 GMT] getSecurityTokenHandler invoked
[DEBUG] [17-12-2024 17:17:15:960.981 GMT] Refreshing credentials. Force refreshing: 0 Now time is: 1734455835960973962 Expiration: 0
[INFO ] [17-12-2024 17:17:15:960.994 GMT] New credentials expiration is 1734459435
[INFO ] [17-12-2024 17:17:15:961.020 GMT] createDeviceResultEvent(): Create device result event.
[DEBUG] [17-12-2024 17:17:15:961.035 GMT] clientReadyHandler invoked
[DEBUG] [17-12-2024 17:17:15:961.046 GMT] Client is ready
[INFO ] [17-12-2024 17:17:15:961.071 GMT] Creating Kinesis Video Stream raspberry_pi_stream
[INFO ] [17-12-2024 17:17:15:961.086 GMT] createKinesisVideoStream(): Creating Kinesis Video Stream.
[INFO ] [17-12-2024 17:17:15:961.097 GMT] logStreamInfo(): SDK version: 
[DEBUG] [17-12-2024 17:17:15:961.107 GMT] logStreamInfo(): Kinesis Video Stream Info
[DEBUG] [17-12-2024 17:17:15:961.117 GMT] logStreamInfo():      Stream name: raspberry_pi_stream 
[DEBUG] [17-12-2024 17:17:15:961.131 GMT] logStreamInfo():      Streaming type: STREAMING_TYPE_REALTIME 
[DEBUG] [17-12-2024 17:17:15:961.144 GMT] logStreamInfo():      Content type: video/h264 
[DEBUG] [17-12-2024 17:17:15:961.158 GMT] logStreamInfo():      Max latency (100ns): 600000000
[DEBUG] [17-12-2024 17:17:15:961.171 GMT] logStreamInfo():      Fragment duration (100ns): 20000000
[DEBUG] [17-12-2024 17:17:15:961.184 GMT] logStreamInfo():      Key frame fragmentation: Yes
[DEBUG] [17-12-2024 17:17:15:961.196 GMT] logStreamInfo():      Use frame timecodes: Yes
[DEBUG] [17-12-2024 17:17:15:961.209 GMT] logStreamInfo():      Absolute frame timecodes: Yes
[DEBUG] [17-12-2024 17:17:15:961.221 GMT] logStreamInfo():      Nal adaptation flags: 0
[DEBUG] [17-12-2024 17:17:15:961.235 GMT] logStreamInfo():      Average bandwidth (bps): 4194304
[DEBUG] [17-12-2024 17:17:15:961.248 GMT] logStreamInfo():      Framerate: 25
[DEBUG] [17-12-2024 17:17:15:961.262 GMT] logStreamInfo():      Buffer duration (100ns): 1200000000
[DEBUG] [17-12-2024 17:17:15:961.275 GMT] logStreamInfo():      Replay duration (100ns): 400000000
[DEBUG] [17-12-2024 17:17:15:961.287 GMT] logStreamInfo():      Connection Staleness duration (100ns): 600000000
[DEBUG] [17-12-2024 17:17:15:961.301 GMT] logStreamInfo():      Store Pressure Policy: 1
[DEBUG] [17-12-2024 17:17:15:961.314 GMT] logStreamInfo():      View Overflow Policy: 1
[DEBUG] [17-12-2024 17:17:15:961.327 GMT] logStreamInfo():      Allow stream creation: Yes
[DEBUG] [17-12-2024 17:17:15:961.339 GMT] logStreamInfo():      Segment UUID: NULL
[DEBUG] [17-12-2024 17:17:15:961.353 GMT] logStreamInfo():      Frame ordering mode: 0
[DEBUG] [17-12-2024 17:17:15:961.365 GMT] logStreamInfo(): Track list
[DEBUG] [17-12-2024 17:17:15:961.374 GMT] logStreamInfo():      Track id: 1
[DEBUG] [17-12-2024 17:17:15:961.387 GMT] logStreamInfo():      Track name: kinesis_video
[DEBUG] [17-12-2024 17:17:15:961.401 GMT] logStreamInfo():      Codec id: V_MPEG4/ISO/AVC
[DEBUG] [17-12-2024 17:17:15:961.413 GMT] logStreamInfo():      Track type: TRACK_INFO_TYPE_VIDEO
[DEBUG] [17-12-2024 17:17:15:961.432 GMT] logStreamInfo():      Track cpd: NULL
[DEBUG] [17-12-2024 17:17:15:961.524 GMT] setRequestHeader(): Appending header to request: user-agent -> AWS-SDK-KVS/1.5.3 GCC/12.2.0 Linux/6.6.62+rpt-rpi-2712 aarch64 CPPSDK
[DEBUG] [17-12-2024 17:17:15:961.650 GMT] setRequestHeader(): Appending header to request: host -> kinesisvideo.us-east-1.amazonaws.com/describeStream
[DEBUG] [17-12-2024 17:17:15:961.732 GMT] setRequestHeader(): Appending header to request: X-Amz-Date -> 20241217T171715Z
[DEBUG] [17-12-2024 17:17:15:961.783 GMT] setRequestHeader(): Appending header to request: content-type -> application/json
[DEBUG] [17-12-2024 17:17:15:961.842 GMT] setRequestHeader(): Appending header to request: content-length -> 40
[DEBUG] [17-12-2024 17:17:15:961.920 GMT] setRequestHeader(): Appending header to request: Authorization -> AWS4-HMAC-SHA256 Credential=ASIA3FLDXY3CHJRP7JCY/20241217/us-east-1/kinesisvideo/aws4_request, SignedHeaders=host;user-agent;x-amz-date, Signature=254af9b997afdd7b9b11e400aa4d2285c904b76ae551452018a4ca4168aeca8b
[DEBUG] [17-12-2024 17:17:15:961.983 GMT] setRequestHeader(): Appending header to request: x-amz-security-token -> 
[INFO ] [17-12-2024 17:17:16:917.924 GMT] writeHeaderCallback(): RequestId: 8845b
[DEBUG] [17-12-2024 17:17:17:937.091 GMT] describeStreamCurlHandler(): [raspberry_pi_stream] DescribeStream API response: {"StreamInfo":{"CreationTime":1.734341183193E9,"DataRetentionInHours":24,"DeviceName":null,"IngestionConfiguration":null,"KmsKeyId":"arn:aws:kms:us-east-1:8:alias/aws/kinesisvideo","MediaType":null,"Status":"ACTIVE","StreamARN":"arn:aws:kinesisvideo:us-east-1:767397709508:stream/raspberry_pi_stream/1734341183193","StreamName":"raspberry_pi_stream","Version":"3KmtzX7PORhFxtulqWIx"}}
[INFO ] [17-12-2024 17:17:17:938.511 GMT] describeStreamResultEvent(): Describe stream result event.
[WARN ] [17-12-2024 17:17:17:938.553 GMT] describeStreamResult(): Retention period returned from the DescribeStream call doesn't match the one specified in the StreamInfo
[WARN ] [17-12-2024 17:17:17:938.561 GMT] describeStreamResult(): Content type returned from the DescribeStream(null) call doesn't match the one specified in the StreamInfo(video/h264)
[WARN ] [17-12-2024 17:17:17:938.568 GMT] describeStreamResult(): [raspberry_pi_stream] Retention period returned from the DescribeStream call doesn't match the one specified in the StreamInfo
[WARN ] [17-12-2024 17:17:17:938.574 GMT] describeStreamResult(): [raspberry_pi_stream] Content type returned from the DescribeStream call doesn't match the one specified in the StreamInfo
[DEBUG] [17-12-2024 17:17:17:938.930 GMT] setRequestHeader(): Appending header to request: user-agent -> AWS-SDK-KVS/1.5.3 GCC/12.2.0 Linux/6.6.62+rpt-rpi-2712 aarch64 CPPSDK
[DEBUG] [17-12-2024 17:17:17:939.017 GMT] setRequestHeader(): Appending header to request: host -> kinesisvideo.us-east-1.amazonaws.com/getDataEndpoint
[DEBUG] [17-12-2024 17:17:17:939.034 GMT] setRequestHeader(): Appending header to request: X-Amz-Date -> 20241217T171717Z
[DEBUG] [17-12-2024 17:17:17:939.042 GMT] setRequestHeader(): Appending header to request: content-type -> application/json
[DEBUG] [17-12-2024 17:17:17:939.049 GMT] setRequestHeader(): Appending header to request: content-length -> 65
[DEBUG] [17-12-2024 17:17:17:939.082 GMT] setRequestHeader(): Appending header to request: Authorization -> AWS4-HMAC-SHA256 Credential=ASIA3FLDXY3CHJRP7JCY/20241217/us-east-1/kinesisvideo/aws4_request, SignedHeaders=host;user-agent;x-amz-date, Signature=b65cfd0b2024be4ef86da54f9f828a878a35075e93eac3a07251de4e5045e82f
[DEBUG] [17-12-2024 17:17:17:939.103 GMT] setRequestHeader(): Appending header to request: x-amz-security-token -> 
[INFO ] [17-12-2024 17:17:18:802.533 GMT] writeHeaderCallback(): RequestId: dc29-481d-9635-1471c4994c0f
[DEBUG] [17-12-2024 17:17:18:802.788 GMT] getStreamingEndpointCurlHandler(): [raspberry_pi_stream] GetStreamingEndpoint API response: {"DataEndpoint":"https://s-.kinesisvideo.us-east-1.amazonaws.com"}
[... 1 lines cut ...]
[DEBUG] [17-12-2024 17:17:18:805.033 GMT] getStreamingTokenHandler invoked
[DEBUG] [17-12-2024 17:17:18:805.132 GMT] Refreshing credentials. Force refreshing: 1 Now time is: 1734455838805126613 Expiration: 1734459435
[INFO ] [17-12-2024 17:17:18:805.210 GMT] New credentials expiration is 1734459438
[INFO ] [17-12-2024 17:17:18:805.957 GMT] getStreamingTokenResultEvent(): Get streaming token result event.
[DEBUG] [17-12-2024 17:17:18:806.092 GMT] streamReadyHandler invoked
[... cut ...]
[DEBUG] [17-12-2024 17:17:19:653.470 GMT] freeKinesisVideoClientInternal(): Total allocated memory 0
[INFO ] [17-12-2024 17:17:19:653.528 GMT] Completed freeing client
[WARN ] [17-12-2024 17:17:19:653.551 GMT] curlApiCallbacksShutdown(): curlApiCallbacksShutdown called when already in progress of shutting down
root@raspberrypi:/greengrass/v2/work/com.dev.kvsComponent/amazon-kinesis-video-streams-producer-sdk-cpp/build# ^C
root@raspberrypi:/greengrass/v2/work/com.dev.kvsComponent/amazon-kinesis-video-streams-producer-sdk-cpp/build# `

stefankiesz, 2024-12-17:
Hi, I see the error is `Service Unavailable (503)` coming from `gstrtspsrc`. Can you provide the full pipeline or which sample you are streaming from.

srivatsavavuppala, 2024-12-17:
i am using the same
[amazon-kinesis-video-streams-producer-sdk-cpp](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp)

and the complete error i am attaching below

`2024-12-17T19:01:47.763Z [WARN] (Copier) com.dev.kvsComponent: stderr. 2024-12-17 19:01:47 - INFO - [Thread-79 (monitor_process)] - KVS Stream Output: [INFO ] [17-12-2024 19:01:46:878.611 GMT] Using region: us-east-1. {scriptName=services.com.dev.kvsComponent.lifecycle.Run.Script, serviceName=com.dev.kvsComponent, currentState=RUNNING}
2024-12-17T19:01:47.764Z [WARN] (Copier) com.dev.kvsComponent: stderr. 2024-12-17 19:01:47 - INFO - [Thread-79 (monitor_process)] - KVS Stream Output: [INFO ] [17-12-2024 19:01:46:878.672 GMT] Using aws credentials for Kinesis Video Streams. {scriptName=services.com.dev.kvsComponent.lifecycle.Run.Script, serviceName=com.dev.kvsComponent, currentState=RUNNING}
2024-12-17T19:01:47.764Z [WARN] (Copier) com.dev.kvsComponent: stderr. 2024-12-17 19:01:47 - INFO - [Thread-79 (monitor_process)] - KVS Stream Output: [INFO ] [17-12-2024 19:01:46:878.678 GMT] Session token detected.. {scriptName=services.com.dev.kvsComponent.lifecycle.Run.Script, serviceName=com.dev.kvsComponent, currentState=RUNNING}
2024-12-17T19:01:47.764Z [WARN] (Copier) com.dev.kvsComponent: stderr. 2024-12-17 19:01:47 - INFO - [Thread-79 (monitor_process)] - KVS Stream Output: [INFO ] [17-12-2024 19:01:46:879.431 GMT] createKinesisVideoClient(): Creating Kinesis Video Client. {scriptName=services.com.dev.kvsComponent.lifecycle.Run.Script, serviceName=com.dev.kvsComponent, currentState=RUNNING}
2024-12-17T19:01:47.764Z [WARN] (Copier) com.dev.kvsComponent: stderr. 2024-12-17 19:01:47 - INFO - [Thread-79 (monitor_process)] - KVS Stream Output: [WARN ] [17-12-2024 19:01:46:879.459 GMT] fixupClientInfo(): Connection timeout is invalid...setting to default. {scriptName=services.com.dev.kvsComponent.lifecycle.Run.Script, serviceName=com.dev.kvsComponent, currentState=RUNNING}
2024-12-17T19:01:47.764Z [WARN] (Copier) com.dev.kvsComponent: stderr. 2024-12-17 19:01:47 - INFO - [Thread-79 (monitor_process)] - KVS Stream Output: [WARN ] [17-12-2024 19:01:46:879.464 GMT] fixupClientInfo(): Completion timeout is invalid...setting to default. {scriptName=services.com.dev.kvsComponent.lifecycle.Run.Script, serviceName=com.dev.kvsComponent, currentState=RUNNING}
2024-12-17T19:01:47.764Z [WARN] (Copier) com.dev.kvsComponent: stderr. 2024-12-17 19:01:47 - INFO - [Thread-79 (monitor_process)] - KVS Stream Output: [INFO ] [17-12-2024 19:01:46:879.477 GMT] heapInitialize(): Initializing native heap with limit size 134217728, spill ratio 0% and flags 0x00000001. {scriptName=services.com.dev.kvsComponent.lifecycle.Run.Script, serviceName=com.dev.kvsComponent, currentState=RUNNING}
2024-12-17T19:01:47.764Z [WARN] (Copier) com.dev.kvsComponent: stderr. 2024-12-17 19:01:47 - INFO - [Thread-79 (monitor_process)] - KVS Stream Output: [INFO ] [17-12-2024 19:01:46:879.481 GMT] heapInitialize(): Creating AIV heap.. {scriptName=services.com.dev.kvsComponent.lifecycle.Run.Script, serviceName=com.dev.kvsComponent, currentState=RUNNING}
2024-12-17T19:01:47.764Z [WARN] (Copier) com.dev.kvsComponent: stderr. 2024-12-17 19:01:47 - INFO - [Thread-79 (monitor_process)] - KVS Stream Output: [INFO ] [17-12-2024 19:01:46:879.502 GMT] heapInitialize(): Heap is initialized OK. {scriptName=services.com.dev.kvsComponent.lifecycle.Run.Script, serviceName=com.dev.kvsComponent, currentState=RUNNING}
2024-12-17T19:01:47.764Z [WARN] (Copier) com.dev.kvsComponent: stderr. 2024-12-17 19:01:47 - INFO - [Thread-79 (monitor_process)] - KVS Stream Output: [DEBUG] [17-12-2024 19:01:46:879.568 GMT] getSecurityTokenHandler invoked. {scriptName=services.com.dev.kvsComponent.lifecycle.Run.Script, serviceName=com.dev.kvsComponent, currentState=RUNNING}
2024-12-17T19:01:47.764Z [WARN] (Copier) com.dev.kvsComponent: stderr. 2024-12-17 19:01:47 - INFO - [Thread-79 (monitor_process)] - KVS Stream Output: [DEBUG] [17-12-2024 19:01:46:879.584 GMT] Refreshing credentials. Force refreshing: 0 Now time is: 1734462106879576915 Expiration: 0. {scriptName=services.com.dev.kvsComponent.lifecycle.Run.Script, serviceName=com.dev.kvsComponent, currentState=RUNNING}
2024-12-17T19:01:47.764Z [WARN] (Copier) com.dev.kvsComponent: stderr. 2024-12-17 19:01:47 - INFO - [Thread-79 (monitor_process)] - KVS Stream Output: [INFO ] [17-12-2024 19:01:46:879.590 GMT] New credentials expiration is 1734465706. {scriptName=services.com.dev.kvsComponent.lifecycle.Run.Script, serviceName=com.dev.kvsComponent, currentState=RUNNING}
2024-12-17T19:01:47.764Z [WARN] (Copier) com.dev.kvsComponent: stderr. 2024-12-17 19:01:47 - INFO - [Thread-79 (monitor_process)] - KVS Stream Output: [INFO ] [17-12-2024 19:01:46:879.603 GMT] createDeviceResultEvent(): Create device result event.. {scriptName=services.com.dev.kvsComponent.lifecycle.Run.Script, serviceName=com.dev.kvsComponent, currentState=RUNNING}
2024-12-17T19:01:47.764Z [WARN] (Copier) com.dev.kvsComponent: stderr. 2024-12-17 19:01:47 - INFO - [Thread-79 (monitor_process)] - KVS Stream Output: [DEBUG] [17-12-2024 19:01:46:879.611 GMT] clientReadyHandler invoked. {scriptName=services.com.dev.kvsComponent.lifecycle.Run.Script, serviceName=com.dev.kvsComponent, currentState=RUNNING}
[... cut ...]
2024-12-17T19:01:48.858Z [WARN] (Copier) com.dev.kvsComponent: stderr. 2024-12-17 19:01:48 - INFO - [Dummy-7] - Current stream status: running. {scriptName=services.com.dev.kvsComponent.lifecycle.Run.Script, serviceName=com.dev.kvsComponent, currentState=RUNNING}
2024-12-17T19:01:49.387Z [WARN] (Copier) com.dev.kvsComponent: stderr. 2024-12-17 19:01:49 - INFO - [Dummy-7] - Shadow update accepted:awsiot.iotshadow.ShadowState(desired=None, desired_is_nullable=False, reported={'stream_status': 'running'}, reported_is_nullable=False). {scriptName=services.com.dev.kvsComponent.lifecycle.Run.Script, serviceName=com.dev.kvsComponent, currentState=RUNNING`

srivatsavavuppala, 2024-12-17:
> Hi, I see the error is `Service Unavailable (503)` coming from `gstrtspsrc`. Can you provide the full pipeline or which sample you are streaming from.

Hi @stefankiesz 

i am using the same
[amazon-kinesis-video-streams-producer-sdk-cpp](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp)

and the complete error i am attaching below

2024-12-17T19:01:47.763Z [WARN] (Copier) com.dev.kvsComponent: stderr. 2024-12-17 19:01:47 - INFO - [Thread-79 (monitor_process)] - KVS Stream Output: [INFO ] [17-12-2024 19:01:46:878.611 GMT] Using region: us-east-1. {scriptName=services.com.dev.kvsComponent.lifecycle.Run.Script, serviceName=com.dev.kvsComponent, currentState=RUNNING} 2024-12-17T19:01:47.764Z [WARN] (Copier) com.dev.kvsComponent: stderr. 2024-12-17 19:01:47 - INFO - [Thread-79 (monitor_process)] - KVS Stream Output: [INFO ] [17-12-2024 19:01:46:878.672 GMT] Using aws credentials for Kinesis Video Streams. {scriptName=services.com.dev.kvsComponent.lifecycle.Run.Script, serviceName=com.dev.kvsComponent, currentState=RUNNING} 2024-12-17T19:01:47.764Z [WARN] (Copier) com.dev.kvsComponent: stderr. 2024-12-17 19:01:47 - INFO - [Thread-79 (monitor_process)] - KVS Stream Output: [INFO ] [17-12-2024 19:01:46:878.678 GMT] Session token detected.. {scriptName=services.com.dev.kvsComponent.lifecycle.Run.Script, serviceName=com.dev.kvsComponent, currentState=RUNNING} 2024-12-17T19:01:47.764Z [WARN] (Copier) com.dev.kvsComponent: stderr. 2024-12-17 19:01:47 - INFO - [Thread-79 (monitor_process)] - KVS Stream Output: [INFO ] [17-12-2024 19:01:46:879.431 GMT] createKinesisVideoClient(): Creating Kinesis Video Client. {scriptName=services.com.dev.kvsComponent.lifecycle.Run.Script, serviceName=com.dev.kvsComponent, currentState=RUNNING} 2024-12-17T19:01:47.764Z [WARN] (Copier) com.dev.kvsComponent: stderr. 2024-12-17 19:01:47 - INFO - [Thread-79 (monitor_process)] - KVS Stream Output: [WARN ] [17-12-2024 19:01:46:879.459 GMT] fixupClientInfo(): Connection timeout is invalid...setting to default. {scriptName=services.com.dev.kvsComponent.lifecycle.Run.Script, serviceName=com.dev.kvsComponent, currentState=RUNNING} 2024-12-17T19:01:47.764Z [WARN] (Copier) com.dev.kvsComponent: stderr. 2024-12-17 19:01:47 - INFO - [Thread-79 (monitor_process)] - KVS Stream Output: [WARN ] [17-12-2024 19:01:46:879.464 GMT] fixupClientInfo(): Completion timeout is invalid...setting to default. {scriptName=services.com.dev.kvsComponent.lifecycle.Run.Script, serviceName=com.dev.kvsComponent, currentState=RUNNING} 2024-12-17T19:01:47.764Z [WARN] (Copier) com.dev.kvsComponent: stderr. 2024-12-17 19:01:47 - INFO - [Thread-79 (monitor_process)] - KVS Stream Output: [INFO ] [17-12-2024 19:01:46:879.477 GMT] heapInitialize(): Initializing native heap with limit size 134217728, spill ratio 0% and flags 0x00000001. {scriptName=services.com.dev.kvsComponent.lifecycle.Run.Script, serviceName=com.dev.kvsComponent, currentState=RUNNING} 2024-12-17T19:01:47.764Z [WARN] (Copier) com.dev.kvsComponent: stderr. 2024-12-17 19:01:47 - INFO - [Thread-79 (monitor_process)] - KVS Stream Output: [INFO ] [17-12-2024 19:01:46:879.481 GMT] heapInitialize(): Creating AIV heap.. {scriptName=services.com.dev.kvsComponent.lifecycle.Run.Script, serviceName=com.dev.kvsComponent, currentState=RUNNING} 2024-12-17T19:01:47.764Z [WARN] (Copier) com.dev.kvsComponent: stderr. 2024-12-17 19:01:47 - INFO - [Thread-79 (monitor_process)] - KVS Stream Output: [INFO ] [17-12-2024 19:01:46:879.502 GMT] heapInitialize(): Heap is initialized OK. {scriptName=services.com.dev.kvsComponent.lifecycle.Run.Script, serviceName=com.dev.kvsComponent, currentState=RUNNING} 2024-12-17T19:01:47.764Z [WARN] (Copier) com.dev.kvsComponent: stderr. 2024-12-17 19:01:47 - INFO - [Thread-79 (monitor_process)] - KVS Stream Output: [DEBUG] [17-12-2024 19:01:46:879.568 GMT] getSecurityTokenHandler invoked. {scriptName=services.com.dev.kvsComponent.lifecycle.Run.Script, serviceName=com.dev.kvsComponent, currentState=RUNNING} 2024-12-17T19:01:47.764Z [WARN] (Copier) com.dev.kvsComponent: stderr. 2024-12-17 19:01:47 - INFO - [Thread-79 (monitor_process)] - KVS Stream Output: [DEBUG] [17-12-2024 19:01:46:879.584 GMT] Refreshing credentials. Force refreshing: 0 Now time is: 1734462106879576915 Expiration: 0. {scriptName=services.com.dev.kvsComponent.lifecycle.Run.Script, serviceName=com.dev.kvsComponent, currentState=RUNNING} 2024-12-17T19:01:47.764Z [WARN] (Copier) com.dev.kvsComponent: stderr. 2024-12-17 19:01:47 - INFO - [Thread-79 (monitor_process)] - KVS Stream Output: [INFO ] [17-12-2024 19:01:46:879.590 GMT] New credentials expiration is 1734465706. {scriptName=services.com.dev.kvsComponent.lifecycle.Run.Script, serviceName=com.dev.kvsComponent, currentState=RUNNING} 2024-12-17T19:01:47.764Z [WARN] (Copier) com.dev.kvsComponent: stderr. 2024-12-17 19:01:47 - INFO - [Thread-79 (monitor_process)] - KVS Stream Output: [INFO ] [17-12-2024 19:01:46:879.603 GMT] createDeviceResultEvent(): Create device result event.. {scriptName=services.com.dev.kvsComponent.lifecycle.Run.Script, serviceName=com.dev.kvsComponent, currentState=RUNNING} 2024-12-17T19:01:47.764Z [WARN] (Copier) com.dev.kvsComponent: stderr. 2024-12-17 19:01:47 - INFO - [Thread-79 (monitor_process)] - KVS Stream Output: [DEBUG] [17-12-2024 19:01:46:879.611 GMT] clientReadyHandler invoked. {scriptName=services.com.dev.kvsComponent.lifecycle.Run.Script, serviceName=com.dev.kvsComponent, currentState=RUNNING} 2024-12-17T19:01:47. [... cut ...] =services.com.dev.kvsComponent.lifecycle.Run.Script, serviceName=com.dev.kvsComponent, currentState=RUNNING} 2024-12-17T19:01:49.387Z [WARN] (Copier) com.dev.kvsComponent: stderr. 2024-12-17 19:01:49 - INFO - [Dummy-7] - Shadow update accepted:awsiot.iotshadow.ShadowState(desired=None, desired_is_nullable=False, reported={'stream_status': 'running'}, reported_is_nullable=False). {scriptName=services.com.dev.kvsComponent.lifecycle.Run.Script, serviceName=com.dev.kvsComponent, currentState=RUNNING

stefankiesz, 2024-12-17:
```
Error received from element source: Unhandled error
Debugging information: ../gst/rtsp/gstrtspsrc.c(6954): gst_rtspsrc_send (): /GstPipeline:rtsp-kinesis-pipeline/GstRTSPSrc:source:
Service Unavailable (503)
```

The error is not KVS related and could be from a malformed GStreamer pipeline or from a network issue affecting the `rtspsrc` element. Please refer to [GStreamer](https://gstreamer.freedesktop.org/documentation/index.html?gi-language=c) and [rtspsrc](https://gstreamer.freedesktop.org/documentation/rtsp/rtspsrc.html?gi-language=c) documentation to debug. As a potential quick fix, you can try NULLing and restarting the pipeline in your application upon such an error.

closed 2024-12-17 by stefankiesz