awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #1223: [Erronedus PipelineL No Element "kvssink" ]

waleedshkt, 2025-01-11:
**Logging**
```
$ gst-launch-1.0 -v v4l2src device=/dev/video1 ! videoconvert ! video/x-raw,format=I420,width=640,height=480,framerate=30/1 ! x264enc  bframes=0 key-int-max=45 bitrate=500 tune=zerolatency ! video/x-h264,stream-format=avc,alignment=au ! kvssink stream-name=test-stream storage-size=128 access-key="..." secret-key="..."

WARNING: erroneous pipeline: no element "kvssink"
```

**Describe the bug**
I have made a near successful attempt to connect to launch live stream. I followed the given official guide of this repo and further of linux system. My OS is Windows 10 and with WSL2 I made a great effort to custom build a video-device supporting linux kernel. All worked. Project got complied successfully, path variables got loaded from the root of repo, and `kvssink` info got displayed. But upon executing above command, it failed.

I doubled checked the USB HD camera was attached using `lsusb` and it was. I also checked by running `/dev/device*` and there were two devices. Though strangely, upon running `gst-device-monitor-1.0`, it did not show up in the list; only the audio sources and sinks were present

This now begs the question, whether it is the camera which is incompatible or some other undocumented steps along the build process that I missed. The camera is not a webcam but a tiny one which runs with android-app and claims to work on PC 

Another thing I did during debugging was I removed the last kvssink command from the above larger pipeline command, and it threw another error: 

`WARNING: erroneous pipeline: no element "video"`

**To Reproduce**
Steps to reproduce the behavior:
1. In WSL2, follow the documentation
2. Using Usblib-win, attach usb camera to WSL2
3. Run the above command
4. See error

**Expected behavior**
Log or response from KVS telling video is streaming

**SDK version number**
I just cloned this repo and complied the project in a newly created `build` folder as directed in docs

**Platform (please complete the following information):**
 - OS: Windows 10 with WSL2 (Ubuntu 22.04 w/ video-enabled kernel)
 - Processor: Intel i5

stefankiesz, 2025-01-13:
Hi, can you confirm when you run `gst-inspect-1.0 kvssink` is it displaying info about kvssink or do you get a similar error?

waleedshkt, 2025-01-13:
It displays info about kvssink

sirknightj, 2025-04-09:
Hi @ waleedshkt,

I've added WSL build & running kvssink to the GitHub actions CI, which runs kvssink and checks that the video is generated using `videotestsrc` element. It seems to be working fine there, so I don't think it's a `kvssink` issue. You can try running the same pipeline to confirm.

For GStreamer pipelines, it needs to end in a sink element (can't end with the caps element, that's why you got the no element 'video' error).

You can try these pipelines to verify that GStreamer can access your camera, then as an intermediate step, send it to a local video file. 

Here are some examples:

```sh
gst-launch-1.0 autovideosrc ! fakesink
```

```sh
gst-launch-1.0 -v -e videotestsrc ! video/x-raw,height=1080,width=720,framerate=15/1 ! x264enc byte-stream=true key-int-max=30 ! h264parse config-interval=-1 ! matroskamux ! filesink location=out.mkv
```

closed 2025-04-09 by sirknightj