awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #1225: [BUG]GStreamer unit tests with Segmentation Fault

shotamishima, 2025-01-22:
**Logging**
```
Unexpected critical/warning: gst_pad_get_peer: assertion 'GST_IS_PAD (pad)' failed

Stack trace:
0   libgstreamer-1.0.0.dylib            0x0000000102f3db8c gst_debug_get_stack_trace + 52
1   libgstcheck-1.0.0.dylib             0x0000000102c26d3c gst_check_log_critical_func + 144
2   libglib-2.0.0.dylib                 0x0000000103063dd0 g_logv + 828
3   libglib-2.0.0.dylib                 0x0000000103063a24 g_log + 28
4   libglib-2.0.0.dylib                 0x0000000103063814 g_return_if_fail_warning + 32
5   libgstreamer-1.0.0.dylib            0x0000000102f4cc8c gst_pad_get_peer + 128
6   libgstcheck-1.0.0.dylib             0x0000000102c27448 gst_check_teardown_pad_by_name + 24
7   gstkvsplugintest                    0x0000000102b71a2c _ZL23kvsproducersinkteststopi + 172
8   libgstcheck-1.0.0.dylib             0x0000000102c32eb0 srunner_run_tagged + 1780
9   libgstcheck-1.0.0.dylib             0x0000000102c287e0 gst_check_run_suite + 116
10  gstkvsplugintest                    0x0000000102b72e8c main + 56
11  dyld                                0x00000001950d0274 start + 2840

Unexpected critical/warning: gst_pad_get_peer: assertion 'GST_IS_PAD (pad)' failed

Stack trace:
0   libgstreamer-1.0.0.dylib            0x0000000102f3db8c gst_debug_get_stack_trace + 52
1   libgstcheck-1.0.0.dylib             0x0000000102c26d3c gst_check_log_critical_func + 144
2   libglib-2.0.0.dylib                 0x0000000103063dd0 g_logv + 828
3   libglib-2.0.0.dylib                 0x0000000103063a24 g_log + 28
4   libglib-2.0.0.dylib                 0x0000000103063814 g_return_if_fail_warning + 32
5   libgstreamer-1.0.0.dylib            0x0000000102f4cc8c gst_pad_get_peer + 128
6   libgstcheck-1.0.0.dylib             0x0000000102c27448 gst_check_teardown_pad_by_name + 24
7   gstkvsplugintest                    0x0000000102b723ac _ZL37check_properties_are_passed_correctlyi + 2168
8   libgstcheck-1.0.0.dylib             0x0000000102c32eb0 srunner_run_tagged + 1780
9   libgstcheck-1.0.0.dylib             0x0000000102c287e0 gst_check_run_suite + 116
10  gstkvsplugintest                    0x0000000102b72e8c main + 56
11  dyld                                0x00000001950d0274 start + 2840

0%: Checks: 6, Failures: 4, Errors: 2
../subprojects/gstreamer/libs/gst/check/gstcheck.c:520:E:AllStateChangeTests:kvsproducersinktestplay:0: (after this point) Received signal 11 (Segmentation fault: 11)
../subprojects/gstreamer/libs/gst/check/gstcheck.c:520:E:AllStateChangeTests:kvsproducersinktestpause:0: (after this point) Received signal 11 (Segmentation fault: 11)
../subprojects/gstreamer/libs/gst/check/gstcheck.c:648:F:AllStateChangeTests:kvsproducersinktestplaytopausetoplay:0: Could not get sink pad from kvssink
../subprojects/gstreamer/libs/gst/check/gstcheck.c:286:F:AllStateChangeTests:kvsproducersinkteststop:0: Unexpected critical/warning: gst_pad_get_peer: assertion 'GST_IS_PAD (pad)' failed
../subprojects/gstreamer/libs/gst/check/gstcheck.c:286:F:AllStateChangeTests:check_properties_are_passed_correctly:0: Unexpected critical/warning: gst_pad_get_peer: assertion 'GST_IS_PAD (pad)' failed
../subprojects/gstreamer/libs/gst/check/gstcheck.c:648:F:AllStateChangeTests:check_playing_to_paused_and_back_to_playing:0: Could not get sink pad from kvssink
```

**Describe the bug**
On master branch, I tried to run unit tests for kvssink after a build but the above error logs showed up. The environment I use is M3 MacOS Sequoia. Thank you for your support.

**To Reproduce**
Steps to reproduce the behavior:
1. Build the project by `cmake -DBUILD_GSTREAMER_PLUGIN=TRUE -DBUILD_TEST=TRUE ..` on `./build` directory
2. Set AWS credentials for environment variables
3. Execute `./tst/gstkvsplugintest`
4. See error

**Expected behavior**
The all of test cases should pass since I don't touch anything from master branch.

**SDK version number**
v3.4.2

**Platform (please complete the following information):**
 - OS: macOS Sequoia 15.2
 - Processor: Apple Sillicon M3 pro

sirknightj, 2025-01-24:
Hi @shotamishima, thanks for the report.

I was able to reproduce this on my mac locally. It seems to be just an issue with the unit tests. The [CI](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/1ba18727583af2a685a81f0b5bfda511143dd113/.github/workflows/raspberry-pi.yaml#L75) which runs `kvssink` in an actual Gstreamer pipeline is working as expected.

We'll be looking into enhancing the test suite at a future date. Is this currently blocking your development?

shotamishima, 2025-01-24:
Thanks @sirknightj for taking care of it. It's not a blocker to me for now.