awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #1233: Unable to Run Multiple Branches in a Single GStreamer Pipeline with Two Different RTSP Sources

Shankar-Bharadwaj, 2025-02-27:
Description: 
I am trying to create a GStreamer pipeline that takes input from two different RTSP streams and processes them in parallel using separate branches. However, when I attempt to run the pipeline, one of the RTSP sources fails to connect, and the pipeline does not function as expected. 

Both the streams work well independently on two different gstreamer pipelines.

Expected Behavior:
Both RTSP sources should be able to connect successfully, and the pipeline should stream both inputs in parallel without errors.

Observed Behavior:
One of the RTSP sources fails to connect, and I receive the following error:

ERROR: from element /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc1: Could not open resource for reading and writing.
Additional debug info:
gstrtspsrc.c(6842): gst_rtspsrc_retrieve_sdp (): /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc1:
Failed to connect. (Generic error)
2025-02-27 13:27:10 [140736842405632] Execution ended after 0:00:19.523974842
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...
INFO - Freeing Kinesis Video Stream KVSStream2

Is there a known limitation when handling multiple rtspsrc elements in a single pipeline?

sirknightj, 2025-02-27:
Hi @Shankar-Bharadwaj, thanks for your interest in kvssink.

I was not able to reproduce the issue. Here's what I ran:

```
docker run -d --rm -e RTSP_PORT=8558 -p 8558:8558 -e GST_PIPELINE="videotestsrc ! clockoverlay ! videoscale ! video/x-raw,width=640,height=480 ! capsfilter ! queue ! x264enc bframes=0 key-int-max=45 bitrate=512 ! rtph264pay name=pay0 
pt=96" ullaakut/rtspatt
docker run --rm -d -e RTSP_PORT=8559 -p 8559:8559 -e GST_PIPELINE="videotestsrc ! clockoverlay ! videoscale ! video/x-raw,width=640,height=480 ! capsfilter ! queue ! x264enc bframes=0 key-int-max=45 bitrate=512 ! rtph264pay name=pay0 pt=96" ullaakut/rtspatt

gst-launch-1.0 \
  rtspsrc location="rtsp://0.0.0.0:8558/live.sdp" short-header=TRUE ! rtph264depay ! h264parse ! kvssink stream-name="KVSStream1" storage-size=128 aws-region="us-west-2" \
  rtspsrc location="rtsp://0.0.0.0:8559/live.sdp" short-header=TRUE ! rtph264depay ! h264parse ! kvssink stream-name="KVSStream2" storage-size=128 aws-region="us-west-2"
```

I was able to stream both successfully.

[image]

You can also verify on your own setup with videotestsrc as well:
```
gst-launch-1.0 \
  videotestsrc is-live=true ! clockoverlay time-format="%a %B %d, %Y %I:%M:%S %p" ! video/x-raw,height=720,width=1080 ! x264enc ! h264parse ! kvssink stream-name="KVSStream1" storage-size=128 aws-region="us-west-2" \
  videotestsrc is-live=true ! clockoverlay time-format="%a %B %d, %Y %I:%M:%S %p" ! video/x-raw,height=720,width=1080 ! x264enc ! h264parse ! kvssink stream-name="KVSStream2" storage-size=128 aws-region="us-west-2"
```

If you're wondering about a single rtsp source crashing the entire pipeline even though the other one is working fine. We recommend creating two separate gst-launch instances, so that you are able to continuously retry connecting to the rtsp source in case of a network failure. rtspsrc doesn't provide retry behavior, that needs to be done through the application layer, which will involve tearing down the pipeline and trying again. You can put `&` at the end of the gst-launch-1.0 command to run multiple launch commands from a single terminal instance.

Shankar-Bharadwaj, 2025-02-27:
Hey, thank you for the reply!

Using the command that you provided,

```
gst-launch-1.0 \
  rtspsrc location="rtsp://0.0.0.0:8558/live.sdp" short-header=TRUE ! rtph264depay ! h264parse ! kvssink stream-name="KVSStream1" storage-size=128 aws-region="us-west-2" \
  rtspsrc location="rtsp://0.0.0.0:8559/live.sdp" short-header=TRUE ! rtph264depay ! h264parse ! kvssink stream-name="KVSStream2" storage-size=128 aws-region="us-west-2"
```

I was able to run both the streams using a single gstreamer pipeline.

But I want to understand why was I unable to achieve this using the following code,

```
gst-launch-1.0 \
  rtspsrc location="rtsp://url_1" protocols=tcp ! queue ! rtph265depay ! h265parse ! kvssink stream-name="KVSStream1" storage-size=128 aws-region="ap-south-1" \
  rtspsrc location="rtsp://url_2" protocols=tcp ! queue ! rtph265depay ! h265parse ! kvssink stream-name="KVSStream2" storage-size=128 aws-region="ap-south-1"
```

I thought queue would create a separate thread for every stream that I send to my pipeline. Please correct me if I'm wrong.

closed 2025-02-27 by Shankar-Bharadwaj